betaflight/betaflight #5852: Feature request: new default settings for 3.4

flint723, 2018-05-08:
I think there is a need for some updated defaults to take advantage of the features 3.4 (and previous releases) have.  It would be best to get some in quickly in my opinion, so we can get feedback on if they are good, or not, and what needs to change before final release.

I've made a list (gone through several revisions) which works pretty well on all of my 8k gyro/pid 5" quads (which range from light to modest weight, I do not own any heavy freestyle+gopro quads).  Not 100% good but it's defaults.  But not sure about 32k.  Of course current default is not good for 32k either, but these setting should be an improvement.  Ideally there should be a filter change when `gyro_use_32khz = on` because of the huge difference in filtering from hardware DLPF, but maybe that can't happen.  Anyway these are what I've come up with, I am open to suggestions.

```
set gyro_lowpass_hz = 120
set gyro_lowpass2_hz = 300
set gyro_lowpass_type = pt1
set gyro_lowpass2_type = pt1
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
feature DYNAMIC_FILTER

set dterm_lowpass_hz = 90
set dterm_lowpass2_hz = 180
set dterm_lowpass_type = pt1

feature ANTI_GRAVITY
set anti_gravity_gain = 2500
set setpoint_relax_ratio = 0
set dterm_setpoint_weight = 50
set throttle_boost = 50
set throttle_boost_cutoff = 40
set rc_interp_ch = RPYT
set p_pitch = 48
set i_pitch = 50
set d_pitch = 26
set p_roll = 46
set i_roll = 45
set d_roll = 24
set p_yaw = 55
set i_yaw = 45

set tpa_rate = 15
set tpa_breakpoint = 1050
```
About TPA breakpoint 1050, I tested 1500 in the first two revisions, then 1050 in 3rd (which I normally use).  None of the quads had a bad reaction to 1050, a couple of them really needed this setting.

flint723, 2018-05-14:
Okay, made several changes based on feedback, and they fly better on my quads too.

smasslich, 2018-05-17:
I tested these settings on JBF4 FC - clracingf4 target, CLracingF4S FC, and 2 Matek F405-OSD FC builds.  Frames are XHover Stingy, Chameleon, AstroX Switch Freestyle, and AstroX X5 silky respectively.  They worked great.  All running 8/8 and after initial test flight, I removed the 2nd gyro lpf.  Thanks!

seaweeduk, 2018-05-17:
I also removed the 2nd gyro lpf, the rest felt good. Only things I changed on the two builds I tested was to raise AGG (5000 and 4500) and the dterm LPF to 120. I also raised the P's up a little, but that's to be expected from default PIDs. Overall the settings felt really good compared to the previous defaults. 

I would also like to see joe lucid's 3.4 changes such as iterm rotation incorporated into the default settings once they have been finalized.

spatzengr, 2018-05-19:
I tested these as well and like their performance and filtering attenuation.  I think it strikes a good balance and provides more modern adequate default filtering (log below).  I would leave Stage 2 on by default because some folks may be enabling 32k and others not.  So the guide could be to fly OTB, and then if all cool and on 8k or 4k, consider turning off the Stage 2 LPF.  If on 32k, you would leave Stage 2 enabled.  OR if you have low motor spikes, you can as an alternative leave Stage 2 enabled (for 8k/4k) and turn off the Dyn. Notch (8k or 32k).

More the come with the Joe Lucid 3.4 I-term adjustments.

[2018.05.18-BF 3.4.0-Build 793_PT1=90+DyN+PT1=300_Dt=PT1-100,200-FIRST FLIGHT.TXT](https://github.com/betaflight/betaflight/files/2019055/2018.05.18-BF.3.4.0-Build.793_PT1.90.DyN.PT1.300_Dt.PT1-100.200-FIRST.FLIGHT.TXT)

SteveKfpv, 2018-05-22:
Looks good. Personally I think Yaw P is better in the 80s. I have had great success with prop-wash handling with Dterm lowpass on 100 and 350 but maybe that's not conservative enough for a default. Also, I have only gyro filter 1 on.

SteveKfpv, 2018-05-22:
Yeah fair enough, I had forgotten about the increased filtering for the faster gyros. A shame there isn't dynamic defaults based on gyro/loop time

seaweeduk, 2018-05-22:
After more time flying with these settings I would definitely prefer dterm_setpoint_weight to be increased. I run 1.5 on my builds, but I think 1 would be a good default.

flint723, 2018-05-22:
SteveKfpv, yes that would be ideal if filtering could change with the parameter `gyro_use_32khz`.  Another possibility would be using profile 1 for 8k specific defaults, and profile 2 for 32k.  Not sure this is a good idea though.

seaweekuk, I will consider it (I use higher weight as well) but with current default weight 0, might not be the best to make such a drastic change.  Not sure why it was changed to zero in the first place though, it used to default to 1 weight, 1 transition I believe.

seaweeduk, 2018-05-22:
I agree it's definitely a drastic change, I just haven't ever encountered anyone who preferred it below 1. 

Hopefully we can get some more feedback on this from other pilots before release. I feel like the two setpoint sliders are a bit of a missed opportunity in general in betaflight, they are not utilized by many pilots because it's difficult to understand how it works.

Saxin, 2018-05-22:
@flint723 what's the reasoning for 2nd lpf @ 300 Hz for 8k/8k ?

flint723, 2018-05-22:
Saxin, it isn't there specifically for 8k8k.  But the reasoning behind the value of 300hz, is that in general the dynamic notch loses effectiveness around 400hz, resulting in a second noise spike in a lot of logs (8k and 32k) anywhere from 400-600hz.  So 300hz lowpass2 is there to cut in by 350hz and start taking care of that second spike.

IMO lowpass2 isn't needed on 8k (2nd dterm pt1 takes care of this noise around 400-600 good enough) but my goal was to have as much filtering as possible (for 32k) and still fly "ok" (better than current defaults) on 8k8k.

Plasmatree, 2018-05-23:
> the dynamic notch loses effectiveness around 400hz (spends most of it's time <300hz), resulting in a second noise spike in a lot of logs (8k and 32k) anywhere from 400-600hz.

Isnt't the dynamic notch even limited to <500hz? On my 3" builds I can clearly see that at high throttle the pure motor noise exceeds the range of the dynamic notch. Could it be that the "second peak" you are talking about (existing in the cumulative spectrum I guess) is the same peak, just at another position and time? If there is always just 1 peak at a time from motors, it should be adressed more effectively by a higher limit for the dynamic notch, I think. Has anybody tried that? [This example is from my 3" quad](https://raw.githubusercontent.com/Plasmatree/PID-Analyzer/master/3inch_example_noise.png)

flint723, 2018-05-23:
Yes, dynamic notch center max is 500hz.  I've never seen it go that high but 5" motor rpm doesn't usually exceed 450hz or so, where 3" motor rpm would.  Even then sometimes dynamic notch doesn't follow motor rpm, if there happens to be a resonance.  On one quad I have, the dynamic notch just stays at 200-300hz regardless of throttle.  But I've seen very few logs with `debug_mode = fft_freq` (only my own) so don't have much data on this.

This "second peak" isn't usually a peak in prenotch gyro trace, it's just noise that hasn't been crushed by the dynamic notch.  It may be beneficial in some cases to raise dynamic notch operating range, I do not believe anyone has tried it.  But I think (don't know for sure) that there is some "weight" effect in dynamic notch, where lower frequency noise takes priority over higher frequency if the amplitude is the same.

atomiclama, 2018-05-24:
Thanks for doing this, settings working very well as a starting point on my hex. Roll pretty much spot on Pitch needs a little tweak.
If it's of any use my hex basic spec is as follows: -
5S 
2306 2100Kv DAL t5051
8k/8K Oneshot42 blheli_s

Lord-Quake, 2018-05-25:
Everyone,

For the last 4 days I was plagued with a problem that I consider dangerous. I am posting this here because it concerns the proposed default settings, however I don't know if the problem has to do with settings directly or the fault is a combination of factors.
I've tried several 3.4 builds with the same results. Right now I'm running build 859.
The installation is always the same. Delete the whole flash, install new FW, change my settings and then add above mentioned proposed default settings.
I arm my QC via stick. When I move the stick to the right to arm the QC the motors immediately go up to speed. If the QC would have props it could potentially lift off and flip which I consider to be a very dangerous situation. Now if I try to disarm by moving the throttle stick to the left the motors will fire up again. The only way to disarm the QC is to turn off the transmitter so the QC enters fail safe mode.
I'm no expert but it seems the min throttle turns to 1000 even though it is set to 1050. At least that is where started to concentrate my search for the possible fault. Long story short.... the culprit turned at to be  "set rc_interp_ch = RPYT". However I don't know if it is the cause of the problem. Anyhow, when I configure "set rc_interp_ch = RP" the problem disappears. BTW, the problem also shows up when settings "set rc_interp_ch = RPY".
I am pointing this problem out because if someone like me had a working QC and did updates they would get a bad surprise.
Please take a look at my DIFF (here from build 859) and look to see if a combination of factors can be found which would cause the unpleasant and dangerous situation.

`# version
# Betaflight / MATEKF405 (MKF4) 3.4.0 May 24 2018 / 14:05:19 (71a1a9789) MSP API: 1.39

# name
name QC_3

# resources

# mixer

# servo

# servo mix

# feature
feature -SOFTSERIAL
feature -TELEMETRY
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE

# map
map TAER1234

# serial

# led

# color

# mode_color

# aux
aux 0 2 0 900 1300 0
aux 1 2 0 1700 2100 0
aux 2 3 0 1700 2100 0
aux 3 13 2 1700 2100 0
aux 4 19 1 1700 2100 0
aux 5 28 0 1300 1700 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 4 s 1000
rxfail 5 s 1000
rxfail 6 s 2000

# master
set gyro_lowpass_hz = 120
set gyro_lowpass2_hz = 300
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set acc_calibration = 33,70,-191
set mag_hardware = NONE
set rssi_channel = 12
set serialrx_provider = SUMD
set blackbox_p_ratio = 64
set motor_pwm_protocol = DSHOT600
set failsafe_off_delay = 50
set failsafe_throttle = 1200
set failsafe_procedure = AUTO-LAND
set bat_capacity = 1550
set beeper_dshot_beacon_tone = 4
set pid_process_denom = 1
set osd_warn_esc_fail = OFF
set osd_cap_alarm = 1500
set osd_alt_alarm = 200
set osd_vbat_pos = 2505
set osd_rssi_pos = 2081
set osd_tim_1_pos = 471
set osd_tim_2_pos = 2103
set osd_throttle_pos = 2087
set osd_ah_pos = 2126
set osd_current_pos = 2518
set osd_mah_drawn_pos = 2512
set osd_altitude_pos = 2094
set osd_warnings_pos = 2409
set osd_avg_cell_voltage_pos = 2497
set osd_disarmed_pos = 2346
set osd_stat_max_dist = ON
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_used_mah = OFF
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set debug_mode = NOTCH
set vcd_video_system = PAL

# profile
profile 0

set dterm_lowpass_type = PT1
set dterm_lowpass2_hz = 200
set dterm_notch_hz = 0
set anti_gravity_gain = 3000
set setpoint_relax_ratio = 0
set dterm_setpoint_weight = 50
set throttle_boost = 50
set throttle_boost_cutoff = 40
set p_pitch = 47
set i_pitch = 20
set d_pitch = 36
set p_roll = 45
set i_roll = 23
set d_roll = 33
set p_yaw = 72

# rateprofile
rateprofile 0

set roll_rc_rate = 90
set pitch_rc_rate = 90
set yaw_rc_rate = 90
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 20
set roll_srate = 82
set pitch_srate = 82
set yaw_srate = 82
set tpa_rate = 15
set tpa_breakpoint = 1050

# `

(set rc_interp_ch = RPYT is removed)

flint723, 2018-05-25:
@Lord-Quake Thank you for posting, I believe you should make a separate issue for this ASAP.

Out of curiosity what other builds of 3.4 did you try?

Lord-Quake, 2018-05-25:
@flint723 
betaflight_3.4.0_MATEKF405-855-8f1df0635, betaflight_3.4.0_MATEKF405-849-88904365f and betaflight_3.4.0_MATEKF405-793-02efd4683

Lord-Quake, 2018-06-08:
@flint723 Are the above rev6 settings still valid? Any new settings to be considered for tests? ATM I'm running on build 911.

flint723, 2018-06-08:
There's a PR now for setting new defaults, it's probably slightly worse flying than these listed here.  The main difference is 100/300 gyro cutoffs (instead of 120/300)  Also it does have a different (less) throttle boost setting, and might have iterm_relax on (not totally sure).

If you have any feedback to post there, would be great.  https://github.com/betaflight/betaflight/pull/6036

Lord-Quake, 2018-06-08:
OK, however as it stands at the moment I'm sticking with 120Hz.

mikeller, 2018-06-10:
@flint723: This has been addressed in #6036. Can you please close this?

closed 2018-06-10 by flint723