betaflight/betaflight #5854: GPS symbols are wrong in OSD

McGiverGim, 2018-05-08:
Since https://github.com/betaflight/betaflight/pull/4940 the symbols used to lon and lat in gps coordinates are missing in the OSD.

The problem is that the references in the file are wrong:

https://github.com/betaflight/betaflight/blob/5dc23e662fc9de719bed42781b2447844771d75a/src/main/drivers/max7456_symbols.h#L92-L95

There are no symbol for lat and lon actually in the font. This is the reason why we were using arrows by the moment. This positions contain spaces of the logo at this moment.

What is the best solution for this? Change this constants file to point to other symbol (the arrows) or change the `osd.c` to use directly the arrow symbols (maybe other dev changes it again in the future)?

McGiverGim, 2018-05-08:
By the moment, added a PR changing the symbol in the `osd.c` and added a comment explaining why this symbol is used.

etracer65, 2018-05-08:
Since symbols 0x10 - 0xff are used for the logo, there are a lot of defines in max7456_symbols.h that aren't valid/usable.

Seems like we need a better way to store/draw the logo...

closed 2018-05-09 by mikeller

mikeller, 2018-05-09:
@etracer65: Created #5858 to address this.