betaflight/betaflight #5859: Decoding mavlink messages

phildue, 2018-05-09:
Hi,

I'm using the **Omnibus F4 V3**  and I want to use UART6 for mavlink messages. I activated the telemetry in Betaflight Configurator:
[image]

When i screen the serial port on my laptop, I can see messages coming in. Then I try to decode the messages using [this script](https://gist.github.com/phildue/da2d7d8921675cfa0a8aaaefed355569).

I deactivated wait_for_heartbeat() as it would get stuck forever. The exact same script works fine with another board. Here messages can be decoded correctly. What could be the problem here?

Any help would be appreciated :)

etracer65, 2018-05-10:
On the Omnibus F4 V3 UART6 is meant to be used to SBUS and has a hardware inverter on the RX pin.  So if you're connecting to the SBUS connection then that could be a problem.  UART6 is exposed as uninverted as well.

[image]

Alternately you could use UART3.

phildue, 2018-05-15:
That was indeed the issue. Using uart1 now. Thank you!

closed 2018-05-15 by phildue