betaflight/betaflight #5863: DSHOT600 not fully working for SPRACINGF4NEO 3.3.x and above

Andi252, 2018-05-09:
I tried to update my SPRACINGF4NEO from 3.2 something to 3.3.2. After the update DSHOT600 was only working for motor 3, when I spin motors in the configurator.

I tested different versions 3.2.5 is the last one which works. I digged into the code and I think I found the problem.
 #4844 changed the timer definition for SPRACINGF4NEO to a "macro-style" and this introduced a "porting"-bug.

I believe the SPRACINGF4EVO is very similiar to the F4NEO, therefore I compared both target.c files and I found some difference for motor 1, 2 and 4.

F4NEO:
```
    DEF_TIM(TIM8,  CH3, PC8,  TIM_USE_MOTOR,               0, 0), // D(2,4,7) ESC 1
    DEF_TIM(TIM8,  CH2, PC7,  TIM_USE_MOTOR,               0, 0), // D(2,3,7) ESC 2
    DEF_TIM(TIM8,  CH4, PC9,  TIM_USE_MOTOR,               0, 0), // D(2,7,7) ESC 3
    DEF_TIM(TIM8,  CH1, PC6,  TIM_USE_MOTOR,               0, 0), // D(2,2,7) ESC 4
```

F4EVO:
```
    DEF_TIM(TIM8,  CH1, PC6,  TIM_USE_MOTOR,               0, 1), // ESC 1
    DEF_TIM(TIM8,  CH2, PC7,  TIM_USE_MOTOR,               0, 1), // ESC 2
    DEF_TIM(TIM8,  CH4, PC9,  TIM_USE_MOTOR,               0, 0), // ESC 3
    DEF_TIM(TIM8,  CH3, PC8,  TIM_USE_MOTOR,               0, 1), // ESC 4
```

For motor 1, 2 and 4 the last parameter for the macro is different 1 instead of 0.

I changed the target.c and complied the code locally and DSHOT600 works again for all 4 outputs.

There are two more differences for the last two lines and there I'm not sure which one is correct.

mikeller, 2018-05-09:
One possible cause for this could be an invalid configuration. Can you please try to re-flash the official firmware, and then do a `defaults` in CLI, to see if that fixes it?

jflyper, 2018-05-09:
My theory is as follow.

- SPRacing F4 NEO has a nice pins for first four suitable for burst Dshot.
- However, SPRACINGF4NEO target could not enjoy the benefit as it was not using the TIM_DEF macro that is required for insertion of TIM_UP stream def.
- The burst mode was enabled by default for all targets to detect targets that has problems with the mode.
- I converted SPRACINGF4NEO to TIM_DEF style, but did not enable  `USE_DSHOT_DMAR`.
- Sometime later, the default enable was removed, and SPRACINGF4NEO started to fail due to DMA stream conflicts.

jflyper, 2018-05-09:
@Andi252 

Can you check if `set dshot_burst = ON` fixes your problem?

Andi252, 2018-05-10:
@mikeller I tried the original version 3.3.2 with full chip erase and command default
```
# diff
# version
# Betaflight / SPRACINGF4NEO (SP4N) 3.3.2 May  4 2018 / 10:06:55 (64a292f37) MSP API: 1.38
# name
# resources
# mixer
# servo
# servo mix
# feature
feature -LED_STRIP
feature -TRANSPONDER
# beeper
# map
# serial
# led
# color
# mode_color
# aux
# adjrange
# rxrange
# vtx
# rxfail
# master
set motor_pwm_protocol = DSHOT600
# profile
profile 0
# rateprofile
rateprofile 0
```
still only motor 3 works.

@jflyper 

with 
`set dshort_burst = on`

all motors work

thanks for the tip

mikeller, 2018-05-10:
Thanks for testing @Andi252. With #5865 `dshot_burst` will default to 'on' from 3.3.3 on.

DieHertz, 2018-05-10:
For burst dshot these values aren't used

jflyper, 2018-05-10:
@Andi252 I understand your concern.

@mikeller May be technically correct to amend #5865 with correct dmaopt values so that DMA streams don't collide even without the `dshot_burst` turned on?

Andi252, 2018-05-10:
@mikeller if you add the modification to your pull request I can test both cases.
Are there automatically created HEX files for each pull request from the CI server, or do I have to compile to code locally?

And one "stupid" github beginner question: Who will close the issue? The pull request from mikeller or do I have to close the issue or do you have to close the issue?

DieHertz, 2018-05-10:
Pull Request can close an issue if it has a pattern `Fixes/Closes #IssueNo` somewhere in the name, description or commit messages

mikeller, 2018-05-10:
@jflyper: Done.

@Andi252: Please test with `set dshort_burst = off`.

Andi252, 2018-05-11:
@mikeller I tested your pull request, both opions work for DSHOT600 and 4 motors.

```
# diff
# version
# Betaflight / SPRACINGF4NEO (SP4N) 3.4.0 May 11 2018 / 15:24:20 (1877a6f) MSP API: 1.39
# name
# resources
# mixer
# servo
# servo mix
# feature
# beeper
# map
# serial
# led
# color
# mode_color
# aux
# adjrange
# rxrange
# vtx
# rxfail
# master
set dshot_burst = OFF
set motor_pwm_protocol = DSHOT600
# profile
profile 0
# rateprofile
rateprofile 0
```

PR #5865 (2018-05-11): Enabled Dshot DMAR for SPRACINGF4NEO.
Fixes #5863.
closed 2018-05-11 by jflyper