betaflight/betaflight #5864: Occurrences of target names outside the target directories

jflyper, 2018-05-10:
```
src/main/drivers/system_stm32f10x.c:77:#ifdef CC3D
src/main/drivers/timer.c:777:#ifdef CC3D
src/main/fc/fc_init.c:789:#ifdef CJMCU
src/main/io/gps.c:284:#if !defined(COLIBRI_RACE) || !defined(LUX_RACE)
src/main/io/gps.c:303:#if defined(COLIBRI_RACE) || defined(LUX_RACE)
src/main/io/gps.c:308:#if defined(COLIBRI_RACE) || defined(LUX_RACE)
src/main/io/gps.c:327:#if defined(COLIBRI_RACE) || defined(LUX_RACE)
src/main/io/gps.c:343:#if defined(COLIBRI_RACE) || defined(LUX_RACE)
src/main/io/gps.c:284:#if !defined(COLIBRI_RACE) || !defined(LUX_RACE)
src/main/io/gps.c:303:#if defined(COLIBRI_RACE) || defined(LUX_RACE)
src/main/io/gps.c:308:#if defined(COLIBRI_RACE) || defined(LUX_RACE)
src/main/io/gps.c:327:#if defined(COLIBRI_RACE) || defined(LUX_RACE)
src/main/io/gps.c:343:#if defined(COLIBRI_RACE) || defined(LUX_RACE)
src/main/io/gps.c:284:#if !defined(COLIBRI_RACE) || !defined(LUX_RACE)
src/main/io/gps.c:303:#if defined(COLIBRI_RACE) || defined(LUX_RACE)
src/main/io/gps.c:308:#if defined(COLIBRI_RACE) || defined(LUX_RACE)
src/main/io/gps.c:327:#if defined(COLIBRI_RACE) || defined(LUX_RACE)
src/main/io/gps.c:343:#if defined(COLIBRI_RACE) || defined(LUX_RACE)
src/main/io/vtx_smartaudio.h:107:#ifdef OMNIBUSF4
src/main/io/vtx_smartaudio.h:107:#ifdef OMNIBUSF4
src/main/drivers/adc.h:94:#ifndef SITL
src/main/fc/fc_init.c:441:#if defined(USE_INVERTER) && !defined(SITL)
src/main/fc/fc_rc.c:241:#if defined(SITL)
src/main/fc/fc_rc.c:246:#if defined(SITL)
src/main/io/serial.c:44:#ifdef SITL
src/main/io/serial.c:393:#ifdef SITL
src/main/io/serial.c:462:#ifndef SITL
src/main/osd_slave/osd_slave_init.c:188:#if defined(USE_UART) && !defined(SITL)
src/main/sensors/compass.c:118:#if !defined(SITL)
```

mikeller, 2018-05-10:
Good grepping, @jflyper! Do you want to come up with a pull request to get rid of these, or shall I?

jflyper, 2018-05-11:
@mikeller Can you do it? I'm working on gyro thing atm.

mikeller, 2018-05-11:
Yeah, I am happy to do it. I just wanted to make sure we are not **both** doing it.

brucesdad13, 2018-05-13:
Keeping it clean 😁

mikeller, 2018-05-13:
More like 'keeping nasty surprises away when we're moving to universal targets'. :stuck_out_tongue_closed_eyes:

mikeller, 2018-05-25:
Fixed by #5877.

closed 2018-05-25 by mikeller