betaflight/betaflight #5867: ESC/MOTORS start beeping on rc lost BF3.3X BLH16.7 BLH32

RotorVulura, 2018-05-10:
HI there ! 
That's not pretty good just because, if you leave your quad up a tree and you need to much time to get a ladder or other exotic things to bring it back to the ground, you will find your expensive battery drained out and\or your motors/escs boiled.
This is not about beacon time (already set to infinite). All you need is just turn off the radio.
Thx for your answer.
First build kakute AIO v1 - racerstar 30A V2
Other build kaute AIO v2 - tekko32 35 A
DIFF 
# version
# Betaflight / KAKUTEF4 (KTV1) 3.3.1 Apr  3 2018 / 09:43:09 (4b7eaf79d) MSP API: 1.37

# name
name GURZO

# resources

# mixer

# servo

# servo mix

# feature
feature -TELEMETRY
feature MOTOR_STOP
feature DYNAMIC_FILTER

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -MULTI_BEEPS
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE

# map

# serial
serial 5 8192 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 900 1200 0
aux 1 13 1 1800 2100 0
aux 2 28 3 1300 1700 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_lowpass_hz = 0
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set gyro_stage2_lowpass_hz = 90
set rssi_channel = 7
set dshot_idle_value = 350
set motor_pwm_protocol = DSHOT600
set current_meter = ADC
set vbat_scale = 112
set beeper_dshot_beacon_tone = 5
set small_angle = 180
set deadband = 2
set yaw_deadband = 2
set pid_process_denom = 1

Just OSD and pid deleted

ROTOR VULURA !

theSkonk, 2018-05-11:
you have the beeper configured to RX_LOST turn that off and it will stop

RotorVulura, 2018-05-11:
@theSkonk
beeper -RX_LOST
beeper -RX_LOST_LANDING
Minus (-) tell me that it is disabled, isn’t it ? 
This should disable the piezo beeper and enable the esc beeping ? Maybe ?
Thx in advance :)

closed 2018-06-17 by stale[bot]