betaflight/betaflight #5869: Omnibus F4 Nano LED flicker

TIMSCHMITZ, 2018-05-10:
Problem:
https://youtu.be/M2-diNJFl4I
Expected solid color

version
# Betaflight / OMNIBUSF4SD (OBSD) 3.3.1 Apr  3 2018 / 09:55:08 (4b7eaf79d) MSP API: 1.37

# name
name Spatz

# resources
resource SERIAL_TX 1 NONE
resource SERIAL_TX 11 A09

# mixer

# servo

# servo mix

# feature
feature SOFTSERIAL
feature TELEMETRY
feature LED_STRIP

# beeper

# map

# serial
serial 0 64 115200 57600 0 115200
serial 2 1024 115200 57600 0 115200
serial 5 2048 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# led
led 0 7,15::C:2
led 1 8,15::C:2

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0
aux 1 1 1 900 1300 0
aux 2 2 1 1300 1700 0
aux 3 3 2 1700 2100 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set align_gyro = CW180FLIP
set align_acc = CW180FLIP
set fpv_mix_degrees = 35
set serialrx_provider = SBUS
set min_command = 950
set dshot_idle_value = 475
set motor_pwm_protocol = PROSHOT1000
set osd_vbat_pos = 2049
set osd_warnings_pos = 2506
set osd_disarmed_pos = 2282
set vtx_band = 3
set vtx_channel = 5
set vtx_freq = 5885

# profile
profile 1

# rateprofile
rateprofile 0

Arakon, 2018-05-10:
That's either because you have 2 warnings fighting with each other, or because you didn't put a diode on the 5V line to the LEDs.

TIMSCHMITZ, 2018-05-10:
How to connect the diode?

etracer65, 2018-05-11:
The diode goes on the 5V positive power connection to the LED.  It has the effect of dropping the voltage by about 0.6V which the puts the addressable LED's better in range.

Arakon, 2018-05-11:
And the marked side of the diode (usually a black stripe) faces the LEDs.

closed 2018-06-17 by stale[bot]