betaflight/betaflight #5872: Upload .Hex file with Xloader

Nightcr4wler, 2018-05-11:
Hi, I'm new here and I want to know if is possible upgrade the betaflight 3.2.5 .hex with usb using the xloader software?

mikeller, 2018-05-11:
Erm, the only 'xloader' that I found seems to be targeting arduino boards. Can you provide a link to the 'xloader' that you want to use for STM32?

jflyper, 2018-05-12:
@Nightcr4wler 
I recommend you use the firmware flasher that is a part of the BetaFlight Configurator (BFC) software, as you will use the BFC anyway to perform post-flash configuration process.

Nightcr4wler, 2018-05-12:
@mikelle I was thinking this should work for the STM32 :(

@jflyper I need a uploader to put the CC3D F1 without a converter with a BF 3.2.5 because don't have enough space to update .hex file from BF Configurator

mikeller, 2018-05-12:
@Nightcr4wler: Ah, what you want to use in this case is ST-Link, as described here: https://librepilot.atlassian.net/wiki/spaces/LPDOC/pages/40402972/Recover+board+using+ST-Link

Nightcr4wler, 2018-05-13:
@mikeller Wow, I'll certainly use this method, very useful! Thanks for your help!

closed 2018-06-19 by stale[bot]