betaflight/betaflight #5881: Servo PWM doesn't go lower than PWM_PULSE_MIN

mankeli, 2018-05-12:
I would like to control a LED from my controller, but it seems that the servo output mapping is limited by PWM_PULSE_MIN and PWM_PULSE_MAX.
Looking at the code, it seems that after mapping the value goes to timer CCR reg. At a quick glance (and without too much prior RC experience) it seems to me that this kind of clamping is pretty much unnecessary. 

Can somebody enlighten me if there's any reason why it shouldn't be removed?

mikeller, 2018-05-13:
What you are looking at are **RC** PWM values. These are defined to have a range from 1000 to 2000 us.  There is no point in extending this and making it unsafe to use with any PWM controlled RC devices.
There is a driver for power control PWM values that is used for brushed motors. This can be used to drive an LED as well.

mankeli, 2018-05-13:
Okay, sounds reasonable. @mikeller Can you point me into right direction on configuring PWM pin as brushed motor output that can be directly controlled from RX aux channel? My actual motors use dshot which DMA i believe. I can't seem to find an option to mark just one motor output as "brushed".

After I get this working, I can write a small tutorial to the wiki.

jflyper, 2018-05-13:
You have add a code to switch passed through output to power PWM control mode. In addition, I think we can not expect some nice linear response as brightness by applying straight PWM to LED; it's probably lots more complicated.

mankeli, 2018-05-13:
okay. :/ In this case even linearly controlled intensity would have been enough though. But since I have to add code to do this myself I can add this. (not too big task to convert perceptual intensity to linear) And maybe I can also output PWM on higher rate, as servo output looks a bit too flickery on the led.

closed 2018-05-13 by mankeli