betaflight/betaflight #5882: Brushed quad hard 180 faceplant

Skynectar, 2018-05-13:
With brushed models, when flying too fast if the pilot does a hard 180 the quadcopter will faceplant straight into the ground. This problem has been found on multiple FC, as well as multiple Betaflight firmware versions tested from 3.1-3.2 on these boards. 

Fly the brushed quadcopter straight forward, and quickly add some yaw and a bit of roll to do a quick 180. If the pilot is flying too fast and 180 turns too quickly, the brushed quadcopter will "glitch" forward into the ground. 

 This has been observed on the f3 evo brushed FC, multiflite pico FC, betafpv w/ OSD dsmx rx FC, all using various firmware versions. 

I don't have a specific diff to post until I get to my computer in a few hours. I will post the dump/diff for the latest board with the problem (betafpv f3 osd w/ dsmx rx) in a few hours.

Skynectar, 2018-05-13:
diff

CLI

# diff
# Betaflight / BETAFLIGHTF3 (BFF3) 3.2.4 Jan 10 2018 / 22:39:42 (90d32fd56) MSP API: 1.36
name QL720
resource MOTOR 1 B09
resource MOTOR 2 B08
resource MOTOR 3 A07
resource MOTOR 4 A06

feature -TELEMETRY
feature AIRMODE
map TAER1234
aux 0 0 0 1500 2100
aux 1 13 1 1800 2100
set acc_hardware = NONE
set blackbox_device = NONE
set motor_pwm_protocol = BRUSHED
set align_board_yaw = 180
set vbat_scale = 115
set deadband = 20
set yaw_deadband = 20
set osd_rssi_pos = 40
set osd_tim_1_pos = 2103
set osd_flymode_pos = 333
set osd_throttle_pos = 2241
set osd_vtx_channel_pos = 377
set osd_ah_sbar = 200
set osd_ah_pos = 200
set osd_mah_drawn_pos = 353
set osd_craft_name_pos = 2456
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 329
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_tim_2 = OFF
set vcd_video_system = 2
profile 0

set setpoint_relax_ratio = 30
set dterm_setpoint_weight = 75
set p_pitch = 75
set d_pitch = 105
set p_roll = 75
set i_roll = 50
set d_roll = 105
set p_yaw = 75
[... 558 lines cut ...]
set p_vel = 55
set i_vel = 55
set d_vel = 75
set level_limit = 55
set horizon_tilt_effect = 75
set horizon_tilt_expert_mode = OFF
rateprofile 0

set rc_rate = 135
set rc_rate_yaw = 135
set rc_expo = 0
set rc_expo_yaw = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 71
set pitch_srate = 71
set yaw_srate = 71
set tpa_rate = 10
set tpa_breakpoint = 1650

# 

_________________

Thanks to anyone who helps out, has been an issue for a few months now and has been really hindering performance the brushed models.

chime13, 2018-05-13:
Have you tried flying with AIRMODE enabled? Seems BF introduced this bug on 3.1.7 that requires AIRMODE to be on or you'll experience some weird behavior like you described. Happens to me all the time when AIRMODE is not enabled.
Put AM on an AUX so you can toggle it off when landing, so the craft doesn't go bouncy bounce on the ground.

etracer65, 2018-05-13:
I think you're simply expecting too much from brushed motors and tiny props.  In the scenario you describe the quad is going to experience substantial prop wash caused by falling back through its turbulent air.  Brushed motors generally can't accelerate/decelerate quickly enough to respond to dynamic conditions quickly enough.

chime13, 2018-05-13:
I disagree - there is a totally reproducible bug introduced in 3.1.7 that makes brushed builds exhibit the issue described. The solution at that time was to leave AM on.
It doesn't happen on 3.1.6 and it definitely never happens in Silverware "for micro brushed".

Skynectar, 2018-05-13:
Thanks to both of you for the replies. Ever since I found out about AM I've been using it on all my brushed models. I don't know if this matters or not, but I have not checked AM in the Configuration tab, I do have it enabled in the Modes Tab all the time for all my models, that's usually just how I set it up. I will try to enable it through the configuration tab to see if that helps and report back the results.

I agree, it is an extremely harsh testing condition for a brushed quad to go through, however i truly feel as Chime mentioned it is a result of the gyro freaking out for a second and not from a lack of stable air. It almost "locks up" when it face plants into the ground. 

I am just trying to get some more love on betaflight for the brushed models if possible. They can be awesome, fun and stress free flying. Hope to see more development into the brushed category.

Skynectar, 2018-05-13:
That makes sense. It may be hard to put a blackbox logger on a micro brushed model, and idt any brushed FCs come with those, so i understand this issue may not be very easy to solve but I figured it was worth a shot bringing it up.

chime13, 2018-05-13:
If possible, try flashing 3.1.5 or 3.1.6 to see if the issue goes away without having to engage AM. I never used AM in the beginning and got used to flying that way until 3.1.7. From that point on, I've had to resort to using AM. It's alright except for the bouncy bounce when landing.
Unfortunately, I believe BF is mostly focussing on larger, brushless builds which do exhibit different issues than the micro brushed type.
IMVHO - Without trying to offend anyone or their efforts here, you have to try Silverware in order to truly understand un-bloated, phenomenal performance out of a brushed micro quad.

On another note: I noticed "set motor_pwm_rate = 480". For brushed, should be between 8000 and 32000.

Skynectar, 2018-05-13:
I sure will try that chime! 3.1.5 and 3.1.6 will be tested without airmode, and I will report back with my findings, and and thanks, idkh it got set to 480.

chime13, 2018-05-13:
@Skynectar,
Here's my setup for most of my brushed builds.
I am running 3.4 though on F3 boards, SpracingF3EVO and OmnibusF3
On my F1, 3.2.5 is the highest you can go.
[Lemon.txt](https://github.com/betaflight/betaflight/files/1998680/Lemon.txt)

Skynectar, 2018-05-14:
Hey brucesdad13 , that would be really neat if you could try to post a bblog of this issue for further investigations!!!

The most recent brushed quad I've built that has suffered from this issue is a NotFastEnuf Boss Edition 7XL (720 motors) at 17500kv, using afunta 55mm props. (380mah battery), weighs 28.2 without battery, 38.5 with battery. (Best brushed model I've found ever, imho ) 

If you are unable to replicate those exact flight conditions though, I'm sure you can still replicate the issue using any brushed model of your choice, it seems to happen to all I've flown basically... 

It's been raining, and I've been unable to test bringing it back to an older firmware 3.1.5 and 3.1.6, but as soon as it clears up I'll be reporting my findings.

Skynectar, 2018-05-16:
Just built a 6mm Boss frame at 18.2g without battery, 24g with battery on betaflight 3.2.4,  using a betafpv F3 FC and the same thing is happening. Going to flash both quads to 3.1.5 and see if it fixes things.

closed 2018-06-22 by stale[bot]

Skynectar, 2019-01-15:
Sorry for bringing this issue up again, but i found the problem to the issue so i figured it would probably be very helpful to post and share....

Those 180 faceplant glitches when doing hard 180s are caused by running "props in" on brushed... Betaflight does not get along with brushed models running standard prop orientation (props in).... it'll fly that way, but with many issues creating a terrible flight experience.

for about a year i was always just thinking Betaflight firmware was just broken when it came to brushed, BUT

running reversed props (props out) literally ELIMINATES this issue and flies 4 times better on Betaflight this way...

I wish i was "in the know" sooner on this issue... Maybe if i could make a recommendation for Betaflight, post this in the "helpful tips" section so newer brushed users can be "in the know" about this issue and will spend more time flying props out and less down time trying to figure out whats causing the poor flight performance.

mikeller, 2019-01-16:
That is an interesting finding, @Skynectar.