betaflight/betaflight #5884: BLHELI32 DSHOT BEEPING when RX LOST.  No option to disable while keeping AUX switch Beeping enabled.

attomic, 2018-05-13:
It appears that while the dshot beeping can be set to a an AUX switch after enabling the DSHOT Beacon in configuration,  it also beeps when RX is lost and there is no option to disable this.

BF 3.3.2
DYS F4 and PYRO F4 OSD
AIkon AK32 4in1

attomic, 2018-05-13:
Im not quite sure how to read your response but here is the diff:

# feature
feature TELEMETRY
feature AIRMODE
feature ESC_SENSOR
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE

# map

# serial
serial 0 64 115200 57600 0 115200
serial 2 1 115200 57600 0 115200
serial 4 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 2 900 1150 0
aux 1 13 1 1875 2100 0
aux 2 35 3 1850 2100 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 0
set gyro_notch2_hz = 0
set gyro_notch2_cutoff = 0
set moron_threshold = 128
set acc_hardware = NONE
set min_check = 1020
set rssi_channel = 16
set rssi_scale = 30
set rc_interp = MANUAL
set rc_interp_ch = RPYT
set rc_interp_int = 14
set serialrx_provider = SBUS
set adc_device = 2
set min_throttle = 1050
set dshot_idle_value = 250
set dshot_burst = ON
set use_unsynced_pwm = ON
set motor_pwm_protocol = DSHOT1200
set motor_pwm_rate = 32000
set failsafe_delay = 10
set vbat_max_cell_voltage = 45
set current_meter = VIRTUAL
set beeper_inversion = OFF
set beeper_od = ON
set deadband = 5
set yaw_deadband = 5
set pid_process_denom = 1
set ibus_sensor = 0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
set osd_units = IMPERIAL
set osd_warnings = 1
set osd_rssi_alarm = 60
set osd_cap_alarm = 234
set osd_alt_alarm = 2200
set osd_tim1 = 257
set osd_tim2 = 2560
set osd_vbat_pos = 2497
set osd_rssi_pos = 2106
set osd_tim_1_pos = 2518
set osd_tim_2_pos = 33
set osd_flymode_pos = 333
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 2081
set osd_ah_pos = 200
set osd_current_pos = 385
set osd_mah_drawn_pos = 353
set osd_craft_name_pos = 2505
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 329
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 2283
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_min_rssi = OFF
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_tim_2 = OFF
set debug_mode = FFT
set vtx_channel = 2
set vtx_power = 2
set vtx_freq = 5760
set vcd_video_system = 1
set camera_control_key_delay = 150

# profile
profile 0

set dterm_lowpass_type = PT1
set dterm_lowpass = 80
set dterm_notch_hz = 0
set vbat_pid_gain = ON
set dterm_setpoint_weight = 60
set crash_gthreshold = 200
set p_pitch = 40
set i_pitch = 65
set d_pitch = 18
set p_roll = 35
set i_roll = 65
set d_roll = 18
set p_yaw = 60
set i_yaw = 55
set d_yaw = 10

# rateprofile
rateprofile 0

set thr_mid = 35
set thr_expo = 10
set roll_srate = 73
set pitch_srate = 73
set tpa_rate = 0

jflyper, 2018-05-13:
Can I see the screen shot of the receiver tab when it is beeping on RX_LOSS?

mikeller, 2018-05-14:
RX_LOSS is hard wired to use the Dshot beacon. Disable Dshot beacon if you do not want this.

RotorVulura, 2018-05-14:
not good :(
This will result in esc/motor/battery ready for trash if you can’t rescue your quad in time. May I ask why it is hard wired to esc beacon now ? In BF 3.2.X, rx_lost come up on the “real” beeper and only if you want it.

mikeller, 2018-05-14:
The idea was to give users the best possible chance to find their craft in case of RX loss by activating the Dshot beacon. If you prefer to lose your craft instead of risking damage to your motors / ESCs, you can simply turn off the Dshot beacon.

attomic, 2018-05-14:
I would not lose my craft because I would have it on a switch as it was before.  The fact is that it was hard coded without much thought to leaving it as an option.  If you use a physical buzzer then you have the option do you not?

mikeller, 2018-05-14:
How are you going to use your switch if you have RX loss?

attomic, 2018-05-14:
You still have the option in BLHELI32 based on time.

mikeller, 2018-05-14:
So what you are saying is that enabling the timeout based option (that will even kick in if your craft sits around fully functional and ready) won't risk burning your ESCs / motors, but the one that only kicks in when you have really lost your craft will? This does not make any sense.

attomic, 2018-05-14:
OK does it make sense that since it is coded into betaflight now you have negated an option in belheli32?

Also you are acknowledging the danger of ruining hardware depending on the scenario so there should be an option in betaflight just as there is in blheli based on the flyers judgement.

mikeller, 2018-05-14:
Huh?

If you had read my comments above, you would have learned that you already can disable the Dshot beacon if you are so worried about it burning up your hardware. Please stop making nonsensical statements.

attomic, 2018-05-14:
If you disable the dshot beacon does it also disable the ability to use an AUX switch?

mikeller, 2018-05-14:
Yes, of course. Then you don't have to worry any more about burning up your hardware - why would you want to use this functionality if you are so worried about it burning up your hardware?

RotorVulura, 2018-05-14:
I would like to be able to make my own choice to activate the beacon using an AUX switch and/or triggered by rx_lost.

mikeller, 2018-05-14:
Why would you do that - if it is as you say and it will burn up your hardware when it is activated in an emergency, why would you ever want to activate the beacon in a non-emergency?

Also, there is no beaper - only a beacon.

wind0r, 2018-05-14:
When stuck in a tree and getting a ladder from home. I dont want my quad to annoying beep ;)

beeper -RX_LOST should disable it. i dont see any reason why that command should have some special case where it still is enabled

RotorVulura, 2018-05-14:
@wind0r 
beeper -RX_LOST regard the buzzer, esc beacon on rx_lost remain active :(

wind0r, 2018-05-14:
@RotorVulura yes, thats the problem here and i also dont understand why rx_lost with dshot doesnt care about those setting. i also dont see a valid reason to ignore those settings

RotorVulura, 2018-05-14:
@mikeller This rx-lost esc beacon drive me crazy ... you can configure 36 led but you can't control your esc beeping. Maybe better to open a formally feature request ?

mikeller, 2018-05-14:
@wind0r: The reason for that is that some users complained because they wanted to be able to disable the beeper on AUX, but not the beacon. I really don't understand why users to want to have the Dshot beacon for non-emergencies, but not in emergencies, when a switch won't help them activate it because it's lost without contact. You will never go and get a ladder when your craft is on the roof upside down, and has lost contact for that reason - you will go and spend money on a new one because  tghe first one is lost and a write-off...

@RotorVulura: #5891 gives you the option to make yourself unhappy.

attomic, 2018-05-14:
@mikeller We all knew where it was approximately so going near to it, beeping on the switch to find it, would be normally do.  Instead it sat there beeping to ruin.

wind0r, 2018-05-14:
@attomic @RotorVulura 
there also is https://github.com/betaflight/betaflight/pull/5891
both prs are for disscusion and another solution would be having a setting/command similar to the beeper one.

personal I would prefer just handling the beacon similar to the beeper as done with my pr since i dont think user care or understand the difference between beacon or beeper. Also because most people call it dshot beeper and since it needs to be could one per beep i think it is a beeper.

if we find some use case where different beeper and beacon settings i would prefer the beacon cli option

mikeller, 2018-05-14:
@attomic: If the rx lost beacon was going then rx was lost. Saying now 'going to where we thought it should be would have restored connectivity' is purely speculative.

RotorVulura, 2018-05-14:
@wind0r Are you talking about the configurator's GUI ? I think that to leave rx_lost esc beacon enabled by default while a given CLI command to disable it, will be enought as I can see just 2 people (I'm one of this two ahah !) that starts hating this new hardcoded feature . . 
@mikeller I have understood your intention. I just don't like hard coded things, please don't hate me :)

attomic, 2018-05-14:
@mikeller We had the option to have it on switch and now that option is being degraded by what you think is a "feature".  Just keep the options open as you add "features" and everyone's needs are addressed.

wind0r, 2018-05-14:
@RotorVulura i think most people didnt noticed this behaivor. i also didnt noticed it. looks like i never was stuck in a tree after starting using dshot. i prefer having not having a single sound on rx_lost but want to use the aux beeper

i think having rx_lost disabled also should disabled it for dshot. handling dshot and normal beeper different is strange and confusing to me.

attomic, 2018-05-14:
> i think having rx_lost disabled also should disabled it for dshot. handling dshot and normal beeper different is strange and confusing to me.

Yes another excellent point is that you want consistency.

mikeller, 2018-05-14:
@attomic: You are wrong, the Dshot beacon for rx lost was never on a switch since it got added. There is no 'degradation', just another feature that got added.
If you and others are dead set that it has to be possible to disable this we can do this.
I am just trying to understand why you seem to be preferring the risk of permanently losing a craft to the risk of damaging your ESCs / motors. I guess the bias that individual bad experiences create is strong for these things...

Bahecuk, 2018-05-14:
I use dshot beeper as a replacement for active beeper and would like it to behave the same way normal beeper does.
My beacon volume is set to max (255?) and the only time when I want to hear a dshot beeper is when I get to potential area of the crash, stand still, listen carefully and flick the switch on the radio. No exceptions.
Why? Three reasons:
1) Continuous motor beeping is not good for 2204+ size motors and straight up deadly for smaller motors.
2) Dshot beeper is only audible within close range, much closer than any modern Tx/Rx combo can reach  and reconnect enabling you to use aux activated beeper. 
3) I mostly fly in parks relatively close to people and I want my quad to sit still and silent until I get close to it. Beeper attracts children like a magic whistle, I don't need to explain why kids and race quads don't mix well. 
Only situation I can think of when rx_lost beeper can help you find a model is if your Rx stopped working during flight or got disconnected, which is extremely unlikely and still can be solved by timer-based BlHeli beacon.

RotorVulura, 2018-05-14:
looking at [#5891](https://github.com/betaflight/betaflight/pull/5891)
Thank you again for your hard (coded ahah!) work. I think that I (and other fews)  just need a CLI option to disable the ESC Beacon on RX_LOST. So you can do that in a secret way (without trace in release note) and PM me the new command. I'm joking of course, but I think that a CLI command act to disable the ESC BEACON on RX_LOST, will be a good (undocumented) new feature :)
Thx again for your time, and excuse this rx_flame

attomic, 2018-05-14:
@mikeller Thanks for considering the issue I appreciate it.  Thanks to everyone who chimed in.

brycedjohnson, 2018-05-16:
@Bahecuk :smile: I wouldn't put your beeper at max - it doesn't actually get louder, but just gets hotter.  I guess you could just leave the beeper on and use see if you can smell the enamel if you can't hear it...

I have no idea how people fry their motors besides doing stuff like setting the beeper to max... I have been really abusive on a wide variety of motors, big and small and haven't manage to get them to even get very hot, but they still are plenty loud.

Has anyone actually burned up a motor with the beeper besides setting the beacon to be > 80? All I have heard is that it happened to a friend...

bkeinert, 2018-05-24:
I can follow both lines of argumentation (@mikeller, @wind0r) but I'm with @mikeller on this one.

Let me throw in another point of view, from someone who flies mostly overpowered machines <150g (with lipo): I don't have the space/resources for a real buzzer/beeper and heavily rely on DSHOT_BEACON, if I loose a machine in >4cm grass - it's just gone if there was no DSHOT beacon. I prefer burning my motors (which does not really happen anymore with adequate settings and recent blheli_s). Maybe you should consider whether it's a good idea to run DSHOT_BEACON at all on a larger machine. Put a <$2 beeper and disable DSHOT beacon - you have those 0.8g.

On the other hand, the @mikeller variant also gives some benefit for the super paranoid: A beeper wire can break, this is unlikely to happen for all ESCs/Motors. So you at least have the RX_SET beacon available in case you crash and somehow manage to loose your main beeper (this can happen and happened to me) which still would have handled RX_LOSS situations. You still would be screwed with RX_LOSS nevertheless in that case.

RotorVulura, 2018-05-24:
@bkeinert The Line/Point is to be able to do the right choice you need (I think). Piezo buzzer , esc buzzer or total stealth ? ESC beeping on rx_lost is the best default settings. I just don’t like that was hardcoded :)

mikeller, 2018-06-02:
Implemented in #5891.

mikeller, 2018-07-07:
@attomic: Can you please confirm that this has been implemented, and if so, close this?