betaflight/betaflight #5885: Altitude does not show in OSD

IcyLemur276, 2018-05-13:
Running the latest BTFL 3.3.2 Target MLTE MPF4 using the MICRO UBLOX M8N GPS GLONASS W/ (18X18MM) powered off 5v off my MotoLab Tempest F4 Flight Controller FC.  I have the GPS module mounted to the top of my battery via industrial strength velco.  GPS works perfectly, but not getting Altitude in my Goggles. GPS works fine in the Betaflight Configurator with altitude. Any suggestions?

# version
# Betaflight / MLTEMPF4 (MLTE) 3.3.2 May  4 2018 / 09:47:38 (64a292f37) MSP API: 1.38

# name

# resources

# mixer

# servo

# servo mix

# feature
feature GPS
feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -DISARMING
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -ARMED
beeper -BLACKBOX_ERASE

# map

# serial
serial 0 64 115200 57600 0 115200
serial 1 2 115200 57600 0 115200
serial 4 0 115200 57600 0 115200
serial 5 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1800 2100 0
aux 1 1 0 1400 2100 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set rssi_channel = 8
set serialrx_provider = CRSF
set motor_pwm_protocol = DSHOT600
set beeper_dshot_beacon_tone = 3
set small_angle = 15
set gps_provider = UBLOX
set gps_auto_baud = ON
set osd_units = IMPERIAL
set osd_warnings = 63
set osd_tim1 = 1025
set osd_vbat_pos = 2080
set osd_rssi_pos = 2458
set osd_tim_1_pos = 2432
set osd_current_pos = 2048
set osd_mah_drawn_pos = 2400
set osd_gps_speed_pos = 2095
set osd_gps_sats_pos = 2137
set osd_home_dir_pos = 2094
set osd_home_dist_pos = 2073
set osd_altitude_pos = 2102
set osd_disarmed_pos = 2250
set osd_stat_max_dist = ON
set osd_stat_max_alt = ON
set osd_stat_endbatt = ON
set osd_stat_bb_no = OFF
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set vtx_band = 3
set vtx_channel = 4
set vtx_power = 3
set vtx_freq = 5645
set vcd_video_system = 2

# profile
profile 0

set anti_gravity_gain = 3000
set dterm_setpoint_weight = 100
set d_level = 100
set level_limit = 70

# rateprofile
rateprofile 0

set roll_rc_rate = 105
set pitch_rc_rate = 105
set yaw_rc_rate = 105
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 30
set roll_srate = 79
set pitch_srate = 79
set yaw_srate = 79

#

mikeller, 2018-05-25:
@IcyLemur276: Have you been able to verify that this is fixed, and can this be closed?

IcyLemur276, 2018-05-25:
@mikeller This has not been fixed as of this morning running 3.3.2.  I'm getting altitude readings in my Betaflight configurator GPS tab but only reads 0 in my OSD.

mikeller, 2018-05-26:
@IcyLemur276: As I stated above, the fix for it is in 3.4, not in 3.3.2, so testing with 3.3.2 is just a waste of time.

IcyLemur276, 2018-05-26:
@mikeller I don’t see 3.4 available for download.  I am anxiously waiting so that I can confirm if the issue is resolved.

mikeller, 2018-05-26:
@IcyLemur276: You can get a preview from https://ci.betaflight.tech/job/Betaflight/lastStableBuild/, or else release candidates for 3.4 should be available in the configurator in about a week.

IcyLemur276, 2018-05-26:
@mikeller I will wait for the update to show in the configurator and report back when I have it tested. BTW..Where in the link you provided is the 3.4 download?  For me it would be the MLTE target.

mikeller, 2018-05-26:
@IcyLemur276: Sorry, that link was a bit vague. The list of hex files is here: https://ci.betaflight.tech/job/Betaflight/lastStableBuild/artifact/obj/

IcyLemur276, 2018-05-26:
@mikeller I just tried 3.4 on target MLTE and the altitude is still not reading in the OSD, but reads fine in the configuration GPS tab.  OSD just reads 0.

mikeller, 2018-05-26:
@IcyLemur276: That's weird. Both places get the information from the same source. One thing that has changed in 3.4 though is that the altitude (except for the one displayed in the GPS tab in configurator) is tracked as relative altitude from the take-off point, so the altitude displayed is reset to 0 every time the craft is armed.

mikeller, 2018-06-02:
@IcyLemur276: Have you checked if your OSD altitude display actually updates while you are flying?

IcyLemur276, 2018-06-02:
@mikeller yes the OSD altitude display does not update. Only see it updating in the configurator.

mikeller, 2018-06-03:
Unfortunately, I don't have a GPS setup. What I can confirm is that that for a setup with BARO and no GPS, the altitude display in OSD works as expected.

@s0up, @dbasch, @McGiverGim: I think you guys got GPS setups. Can one of you look into this please?

mikeller, 2018-06-03:
@dbasch: @IcyLemur276 reported checked this problem against 3.4 in https://github.com/betaflight/betaflight/issues/5885#issuecomment-392230903. So it looks like there still is a problem.

McGiverGim, 2018-06-03:
Mine works right, but I have GPS+Baro. I will try to deactivate the Baro to test if it works only with GPS.

Smeat, 2018-06-03:
I have the same problem using GPS only.
When changing the task condition to use ``FEATURE_GPS`` and not ``SENSOR_GPS`` everything works just fine:
```c
setTaskEnabled(TASK_ALTITUDE, sensors(SENSOR_BARO) || feature(FEATURE_GPS));
```

mikeller, 2018-06-03:
Good finding, @Smeat!

It looks like the line in question is executed in `fcTasksInit()` before the first frame from the GPS was received, resulting in the GPS sensor not being recognised (we only know it's there once it starts sending data on the UART), and thus the altitude update task is never started.

Starting the altitude update task speculatively based on the GPS feature being enabled (but without knowing that a GPS is present) isn't ideal, but I think in this case it is workable, because waiting for the GPS to be recognised is a worse option.

Do you want to open a pull request to change this, @Smeat?

PR #6034 (2018-06-03): Fix altitude task when using GPS only.
Fixes #5885.
closed 2018-06-03 by mikeller