betaflight/betaflight #5887: BF3.4 #835 NOX - CLI 'bl' does no  longer switch into DFU mode

RipperDrone, 2018-05-13:
BAck to normal after flashing back to BF 3.3.2

smasslich, 2018-05-17:
I have a JBF4 FC that uses the CLRacingF4 target.  I noticed that with 3.4.0 installed (not build specific) the bl command doesn't work either.  After typing command the pc won't recognize the fc, however, if I disconnect and reconnect it comes up normal, just not in DFU mode.  I have had to disconnect the R-XSR Rx and then it will go into dfu.

jflyper, 2018-05-18:
Yeap. MATEKF411 reboots on `bl` also. Let me take a look.

jflyper, 2018-05-18:
It's #5193.

DieHertz, 2018-05-18:
@jflyper have you found it by bisecting?
I'll have a look tonight

jflyper, 2018-05-18:
@DieHertz
> have you found it by bisecting?

Yeap.

ptrizio, 2018-05-19:
RipperDrone ciao hai provato ad entrare tramite gestione dispositivi, porta lpt, stm virtual port ,aggiorna driver,scegli da un elenco, e clicchi sul nome del tuo controllore di volo e dai ok. lui così dovrebbe andare in DFU. ciao patrizio

RipperDrone, 2018-05-20:
@ptrizio Haven't tried, but it clearly reboots instead of switching into DFU. Drivers are ok, since DFU works in BF3.3. Issue has been confirmed for other F4 targets and seems to be linked to timer reworknij BF code...

ptrizio, 2018-05-23:
ok

mikeller, 2018-06-02:
This issue still exists with current master. @jflyper, @DieHertz I thought you had found a fix for this?

ptrizio, 2018-06-02:
now it has happened to me also installing the firmware nox 3.4

PR #6221 (2018-06-26): Fixed call to checkBootloaderRequest()
Fixes #5887, overlooked during #5193

Tested on Nox v2.15, only affects F411 targets.
closed 2018-06-26 by mikeller