betaflight/betaflight #5888: SPRACINGF3 MSP link to BFC become intermittent when mag is enabled.

jflyper, 2018-05-14:
Happens on SPRACINGF3 flashed with latest 3.4 (eab55fc3e) ran with default config.

jflyper, 2018-05-14:
Bisection revealed 19a360b (#5847) being responsible.

jflyper, 2018-05-14:
Disabling `TELEMETRY` feature also stops the phenomenon... So, it has something to do with #5847 also.

mikeller, 2018-05-14:
Weird. Have you tried a reset to defaults?

jflyper, 2018-05-14:
No dice.

- Disabling compass, or
- Disabling `TELEMETRY`

will make it work...

jflyper, 2018-05-14:
[SPRF3-19a360b.mov.zip](https://github.com/betaflight/betaflight/files/2000155/SPRF3-19a360b.mov.zip)

mikeller, 2018-05-14:
Weird - especially the compass part. Could it be some resource conflict?

jflyper, 2018-05-14:
I found it. It's the serial TX buffer size decreased by #5432.
I think BFC is continuously requesting box names, which overflowed the 128 byte buffer with the introduction of the new telemetry box. Compass, baro and telemetry each enabled/disabled at least one box to trigger/untrigger this issue.

So, the challenge by #5432 failed, and dynamically allocated buffer suggested by #3650 is required. Fortunately, there is already https://github.com/iNavFlight/inav/pull/3189/commits/e84d73cf425a4e6afdbd9769738c4ffc44b4b40e. I will look into applicability of the iNav code.

In the mean time, a hot fix PR to temporary increase TX sizes to 256 will be submitted.

mikeller, 2018-05-14:
Good digging, @jflyper!

And yes, that makes sense, #5432 changed the telemetry box to be always available - kind of pointless to hide it behind a parameter that does nothing else but hide it.

The iNav fix looks good for a permanent fix.

(Side note, we might want to trial using `[ci skip]` on PoC pull requests that are not intended to be merged - not much point using CI resources for these.)

closed 2018-06-20 by stale[bot]