betaflight/betaflight #5890: Bad altitude when gps fix while flying

McGiverGim, 2018-05-14:
If I'm not wrong, the altitude now is a mix between baro and gps data. When arming is calibrated to zero.

The problem is that if I don't have GPS lock when arming, and I get GPS lock while flying, then the altitude is wrong by a lot. I've seen more than 200m while in ground.

Maybe is better to use only baro in this case or calibrate GPS height with baro when first GPS lock.

mikeller, 2018-05-14:
Is this with 3.4? I think we are mixing GPS and BARO for altitude, and I haven't seen code waiting for a GPS lock indeed. @s0up might know more.

McGiverGim, 2018-05-14:
Yes, that's with 3.4. I've not looked at the code, only what I'm supposing is happening.

s0up, 2018-05-14:
Currently we don't wait for the state flag gps fix to have estimated
altitude get calculated.  It might be wise to check, but when it comes down
to it, GPS altitude isn't very accurate until you have a decent amount of
sats, not just the bare minimum to be considered a fix.

McGiverGim, 2018-05-14:
My fix are usually good, about 9-10 satellites. If the fix is done before arming, all is good, if the fix is while flying, then is very bad. 
This PR https://github.com/betaflight/betaflight/pull/5834 fixes a problem, resetting to zero the baro altitude when arming. Maybe this is the difference between both situations.

s0up, 2018-05-14:
The altitude offset is constantly being updated when not in an armed
state.  I'm not sure what the ideal solution is for this.  The logic was
that rescue mode should do the best job it can to recover the craft vs
failure if a good lock wasn't present prior to arming.   Definitely up for
any improvement though 😀

McGiverGim, 2018-05-14:
Without arming before landing I have not a gps home position, so the rescue will not work 😋
It's only the info on the osd, is cosmetically very bad.

mikeller, 2018-05-14:
Yeah, I guess this will need some UX improvements. In a way I think it should be done so that if 'GPS rescue' is configured, the craft should not arm until it has a GPS lock. Otherwise users will take off on long distance flights thinking that they've got the 'safety net', and then realise they don't when they are lost.

s0up, 2018-05-14:
That is possibly a decent solution.  Inav has some stuff in place to
prevent that exact scenario from happening.  I need to look at how OSD
warnings are done, but maybe we can put in something similar.

McGiverGim, 2018-05-15:
I don't see the not arm if no GPS lock as a good solution. I think is enough to show a warning when arming "rescue deactivated" or similar with some beeps. In this way you can decide if land and wait until lock, or continue flying because you are not going to fly far.

mikeller, 2018-05-15:
@McGiverGim: Depends on how it's done - it should be obvious enough so that users won't accidentally take off without a fix only to find out that GPS rescue won't work when they're lost.

McGiverGim, 2018-05-15:
Maybe a CLI variable, MANDATORY, WARNING, NONE or similar can help. In this way the user can have as much security as he wants.

mikeller, 2018-05-15:
Possibly. Or the behaviour can be made configurable: `no_fix_arming = ALLOW | WARNING | PREVENT` or suchlike.

McGiverGim, 2018-05-15:
Yes, that is what I wanted to say :)

beeb, 2019-02-23:
Rescuciting this issue to say that the initial issue is still present: when both a baro and gps are present, getting a GPS fix in flight will completly change the altitude value. I uploaded a DVR https://youtu.be/pM8ZUZinun4

Would it be possible to force using one or the other? How would it affect GPS rescue is the value starts jumping between baro and GPS? Note that I took off having plenty of sats, but for some reason they dropped to 0 after a while and when coming back the value was jumping like shown in the video.

McGiverGim, 2019-02-23:
The GPS Rescue is deactivated if no fix while arming, so in theory it must not affect.

beeb, 2019-03-02:
> The GPS Rescue is deactivated if no fix while arming, so in theory it must not affect.

Well but in my video, there was a fix when arming, so GPS rescue was not deactivated. The fix was lost afterwards during flight. That's when I wonder what happens, because the displayed altitude jumps like crazy.

beeb, 2019-03-02:
Could it be that in https://github.com/BrainFPV/betaflight/blob/7fbee96449e0e522d908154c815ba0bc60331145/src/main/flight/position.c#L91 the `altitudeOffsetSet` flag is reset when the GPS fix is lost during flight? That could explain why values jump if the reference altitude from arming time is lost.

bakwc, 2021-05-30:
Same issue here, in 4.2.9 I got bad OSD altitude value.

FenziRocks, 2023-04-29:
Same issue with 4.4.0
I arm and the altitude is on zero.
Then I fly 300 meters away on 5 meters altitude.
Then I switch on gps rescue and altitude shows -40 meters.
What means it will try to fly to 5 meters + 20 meters height what ands up in 65 meters height.

Why is gps rescue not using barometer information if available instead of the bad gps altitude information?

FenziRocks, 2023-04-29:
If this is really working.
The issue can be closed.
https://github.com/betaflight/betaflight/pull/8127

haslinghuis, 2023-04-29:
Lols, we have two settings now:

      set altitude_source = [BARO_ONLY | DEFAULT | GPS_ONLY]
      set altitude_prefer_baro = [0 ...100]

https://github.com/betaflight/betaflight/pull/11834
https://github.com/betaflight/betaflight/pull/11841

closed 2023-04-29 by haslinghuis

FenziRocks, 2023-05-01:
@haslinghuis could you pls exlain me what  "set altitude_prefer_baro" 0-100 means? 100 is default. What happens when I set 0 compaired to 100?

haslinghuis, 2023-05-01:
> improves the existing basic sensor integration of Baro and GPS data in position.c, to favour Barometer more than previously. This is because we found that Baro was more stable, generally. Includes logic to avoid the 'ground effect' problem where throttling up on the ground gives a transient increase in pressure below the quad, causing a false negative baro reading. An altitude_prefer_baro parameter is provided in CLI, defaulting to 100, which sets the extent to which Baro data should be preferred over GPS. When set to zero, and both Baro and GPS are used, the old GPS-dominant method is used, which typically resulted in Baro accounting for 30% of the altitude estimate. When set to 100, Baro provides 90% of the altitude estimate.

Please check: https://github.com/betaflight/betaflight/pull/11834