betaflight/betaflight #5893: Re-enabling BARO mode (ALT HOLD) on F3 (Custom Betaflight Build)

fredericdepuydt, 2018-05-14:
Dear,

I'm trying to re-enable the BARO mode (ALT HOLD) on my SPRacing F3 Flight Controller.
I've already freed up enough space (removed all peripherals I won't be using). But I can't seem to find what flag/define I need to use for re-enabling the BARO mode.

Kind regards,
Frederic Depuydt

mikeller, 2018-05-14:
Hey @fredericdepuydt.

Altitude hold ('BARO mode') was removed in master because it did not work reliably, and work is under way to replace it with a new, GPS based 'Emergency Return To Home' function.

If you want to use the (almost unusable) BARO mode, you can do your modifications on top of the `v3.3.x-maintenance` branch.

johmarjac, 2018-06-17:
@mikeller I would like to get BARO mode (or Altitude Hold) working on Betaflight. I know Betaflight is totally focused on Performance and Race Quads but in my opinion something like this should not be missing. I am not actually a racer but I want to have good performance too when just flying free-style between trees and if I am fastly flying above ground altitude hold would be so much better to have.

I can see the barometer graph on OMNIBUS 4 in 1 Racerstar FC, just what is disabled is that actualy BARO mode, right? v3.3.x-maintenance branch has this still enabled ? Or would I have to do modifications in every case? I am fairly new when it comes to chip development but I want to give it a try actually. Can you maybe explain why it was jumping around in recent versions of BF ? Were filters the issue? Is it something that can be fixed when being worked on to bring it back?

mikeller, 2018-06-17:
@johmarjac: Altitude hold was completely removed for 3.4, when GPS rescue mode was introduced. The problem was that the old implementation was unmaintained, and did not work in a safe way - least of all close to ground, since it was prone to oscillations. Feel free to propose a new implementation in a pull request.

mikeller, 2018-06-18:
@johmarjac: That's a good start. Don't spend too much time on the altitude hold code in 3.3, this was badly maintained and has been patched up a lot of times just to keep it from completely falling apart. It might be better if you had a look at how iNav is doing altitude hold to get an inspiration for how to do it properly.

Re developing / debugging, it is not as hard as it might sound, have a look at https://github.com/betaflight/betaflight/blob/master/docs/development/Hardware%20Debugging.md to get an idea how little hardware is needed. And you might want to join the Betaflight Slack, registration at https://slack.betaflight.com/.

johmarjac, 2018-06-18:
@mikeller Allright, I requested registration in Slack with johannesmjacobs@gmail.com

Looking how INav is doing it sounds like a good idea actually, thanks for the hint, maybe I can get something to work. Now I will read the Hardware Debugging Markdown first.

aowi7280, 2018-06-30:
I would like to have this back as well. It helps when going high and doing a panorama shot.

Hotwire-, 2018-07-16:
Why is this feature such a troublesome thing to get right? Can I help?
It seams pointless putting a barometer on an FC if it is ignored.

mikeller, 2018-07-17:
@Hotwire-:

- there is more that you can do with a BARO sensor than just altitude hold - just getting an altitude indication in the OSD is valuable;
- BARO sensors are really susceptible to air flow over the sensor, temperature changes, vibrations - all of which there are plenty on an RC craft;
- with respect to air flow, there is also the aspect of feedback from the props to be considered, i.e. detected altitude decreases in altitude hold, prop speed is increased, changes air flow over the sensor, sensor detects change in altitude;
- the existing implementation had a lot of problems due to this, the craft would almost instantly start to oscillate as soon as altitude hold was activated;
- feel free to try and address these issues in a new implementation of altitude hold.

Hotwire-, 2018-07-17:
Hi Mikeller
I have old 3DR FCs that manage to use the barometer to control altitude. Smoothing date from the barometer would most likely quell barometer instability. So why is this a hot potato.
The thing most likely to adversely effect barometers is sunlight. While open cell foam rubber fixes airflow problems really well. 
Measuring altitude would always require calibration but really isn't the most essential thing to know.
Ardupilot is being ported to the STM32 FCs and I expect it's likely to be the best solution. It is funny though sad, that advances in computing often strive to equal previous achievements. Apart from the old PWM input and ESC control, Ardupilot had amazing abilities.
Betaflight is very good at basic flight stablisation and control but continues to refine rather than expand.
Thanks for your reply.

closed 2018-08-23 by stale[bot]

andre92, 2018-09-01:
Is it possible that altitude hold oscillates because the vertical acceleration is only calculated at 100hz, together with the IMU? It worked fine before the IMU loop was linked to the PID loop at 1khz. 
From the top of my head, barometer is sampled at 40hz, but altitude hold relies on the vertical acceleration/velocity as calculated by the IMU. Maybe it works better if IMU frequency is increased to let's say 200hz?

johmarjac, 2018-09-01:
@0Jiahao Hi, I am sorry, I havent been looking into that anymore as I am quite too unexperienced with that and I obviously dont wanna burn/rekt my FC.

mikeller, 2018-09-02:
@andre92: The problem was that the code calculating the altitude estimate from BARO, ACC, GPS, and SONAR (all if available) was outdated and had not been updated for years of changes and improvements to the ACC and IMU code. This led to the input value for altitude hold already badly oscillating for most setups.

chrisalbertson, 2018-09-08:
Altitude hold may have been poorly implemented and that is way it was removed.    But other FC software is getting good BARO based flight modes using the same exact hardware.   

ArduPilot is now running of F4 and F7 racing FCs and of course their altitude based modes are dead-on.   They do something very different.  They run a Kaman filter that fuses the available sensors to track the state of the quad.  So the software i not just using the pressure sensor but also the gyros, accelerometers, GPS and possible LIDAR.  All the sensors have strengths and weaknesses the filter is like magic and combines the best of each.   The trouble is the amount of computing for that kind of sensor fusion.    

One solution is to use one software (Betaflight)  for racing and extreme freestyle and other software for everything else.

I saw a large quad copter set on position hold recently flying indoors.   The owner took a boom stick and pushed it down and to the side rather forcibly.   The machine really did NOT want to move and would return to it's exact hight and location.     There was no magic hardware inside,  just an F4 chip.   It was using the barometer but fusing this with other sensors.

My point is that poor altitude hold is not a fault of the hardware or air plowing of the sensor.

This may come full circle and my bet is that stabilized flight modes will start winning races.     My vision for where this should go is aerodynamic quad bodies that generate lift like airplanes and flight controllers that figure out the fastest route from point A to point B in real-time.  Orders of magnitude increase in computer power is required but it CAN be done -- we do it where I work for MUCH larger airframes.

digitalentity, 2018-09-08:
Kalman filter is not a magical tool - it's a way of fusing together multiple sources of data - pretty much similar to what INAV is doing. Key problem is that on miniquads accelerometer don't have enough dynamic range - momentary vibration amplitude may exceed 16G, accelerometer will saturate and estimate will go crazy - seen this on INAV, seen this on APM, seen this on Pixhawk - it's a hardware design problem which can't be solved in software (unfortunately).

Regarding BARO & Betaflight - Betaflight trades everything for rate mode performance which makes things even less accurate - you simply don't have resporces to make sensor fusion work with sufficient accuracy which means you are pretty much limited to be using raw sensor (baro) data which is not nearly accurate. INAV does sensor fusion and altitude hold is much better - still prone to drift due to pressure changes if using only baro. Rangefinder makes things rock solid.

timman2er, 2018-10-27:
I have a BARO so when I find the time I'd like to reimplement this.

@mikeller if I'm reading this correctly the issue is not only the algorithm for maintaining altitude but also getting good accurate input? If that's so I think I could improve the altitude estimate (which shouldn't be too expensive?) and implement the algorithm both from iNav.

chrisalbertson, 2018-10-27:
Yes, simply looking at the barometer is not the best plan, You need to
estimate the altitude using accelerometer data and suitable lowpass filters.

There is an argument that says the Betaflight is for racing in acro mode
and you don't need good state estimation for that.  But "just wait"  Not to
 many years from now fuly automated flight controllers will be wining races.

chrisalbertson, 2018-10-27:
> @chrisalbertson <https://github.com/chrisalbertson>
>
> I think the earliest time that will happen is low ground races where
> maintaining the same altitude is an advantage. That being said you will
> never be in full control to make a move which could be devastating.
>
> My biggest concern that this project is/should/will remain focused on acro
> performance. But I think a large portion of the users come for good flight
> performance but wouldn't mind trading a little in acro performance for
> better modes. iNav aimed at this but I heard it seems to be focused too
> much on navigation for it to be feasible, as people claimed here that their
> flight performance and other features lack behind on betaflight.

What I meant was not that people would be racing in BARO mode but that that
will be racing is "AUTO" mode.   We don't have an auto mode today.   But in
this mode the data from the FPV camera is used to drive a computer vision
system and the FC finds the best route.     Think of the way an F22 fighter
pilot targets an air to air missile.  He just says "hit that thing" and the
auto pilot works the details.

In the proposed "auto" mode the pilot just says "fly to that point" and
then that point and so on.  Possibly not even using joy sticks but hand
gestures and virtual reality.   With 3D giggles the pilot can touch the
point in 3D space or trace an arc with his finger and the copter flies that
arc.   Not tomorrow but some day the FC will have 100X more compute power

As I type this I'm working on voice control.   It's simulated now.  The
machine can only say back to you what it would do.  I can say "move up 2
meters" and the machine say "moving, 2M upward"   It's a start.   It would
never work for racing, voice is to slow.   I will have his working on a toy
R/C car in a few months, a multicopter later.

Google can make self driving cars that drive at hi way speeds  we will have
self flying racing drones.   In fact that seems like a fun sport

timman2er, 2018-10-27:
I think anything as autonomous as you're describing would require a new fork as I think this will never be added to bef upstream. iNav seems a little inactive.

Also, getting a user/dev base will be incredibly hard as DJI's products work beyond good at these things. 

In what case do you think you can get an open-source non commercial initiative to launch?

mikeller, 2018-10-28:
@timman2er: Yes, altitude hold will have to be based on input from multiple sensors, barometer (slow, prone to drift), accelerometer (faster, but only relative measurements), GPS (if available), sonar / lidar (if available). Porting something from iNav may be a solution. Be aware that this will most likely only be possible for F4 and better - we simply don't have the flash to spare on F3.

timman2er, 2018-10-28:
@chrisalbertson

Ima require you to link the vid.

rs2k, 2018-11-25:
> One of my favorite ArduPilot videos shows a normal "Flamewheel" type quad copter being placed on position/altitude hold in a 60 mile per hour wind storm that was bending trees.

I did something like that 4 years ago. It was crazy gusty in the area I was videoing my hex.
https://www.youtube.com/watch?v=hk88YWpxKFY

Ardupilot is a fantastic project and the devs are great people who really know what they're doing. It really shows that the magic is in the software, not the hardware. PIDs are okay for emergency navigation, but EKF is really the only way to do proper navigation. The sensors by themselves are inadequate at doing the job properly so you must use good sensor fusion and a proper filter like the Kalman.

BobRob77, 2018-11-29:
Many of the issues discussed here is due to legacy: Flight Controllers are traditionally based on MCU:s, predominantly the STM Family of 32 bit ARM units. Even though the F7 generation gives a lot more processing power than earlier generations, they are still not the ideal tool for computation heavy applications, which is evident by developers having to prioritize one capability over the other.

Starting from scratch today however, using STM MCU:S likely would not be the ideal choice. At similar cost and size there are plenty more powerful alternatives, look for example at units like the Beaglebone Pocket Beagle, using a system-in-package containing a 1 Ghz Cortex, 512 MB of RAM and PRU:s for real time communication with peripheral hardware, the entire board is $28 and the entire chip is hardly bigger than an F7, basing an FC on hardware akin to this would allow for features we can only imagine currently.

chrisalbertson, 2018-11-30:
> Many of the issues discussed here is due to legacy: Flight Controllers are
> traditionally based on MCU:s, predominantly the STM Family of 32 bit ARM
> units. Even though the F7 generation gives a lot [...]
If you want " features we can only imagine currently" then move from
Betaflight to ArduPilot.  It can do position and altitude hold in a big
wind storm using F7 based flight controler.

as you say, the Cortex A type ARM processors are more powerful than the
Cortex M that are used by Betaflight.  But the A will run Linux and is not
as good for real-time control as the M.  A great solution you can do today
is to have both the F7 and a Raspberry Pi3.  A system like this with is
suported with current software.  In ArduPilot's terminalogy the Pi3 is a
"companion computer"  I'm working on a system that uses this setup but in
addition the Pi3 is connected vi WiFi to a much larger server class
computer that can do things like computer vision and natural language
processing.   I want a voice and vision controled user interface.    But of
ccourse this is NOT a racing drone.  This copter has 10 inch props.

My conclusion is that Betafligh is for people who fly in ACRO mode.  For
everytrhing else you'd be best off using different software and such
software does exist and it is free.  So I for one have desiged not to try
to use racing software for robotic flight modes.