betaflight/betaflight #5901: 2 RSSI indicators (TBS LQ / RSSI) - custom OSD field.

Szpero, 2018-05-16:
Is there possibility to put 2 (RSSI & LQ) indicators on OSD? Sometimes LQ on TBS is quite confusing especialy when it's switching between modes 4/50/150.

I was even thinking about custom filed for OSD when user can choose channel, range and icon - this will give even more flexibility.

bigd942, 2018-05-20:
I put a similar idea forward on the Crossfire FB page at the start of the week, would be good if the alarm defaults were set at the correct values as well for the particular link quality indicator, so for LQ default warning = 70, for RSSI = XX etc.

laz2wiringpi, 2019-05-21:
#8247, #8180  merged to 4.1  no aux channel needed.
  link quality 300-0

closed 2026-09-26 by blckmn