betaflight/betaflight #5906: Latest Build - Gyro feels like rubber

TheIsotopes, 2018-05-16:
the latest build 841 have an issue with gyro ... on bfc you can see the gyro is not running or its walk like rubber ... imho on of this #5898 or #5898 causes the issue

@jflyper  @etracer65  pls can you both see what's going on there

etracer65, 2018-05-16:
The slow gyro movement in the configurator means that the gyro calibration hasn't completed.

You either need to let the flight controller stay still after powering, or you may need to increase the `moron_threshold`.

That flight controller has an ICM gyro which tends to be noisy.  Before #5898 there was a bug in that for gyros running with 32KHz sampling the calibration was only running for about 0.9 seconds instead of the full 3 seconds.  Thus if you have a noisy gyro it's more likely that the calibration won't complete without a higher `moron_threshold`.

etracer65, 2018-05-16:
Yes, I agree there's some problem.  Not sure how the simple fix in #5898 caused it but seems to be the case.  I'm looking into it.

etracer65, 2018-05-16:
Created #5907 to fix the problem.

PR #5907 (2018-05-16): Fix gyro calibration zero offset calculation
Fixes #5906

It seems like the gyro calibration sample count overflow fix (#5898) caused a downstream problem with the math to calculate the zero offset.  Casting the components of the formula to (float) solves the problem.
closed 2018-05-16 by mikeller