betaflight/betaflight #5918: Led strip not work correctly

DiabloNero92, 2018-05-19:
Matek F405-CTR Flight Controller
Betaflight_3.3.2_MATEKF405
Rubycon ZLH 35V/470uF
Led work correct only when first start controller from usb and then connect battery. If i connect battery first, leds is not expected color and not disable from toggle switch. Betaflight 3.2 similarly

mikeller, 2018-05-19:
@DiabloNero92: The video does not load for me, but what you describe is most likely a voltage level problem, and can be fixed by adding a diode, as descibed in the bottom paragraph of this https://github.com/betaflight/betaflight/blob/master/docs/LedStrip.md#connections

DiabloNero92, 2018-05-19:
@mikeller is it mean my VIN(5.04v) too hight?
after added diode (VIN-diode-led) VIN amounted 4.17V
[image]

may be a problem in the capacitor on XT-60?

mikeller, 2018-05-19:
4.17V should be fine. Weird that it still doesn't work. What type of LED_STRIP are you using?

mikeller, 2018-05-19:
These look ok. What is your `diff`?

DiabloNero92, 2018-05-19:
if I understood correctly what mean 'diff':
when work wrong  VIN=4.15 data=3.27
when work correct  VIN=4.15 data=0.04-0.07

mikeller, 2018-05-19:
> 5. Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); and

DiabloNero92, 2018-05-19:
I don't setup after flashing, only led
```
# diff

# version
# Betaflight / MATEKF405 (MKF4) 3.3.2 May  4 2018 / 09:45:06 (64a292f37) MSP API: 1.38

# name

# resources

# mixer

# servo

# servo mix

# feature
feature LED_STRIP

# beeper

# map

# serial

# led
led 0 2,12::LI:4
led 1 3,12::LI:2
led 2 4,12::LI:2
led 3 5,12::LI:2
led 4 13,12::LI:4
led 5 12,12::LI:2
led 6 11,12::LI:2
led 7 10,12::LI:2

# color

# mode_color
mode_color 7 0 4

# aux

# adjrange

# rxrange

# vtx

# rxfail

# master
set motor_pwm_protocol = DSHOT600
set yaw_motors_reversed = ON
set pid_process_denom = 2

# profile
profile 0

# rateprofile
rateprofile 0
```

mikeller, 2018-05-20:
I had a look into how the target is configured. Unfortunately the pin for the LED_STRIP is shared with I2C, and I2C is initialised first, grabbing the pin. `resource I2C_SCL 1 none` in CLI should fix this.

(Interesting side note: The weird LED pattern that you are getting isn't related to the flight controller's LED_STRIP driver at all - it is just the I2C SCL signal that's output on this pin being interpreted by the WS2812 controller. :laughing:)

DiabloNero92, 2018-05-20:
@mikeller yes, it work correctly, thank you so much

closed 2018-05-20 by DiabloNero92