betaflight/betaflight #5920: Flight controller stuck after reboat

heziben, 2018-05-19:
Hi
Flight controler: MATEK-F405-OSD

I am able to flash the betaflight firmware version: 3.3.2 - MATEKF405 - 04-05-2018(15:21)
After finishing firmware upload I can connect to the board using betaflight configurator version 10.2.0.
If I change any setting in the configuration tab and press "Sava and Reboot" I can't access the board again. If I reinstall the firmware then I can access it but the problem consist.

Thanks for helping,
Hezi

Here is the configurator   log:

2018-05-19 @ 18:53:44 -- Running - OS: Windows, Chrome: 66.0.3359.181, Configurator: 10.2.0

2018-05-19 @ 18:53:44 -- Using cached release information for configurator releases.

2018-05-19 @ 18:53:46 -- Using cached release information for firmware releases.

2018-05-19 @ 18:54:10 -- USB device successfully opened with ID: 3

2018-05-19 @ 18:54:10 -- Detected device with total flash size 1024 KiB

2018-05-19 @ 18:54:10 -- Read protection not active

2018-05-19 @ 18:54:26 -- Erased 1024 kB of flash successfully

2018-05-19 @ 18:54:48 -- USB device successfully closed

2018-05-19 @ 18:54:52 -- Serial port successfully opened with ID: 10

2018-05-19 @ 18:54:52 -- MultiWii API version: 1.38.0

2018-05-19 @ 18:54:52 -- Flight controller info, identifier: BTFL, version: 3.3.2

2018-05-19 @ 18:54:52 -- Running firmware released on: May 4 2018 09:45:06

2018-05-19 @ 18:54:52 -- Board: MKF4, version: 0

2018-05-19 @ 18:54:52 -- Unique device ID: 0x2d00243436510b33323835

2018-05-19 @ 18:54:52 -- Craft name:

2018-05-19 @ 18:54:52 -- Arming Disabled

2018-05-19 @ 18:56:57 -- EEPROM saved

2018-05-19 @ 18:56:57 -- Device - Rebooting

2018-05-19 @ 18:56:57 -- Serial port successfully closed

2018-05-19 @ 18:57:01 -- Serial port successfully opened with ID: 11

2018-05-19 @ 18:57:11 -- No configuration received within 10 seconds, communication failed

2018-05-19 @ 18:57:11 -- Serial port successfully closed

mikeller, 2018-05-19:
What setting are you trying to change? Some settings can cause the flight controller to become unresponsive.

heziben, 2018-05-20:
All kind of settings cause the problem.
The two I recall for sure are the "Mixer" and "ESC/motor protocol" (not together)

heziben, 2018-05-20:
I don't understand? how do I check it?

heziben, 2018-06-15:
Anyone can help?

mikeller, 2018-06-16:
Can you try if this still exists with 3.4.0-RC1?

closed 2018-07-23 by stale[bot]