betaflight/betaflight #5923: Stick inputs open both MWOSD and Camera control menus at once

drprops, 2018-05-20:
I have a MWOSD as well as camera control setup. When I input the stick commands to enter the OSD on MWOSD, because it is the same as for the camera control (pitch forward, yaw left), both menus open at the same time.

I was wondering if there was a way to change what stick inputs are required to enter the camera OSD. Possibly allowing the option of using Yaw right + pitch full instead? This way the minimosd menu won't pop up as well. Thanks!

mikeller, 2018-05-25:
@DieHertz: Do you have any suggestions for what could be done for this?

drprops, 2018-05-25:
Thank you for your responses! I made a mistake, I meant pitch forward + yaw right. I will try moving the sticks differently. 

I think part of the issue is that MWOSD also uses the pitch forward + yaw right combination to open up the MWOSD menu. I was wondering if there was a way to reverse the yaw direction to activate the camera control menu, in order to avoid opening two menus at once?

I am using the following setup:

Lumenier Lux V2 FC
minimOSD micro with MWOSD 1.6
Caddx micro turbo F1 (200 ohm resistor between LED pin on FC and the OSD pin on the camera)
FrSky QX-7
FrSky XSR receiver

DieHertz, 2018-05-26:
`Camera control ENTER` is yaw right, not yaw right + pitch forward, if you move stick simultaneously, it shouldn't trigger entry to camera settings menu when entering MWOSD.

drprops, 2018-05-26:
Thanks for the info. I didn't realize that it was just throttle middle + yaw right.

I am running into the issue of the camera control menu being activated when the MWOSD is already open if I yaw right with the throttle at center. I don't think that the board realizes that the MWOSD is open, so it still allows the camera control menu to open after the MWOSD is already open.

My workaround for now is to just put the throttle at minimum after opening the MWOSD. I can still select with yaw right + throttle minimum in mwosd. But if my throttle is at center and I yaw right, then the camera control menu open behind the MWOSD menu. It would be nice if there were a way for the FC to know when the MWOSD is open in order to avoid opening the camera control menu behind though.

Thanks again for your help!

mikeller, 2018-06-02:
@drprops: Is this working for you now? If so, can you please close this issue?

drprops, 2018-06-05:
I still run into the issue of the camera control menu popping up if I'm in the MWOSD and the throttle is at center, and I touch the yaw.

My work around for now is to keep the throttle at 0 when in MWOSD, since I can still select options that way. It would be nice if there were some way for the FC to know that MWOSD is active so it could prevent the camera control menu from popping up. 

I will close the issue though.

closed 2018-06-05 by drprops