betaflight/betaflight #5925: Max Velocity on Roll axis is not what i set on bf configurator.

Vidalcris, 2018-05-20:
Hi,
I got a weird problem with my OmnibusF4 (HGLRC F440).
The velocity set on betaflight configurator is ok on Yaw and Pitch but not on the Roll axis.
On the Roll axis, i can see 345°/s max in the blackbox (and i also feel it) even if the setting is at 400°/s.

```# Betaflight / OMNIBUSF4 (OBF4) 3.4.0 May  9 2018 / 13:01:02 (3af1610d0) MSP API: 1.39

# name
name BH130-Crisssss

# resources
resource MOTOR 1 B01
resource MOTOR 2 A02
resource MOTOR 3 B00
resource MOTOR 4 A03
resource SERIAL_TX 3 NONE
resource SERIAL_TX 11 B14

# mixer

# servo

# servo mix

# feature
feature SOFTSERIAL
feature LED_STRIP
feature AIRMODE
feature ESC_SENSOR
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -SYSTEM_INIT
beeper -ON_USB

# map

# serial
serial 0 64 115200 57600 0 115200
serial 2 1024 115200 57600 0 115200
serial 5 16384 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200

# led
led 0 7,8::A:13
led 1 8,8::A:13

# color

# mode_color
mode_color 6 0 4
mode_color 6 1 13

# aux
aux 0 0 0 1700 2100 0
aux 1 13 2 1700 2100 0
aux 2 26 1 1700 2100 0
aux 3 35 2 1300 1700 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 3 h
rxfail 6 s 1000

# master
set gyro_lowpass_hz = 100
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 215
set gyro_notch2_hz = 0
set gyro_notch2_cutoff = 150
set gyro_overflow_detect = YAW
set acc_calibration = 68,150,-115
set mag_hardware = NONE
set baro_hardware = NONE
set rc_interp = MANUAL
set rc_interp_int = 12
set serialrx_provider = IBUS
set dshot_idle_value = 550
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT1200
set failsafe_delay = 10
set align_board_yaw = 90
set vbat_min_cell_voltage = 32
set vbat_warning_cell_voltage = 33
set current_meter = NONE
set small_angle = 180
set yaw_deadband = 1
set pid_process_denom = 1
set tlm_inverted = ON
set osd_tim_2_pos = 2416
set osd_vtx_channel_pos = 2071
set osd_craft_name_pos = 2048
set osd_pidrate_profile_pos = 2105
set osd_avg_cell_voltage_pos = 2408
set osd_disarmed_pos = 2347
set osd_esc_tmp_pos = 2065
set osd_esc_rpm_pos = 2444
set debug_mode = NOTCH
set vtx_band = 5
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5658

# profile
profile 1

set dterm_lowpass_type = PT1
set dterm_lowpass2_hz = 200
set dterm_notch_hz = 0
set vbat_pid_gain = ON
set anti_gravity_gain = 4000
set dterm_setpoint_weight = 100
set p_pitch = 30
set i_pitch = 55
set d_pitch = 14
set p_roll = 27
set i_roll = 50
set d_roll = 14
set p_yaw = 40
set i_yaw = 50

# rateprofile
rateprofile 0

set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 10
set roll_srate = 50
set pitch_srate = 50
set yaw_srate = 50
set tpa_rate = 20  ```

McGiverGim, 2018-05-20:
Attach the blackbox file too if you have it

Vidalcris, 2018-05-20:
Here is the BBL : https://ufile.io/ko17w

McGiverGim, 2018-05-20:
What I see in the configurator, is that you are not moving the roll until the end. Maybe a bad calibration of the TX?

McGiverGim, 2018-05-20:
I see zones with until 390º (third log, 1:51:853) so I think all is correct, but for some reason you are not moving the roll until the end. Have you seen some part of the log where you have moved it until the end?

Vidalcris, 2018-05-20:
If you look the BBL you can see that : https://ibb.co/fARC88
We can see that the stick is at the end because of the 0 value right ?
So i think the stick is at the end .. its weird :-/

McGiverGim, 2018-05-20:
The version of blackbox that you have is outdated. Please tell me log and time and I will look at the latest master.

McGiverGim, 2018-05-20:
The capture you have attached, the zero is the PITCH, not the ROLL. The ROLL is at 468 and the maximum is 500.

What I see is that the quad goes more or less exactly at the speed commanded with the roll stick.

I suggest you to use the latest version of blackbox viewer too, you can get it from here if you aren't able of build it directly: http://andwho.sytes.net:8080/job/Borisb_Blackbox_Log_Viewer/

Vidalcris, 2018-05-20:
Oh the zero is the pitch .. my bad :-/

I will make some new test tomorrow !
I will also install latest BBexplorer :)

Thanks a lot for your help !

closed 2018-05-20 by Vidalcris