betaflight/betaflight #5930: RevoltOSD Telemetry issue / Sharing uarts?

ImSundee, 2018-05-21:
With the revoltosd board, they have done mix and match uarts.

To run CRSF and get telemetry, the ports you put the cables into are TX1 and RX3.

TX3 is used for smart audio. 

Currently BF doesn't appear to allow you to do this, so we can only get signal and no telemetry. Is there a way we can make uarts multi use like this?

Cheers

mikeller, 2018-05-22:
This is a weird and unusual configuration. Maybe @proggod can shed some light on how to connect a full duplex, non inverted UART (e.g. Crossfire) to a REVOLTOSD board?

ImSundee, 2018-05-22:
I agree @mikeller it is a unusual setup on the revolts and how flightone have done this. 

[image]

This is the port that you use for your RX and as you can see its TX1 and RX3 for CRSF.

mikeller, 2018-05-22:
This is an utterly weird way of doing things, and probably means that both UART 1 and 3 are blocked just to enable Crossfire RX, where one UART alone is all that is really needed. Looking at the image above, there seems to be TX3 broken out on the connector up top, so it should be possible to connect the Crossfire RX to UART 3 alone.

ImSundee, 2018-05-22:
Indeed you can do that. That one is in the output cable to the vtx and they
set it in flightone for smartaudio.
Im just hoping to get the wiring the same for betaflight so people can
easily change over between both without having to change wiring

mikeller, 2018-05-22:
@ImSundee:

> Im just hoping to get the wiring the same for betaflight so people can easily change over between both without having to change wiring

I am totally with you on this, and that is what we try to do in Betaflight in general. However, one difference between flightone and Betaflight is that we are supporting 130+ different boards, compared to the small number of boards that flightone is supporting. For this reason it is not desirable for us to add support for completely unusual configurations just to enable this for one single board, if this means that the code that is used to support all boards will become more complex and harder to maintain.

Betaflight does have a good solution for serial ports with unusual pin assignments with SOFTSERIAL, but unfortunately I think this will not work for Crossfire, due to the high bit rate used.

We'll have to see if we can come up with a good solution for this, but for now users will just have to rewire their boards if they want to use Crossfire on a REVOLT board.

ImSundee, 2018-05-22:
Thats not a problem. Thought there might be something interesting we could
do, or if there was a CLI command that we could remap the crossfire TX port
:)

mikeller, 2018-05-22:
@ImSundee: There are some UARTs that can be remapped onto the same pins that can be used by other UARTs. Unfortunately the pins for TX1 and RX3 aren't amongst them. :disappointed: 
But if somebody comes up with a way of doing this that does not require us to work around it for all future work on UART support then we can definitely do it.

ImSundee, 2018-05-22:
That makes sense, what about separating out CRSF? as this would allow
people that might damage a pad.

closed 2018-06-28 by stale[bot]