betaflight/betaflight #5931: Betaflight F4 'no gyro' gyro not initializing

iPilotx, 2018-05-21:
So, I keep having an issue were my FC will not be ale to Arm, as a warning flag comes up showing 'no gyro' I have read that this may be a hardware issue however, I have experienced different things which indicate to me that it may be another issue.

- I have tested my quad with a full battery and a discharged battery no change.
- I have flashed multiple firmware 3.2.5; 3.3.1; 3.3.2 with no difference in result.

However when alI seemed lost and I was about to right off this board and replace it with a new one I noticed that when I connected it via USB to my Desktop computer I would initially have no gyro being sensed and after cycling the USB connection, all seems as good as new.

I am able to ARM and Fly no problem. But my frustration comes back in full circle when I simply leave my quad over night or for the rest of the afternoon and find myself with the same problem I initially thought I fixed the previous day. Quite annoying when this happens to you at the field.

I have the exact same setup in another quad the only difference being it has a Betaflight F3 instead of an F4.

Any help to get to the bottom of this before I rip it out and order a new FC would be greatly appreciated 

This is my Diff 

version
# Betaflight / BETAFLIGHTF4 (BFF4) 3.2.5 Feb 10 2018 / 23:55:03 (6e69ff00c) MSP API: 1.36

# name
name PILOT

# resources

# mixer

# servo

# servo mix

# feature
feature MOTOR_STOP
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -ON_USB

# map

# serial
serial 0 64 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 5 0 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100
aux 1 1 2 900 1275
aux 2 13 1 900 1300
aux 3 28 2 1325 2100

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_sync_denom = 2
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set baro_hardware = NONE
set rssi_channel = 12
set serialrx_provider = CRSF
set motor_pwm_protocol = DSHOT1200
set current_meter = ADC
set battery_meter = ADC
set vbat_scale = 109
set ibata_scale = 220
set pid_process_denom = 1
set osd_cap_alarm = 1550
set osd_vbat_pos = 2402
set osd_rssi_pos = 2437
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2421
set osd_flymode_pos = 333
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_crosshairs = 200
set osd_ah_sbar = 200
set osd_ah_pos = 200
set osd_current_pos = 385
set osd_mah_drawn_pos = 2454
set osd_craft_name_pos = 2476
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2442
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 2411
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83

# profile
profile 0

set dterm_lowpass_type = PT1
set dterm_notch_hz = 0
set vbat_pid_gain = ON

# rateprofile
rateprofile 0

set rc_rate = 135
set rc_rate_yaw = 135
set rc_expo = 35
set rc_expo_yaw = 30
set roll_srate = 73
set pitch_srate = 73
set yaw_srate = 65
set tpa_rate = 40
set tpa_breakpoint = 1500

mikeller, 2018-05-22:
Have a look at #5321. This could be another case of insufficiently stabilised voltage for the gyro.

etracer65, 2018-05-23:
The times where I've seen this kind of problem it's always been a hardware issue.  Usually mechanical damage to the actual gyro or the chip having broken solder connections to the circuit board.  When it warms up a little and things expand it can sometimes reestablish connections and appear to work.  Try pressing down on the gyro chip when powering the board as this might help confirm the problem.  You can always try reflowing the chip if you have a hot air station or there's always the toaster oven method...

mikeller, 2018-05-23:
Since these boards are too small to be useful as door stoppers, even the 'cigarette lighter hobo reflow' is an attractive option here. :-D

closed 2018-06-29 by stale[bot]