betaflight/betaflight #5943: Reset altitude at arm

Lord-Quake, 2018-05-22:
I'm new to BetaFlight (coming from MultiWii) and I rely on accurate alt readings displayed in the OSD.

Request: I would like to see the altitude be set to 0 as soon as the copter is armed. This should always be done every time arming is activated.

Presently I using BF 3.4.0. build #793.

Thanks!

mikeller, 2018-05-22:
@Lord-Quake: This has been implemented in the latest versions of 3.4, so updating to a newer build should fix this.

AirBreak69, 2018-05-22:
The latest builds (850+) that incorporate the new GPS RESCUE mode will do so for gps and baro altitude components. This eases the altitude handling during the RTH navigation.
Please close this as it should be solved in next release.

Lord-Quake, 2018-05-23:
OK, that is goods news. Thanks.

closed 2018-05-23 by Lord-Quake