betaflight/betaflight #5951: Betaflight does not Arm

Talha909, 2018-05-23:
I have Eachine Bat . This quard has this board AIOF3PRO_Brushed FLight-control Board Built-in OSD Receiver for Eachine BAT QX105 QX95S Vtail QX110.  

My quard was working great with the old version:-  Betaflight / OMNIBUS (OMNI) 3.2.4 

Yesterday I installed the new version and copy the old dump file.  After that quad did not arm. I give the command #status. But there is no error.   In new version only one time motor spins and stop.  And this also happens with the new setting . After i put the old version. And all the things are good. Can some one tell me what here is problem withe new version.

mikeller, 2018-05-23:
Was the previous version of the firmware installed through the Betaflight configurator, or was it the one that came pre-installed? This is not an original board, but a clone, and the manufacturer does not provide us with information about this board, so there are limited options for us to support you.
I think you best contact the supplier or manufacturer and ask them to provide you with a working firmware / configuration.

Talha909, 2018-05-23:
I have old dump file.  Do you need that one.

mikeller, 2018-05-23:
Were you connected to the configurator when you tried arming for the second time? Recent firmware has got a safety feature that disables arming while connected to the configurator, to prevent accidental arming when users are testing their RX.

Talha909, 2018-05-24:
Thanks for the support.  But when I unplug it . Quard not arm. Quard only arm when in motor tab i unable the motor safety off.

mikeller, 2018-05-24:
Not sure this is supportable, since nothing about the hardware you have has been published by the manufacturer. You said you have a dump of the original firmware that came preinstalled. Can you please post it here?

Talha909, 2018-05-24:
Please see this original file. 

`# Betaflight / OMNIBUS (OMNI) 3.2.1 Oct 15 2017 / 19:42:04 (434db9a36) MSP API: 1.36
name QX105
resource BEEPER 1 C15
resource MOTOR 1 B08
resource MOTOR 2 B09
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 B07
resource MOTOR 6 B06
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B04
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A14
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A15
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource I2C_SCL 1 NONE
resource I2C_SCL 2 NONE
resource I2C_SDA 1 NONE
resource I2C_SDA 2 NONE
resource LED 1 B03
resource LED 2 NONE
resource LED 3 NONE
resource RX_BIND 1 NONE
resource RX_BIND_PLUG 1 NONE
resource TRANSPONDER 1 A08
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 NONE
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
[... 495 lines cut ...]
set d_alt = 0
set p_level = 50
set i_level = 50
set d_level = 100
set p_vel = 55
set i_vel = 55
set d_vel = 75
set level_limit = 55
set horizon_tilt_effect = 75
set horizon_tilt_expert_mode = OFF
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 0
set rc_expo_yaw = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80
set tpa_rate = 10
set tpa_breakpoint = 1650

`

mikeller, 2018-05-24:
The two dumps look reasonable similar, nothing is standing out. But in general, it is wrong and a recipe for trouble to use a `dump` for a backup / restore during a firmware update, because this will dump all of the default values, and then restore them, even if they have changed in the new version.

Can you downgrade to the original 3.2.1, then use a `diff` to print only the modified values, save this, then update to 3.3.2 again, and copy / paste the diff file, just like you did with the dump. Then test again and see if this helps.

Talha909, 2018-05-24:
HI, 

I check the both diff these are min finding

set   motor_pwm_protocol = BRUSHED only seen in the 3.2.1 and this line is not seen in the new version.  

set use_unsynced_pwm = OFF is only seen in the 3.3.2 version

Talha909, 2018-05-24:
thank, 
I will check it and let you know. By the way what this command do. 
what is the meaning of this "set motor_pwm_inversion = OFF or ON"

closed 2018-06-30 by stale[bot]