betaflight/betaflight #5952: Initialization fails on dshot, matekF722, build 850

toiminto, 2018-05-23:
My buddy reports there seems to be an issue with initializing when dshot is enabled, Matek F722, Betaflight 3.4 build 850. On build 849 works fine, when plugging in battery, motors sound all the initialization beeps. 

On build 850, when dshot is selected, there are only the three initial beeps, esc:s dont get initialized. Multishot works fine

diff (with multishot as protocol, otherwise the same as with dshot)
```
# diff

# version

# Betaflight / MATEKF722 (MKF7) 3.4.0 May 21 2018 / 02:43:24 (041362614) MSP API: 1.39

# name

name gm

# resources

resource MOTOR 6 NONE

resource CAMERA_CONTROL 1 A08

# mixer

# servo

# servo mix

# feature

feature -TELEMETRY

feature AIRMODE

feature ANTI_GRAVITY

feature DYNAMIC_FILTER

# beeper

beeper -ON_USB

# map

map TAER1234

# serial

serial 4 8192 115200 57600 0 115200

# led

# color

# mode_color

# aux

aux 0 0 0 1700 2100 0

aux 1 13 1 1700 2100 0

aux 2 26 2 1700 2100 0

aux 3 35 1 1700 2100 0

# adjrange

# rxrange

# vtx

# rxfail

# master

set gyro_lowpass_hz = 120

set gyro_lowpass2_hz = 300

set gyro_notch1_hz = 0

set gyro_notch2_hz = 0

set acc_hardware = NONE

set acc_calibration = 23,15,23

set mag_hardware = NONE

set baro_hardware = NONE

set min_check = 1010

set max_check = 1990

set rc_interp_ch = RPYT

set serialrx_provider = FPORT

set serialrx_inverted = ON

set serialrx_halfduplex = ON

set blackbox_p_ratio = 64

set min_throttle = 1035

set dshot_idle_value = 220

set motor_pwm_protocol = MULTISHOT

set motor_pwm_rate = 16000

set vbat_max_cell_voltage = 44

set vbat_warning_cell_voltage = 34

set vbat_scale = 108

set ibatv_scale = 179

set ibatv_offset = 100

set beeper_dshot_beacon_tone = 2

set yaw_motors_reversed = ON

set small_angle = 180

set deadband = 2

set yaw_deadband = 2

set pid_process_denom = 1

set osd_rssi_alarm = 40

set osd_cap_alarm = 1300

set osd_tim1 = 1

set osd_tim2 = 0

set osd_vbat_pos = 449

set osd_rssi_pos = 2081

set osd_tim_1_pos = 2455

set osd_tim_2_pos = 33

set osd_flymode_pos = 333

set osd_throttle_pos = 225

set osd_vtx_channel_pos = 427

set osd_ah_pos = 200

set osd_current_pos = 2402

set osd_mah_drawn_pos = 2425

set osd_craft_name_pos = 2106

set osd_gps_speed_pos = 218

set osd_gps_lon_pos = 82

set osd_gps_lat_pos = 65

set osd_gps_sats_pos = 51

set osd_home_dir_pos = 302

set osd_home_dist_pos = 303

set osd_compass_bar_pos = 266

set osd_altitude_pos = 344

set osd_pid_roll_pos = 423

set osd_pid_pitch_pos = 455

set osd_pid_yaw_pos = 487

set osd_debug_pos = 1

set osd_power_pos = 321

set osd_pidrate_profile_pos = 345

set osd_warnings_pos = 2442

set osd_avg_cell_voltage_pos = 2434

set osd_pit_ang_pos = 257

set osd_rol_ang_pos = 289

set osd_battery_usage_pos = 330

set osd_disarmed_pos = 2411

set osd_nheading_pos = 311

set osd_nvario_pos = 279

set osd_esc_tmp_pos = 82

set osd_esc_rpm_pos = 83

set osd_core_temp_pos = 89

set osd_stat_tim_1 = ON

set osd_stat_max_dist = ON

set osd_stat_min_batt = OFF

set osd_stat_battery = ON

set osd_stat_max_curr = OFF

set osd_stat_used_mah = OFF

set osd_stat_max_alt = ON

set debug_mode = NOTCH

set vtx_band = 5

set vtx_channel = 8

set vtx_freq = 5917

set vcd_video_system = NTSC

# profile

profile 0

set dterm_lowpass_type = PT1

set dterm_lowpass2_hz = 200

set dterm_notch_hz = 0

set anti_gravity_gain = 3000

set setpoint_relax_ratio = 0

set dterm_setpoint_weight = 50

set iterm_rotation = ON

set throttle_boost = 50

set throttle_boost_cutoff = 40

set p_pitch = 40

set i_pitch = 45

set d_pitch = 23

set i_roll = 45

set d_roll = 23

set p_yaw = 65

# rateprofile

rateprofile 0

set roll_rc_rate = 167

set pitch_rc_rate = 170

set yaw_rc_rate = 167

set roll_expo = 3

set pitch_expo = 3

set yaw_expo = 3

set tpa_rate = 15

set tpa_breakpoint = 1050

# 
```

Wornie, 2018-05-23:
Can confirm DSHOT breaks with 850 on Matek F722, not just your friends board.

'# version
'# Betaflight / MATEKF722 (MKF7) 3.4.0 May 21 2018 / 02:43:24 (041362614) MSP API: 1.39
'# master
'set serialrx_provider = CRSF
'set motor_pwm_protocol = DSHOT1200

The ESCs will not initialise with DSHOT on any current build after 850.

Wornie, 2018-05-23:
If you're asking me, absolutely.

Not sure how to make a diff look nice on here so here's a pastebin. (https://pastebin.com/LTPXxJ5w)

nikolaswidell, 2018-05-23:
Im the "buddy" mentioned in the issue. 

With same diff 849 works. Even if I full chip erase and flash 850 (or newer) and only change default ONESHOT to any DSHOT the escs wont initialize (only first 3 beeps). On 849 DSHOT works. Multishot works on both.

DieHertz, 2018-05-23:
Thanks guys, I'll study it tomorrow first thing in the morning

codecae, 2018-05-23:
Can you try with this hex?

[betaflight_3.4.0_MATEKF722.zip](https://github.com/betaflight/betaflight/files/2031854/betaflight_3.4.0_MATEKF722.zip)

codecae, 2018-05-23:
[image]

jflyper, 2018-05-23:
WTH? I did the same ( checked out 04136261 and ran a build ), and it doesn't work.

codecae, 2018-05-23:
Not sure if this indicates a problem, but there are slight differences in the flash sizes between CI and an offline build:

CI Console Log:
```Building MATEKF722
Memory region         Used Size  Region Size  %age Used
        ITCM_RAM:       15304 B        16 KB     93.41%
      ITCM_FLASH:          0 GB        16 KB      0.00%
ITCM_FLASH_CONFIG:          0 GB        16 KB      0.00%
     ITCM_FLASH1:          0 GB       480 KB      0.00%
      AXIM_FLASH:       12426 B        16 KB     75.84%
AXIM_FLASH_CONFIG:          0 GB        16 KB      0.00%
     AXIM_FLASH1:      334752 B       480 KB     68.11%
        DTCM_RAM:       14972 B        64 KB     22.85%
           SRAM1:       46752 B       176 KB     25.94%
           SRAM2:          0 GB        16 KB      0.00%
       MEMORY_B1:          0 GB         0 GB      -nan%
   text	   data	    bss	    dec	    hex	filename
 336890	  10288	  51432	 398610	  61512	./obj/main/betaflight_MATEKF722.elf
```

Built by hand:
```Linking MATEKF722
Memory region         Used Size  Region Size  %age Used
        ITCM_RAM:       15304 B        16 KB     93.41%
      ITCM_FLASH:          0 GB        16 KB      0.00%
ITCM_FLASH_CONFIG:          0 GB        16 KB      0.00%
     ITCM_FLASH1:          0 GB       480 KB      0.00%
      AXIM_FLASH:       12426 B        16 KB     75.84%
AXIM_FLASH_CONFIG:          0 GB        16 KB      0.00%
     AXIM_FLASH1:      334792 B       480 KB     68.11%
        DTCM_RAM:       14972 B        64 KB     22.85%
           SRAM1:       46748 B       176 KB     25.94%
           SRAM2:          0 GB        16 KB      0.00%
       MEMORY_B1:          0 GB         0 GB      -nan%
   text    data     bss     dec     hex filename
 336930   10288   51428  398646   61536 ./obj/main/betaflight_MATEKF722.elf```

codecae, 2018-05-23:
AXIM_FLASH, DTCM, and SRAM1 are different

DieHertz, 2018-05-23:
They aren't obliged to be the same

codecae, 2018-05-23:
Wasn't sure if it played a part -- just throwing it out there.

mikeller, 2018-05-23:
I'd expect them to be the same when built with the same version of the toolchain. Or what else could cause them to differ?

DieHertz, 2018-05-23:
Different os could cause that, cross-compilation is not that easy

nikolaswidell, 2018-05-29:
Hello,

just an update: with build 876 the issue is still present. DSHOT not working, multishot works OK.

Matek F722

wesk-ms, 2018-06-01:
Not pushing anyone, but just curious if anyone is actively looking into this issue.  Issue is still present in build 889.  I was going to look into it myself, but didn't want to waste time if someone is already doing so.

Thanks!

mikeller, 2018-06-02:
Tested this with CI build 892, and Dshot works fine on a MATEKF722, with the config above.

@toiminto: Can you please test this?

jirif, 2018-06-06:
My custom build (unmodified master) from 30th May confirmed this issue and i reverted it to pre https://github.com/betaflight/betaflight/commit/041362614bac4aaed4c0e4cc163a1ba99139dfdc  , with today's everything is ok. Iam just curious what happened, not just call it an ghost in the machine :)

codecae, 2018-06-06:
Whoops!  Accidentally hit the close button on my phone.

nikolaswidell, 2018-06-07:
Flashed 906 yesterday and works like a charm!

DieHertz, 2018-06-07:
Don't like the fact that it appeared and gone away unnoticed.

mikeller, 2018-06-07:
:ghost:

mikeller, 2018-07-07:
@toiminto: Can you please confirm that this is working now with 3.4.0-RC5, and close this?

closed 2018-07-08 by mikeller