betaflight/betaflight #5958: Feature Request - Failsafe switch to enable GPS Rescue or Drop for 3.4.0

ninova66, 2018-05-24:
This is not a issue just a feature request please delete this if this is inappropriate, as this is my first time submitting anything on GitHub. 

I've been using the GPS Rescue feature for almost a month on 3.4.0 that Mike Cpbl and Diego Basch had been working on and I just installed Betaflight #858 this evening. To make a long story short I bought a defective CF Nano (Waiting for the RMA) and a defective XM+ back to back.  Needless to say a spare XM+ RX has been working well on my kwad for two weeks, but I have been hesitate to turn on the GPS-RESCUE due to all my recent RX loss of signal issues worried that it would try to rescue 10ft from my face if I had it enabled during a RX loss.

With that said I though it would be safer for a user to enable a failsafe mode via a switch. This way if you’re flying in your backyard, a small park in a dense urban area or just testing new equipment where you know you don’t need GPS-RESCUE that you could use a switch to set it to DROP. But if you know your going to do something sketchy then you can flick a switch disable DROP and enable the GPS-RESCUE failsafe feature. Just a though.

McGiverGim, 2018-05-24:
If I'm not wrong, the GPS-RESCUE has a parameter with a minimum distance from home. If the quad is at less distance than that, then it does not activate. I don't know if this is enough for you.

utkinpol, 2018-05-30:
Hi,

I just tried this feature on the rooster model with CL Racing F4S FC using Diego`s code build. Not sure if it is same codebase as this, so i decided to leave this comment in here, in case if this feature is still getting some work. 

It is not clear what GPS_HOME mode has in common with Diego`s GPS_RESCUE, or if it same one or not - but realistically i get it should work similar to the classic RTL function on mode FCs. 
But, with those small models in mind i think it needs more control - like, there is no baro on the model I use, and i think results were wildly inconsistent due to the issues with uncontrollable climb due to wrong reading of the altitude - not sure if code was looking at absolute altitude or the diff from current GPS reading - but i had configured minimal climb to 20m alt and drone jumped up at least 100m up.

From my experience with RTL on Ardupilot i think it would be really important to add a feature via ‘set’ variables to DISABLE any climb to pre-defined altitude upon initiation of this mode.

It is also important to maintain input from manual throttle input as an override over computed throttle value and alter computed throttle to keep current altitude as returned by both baro and the GPS. 
Ideally, in the conditions where one flies in the open field with RSSI not under min threshold it would allow to have drone returned back at the same pre-defined previous altitude as it was prior to init of this mode, as user can always add throttle to raise drone higher.
In case when RSSI is gone then it should be a FAILSAFE param set what altitude adjustment needs to happen, only this adjustment needs to be set as a diff from current previous alt - not a 'set' actual altitude value as this value on model with no baro is quite random. 

It also would be nice if it could be possible to configure a param switch into ‘GPS Hold’ mode when drone arrives within pre-defined landing descent circle instead of landing procedure auto-started, or if landing procedure then could be initiated by a separate control. as in a worst scenario with a good GPS lock if drone cannot return back it should at least 'stop' in the sky and stop moving away. if it flies a circle while doing it - it is also fine, as long as it does not fly sideways who knows where. 

Not sure what went wrong today – when i tested my vehicle – and i had checked i had good GPS lock with 3D and 10 sats – drone went ballistic up to 150m or so alt and then started venturing sideways somewhere. Upon turning off this mode controls were messed up but i managed to return it above the football field and drop it down.

utkinpol, 2018-05-30:
Also, not sure if there is any blog or user test board about the current 3.4 RC code base - i see those GPS modes added into config, but i presume nothing yet works and the only working code base is the Diego`s GPS_RESCUE one?

closed 2018-07-06 by stale[bot]

callagga, 2018-08-01:
Would seem to be a good idea. In some cases depending where you are flying you might not want to enable the gps rescue mode as the failsafe mechanism and turn it off....