betaflight/betaflight #5964: "Use DSHOT beacon" not saved after activation

Lord-Quake, 2018-05-24:
BF Build "betaflight_3.4.0_MATEKF405-859-71a1a9789"

If setting up BF for the first time the following problem occurs:

Activate "Use DSHOT beacon" but not selecting "Beacon Tone" (leaving tone at default "1")
"DSHOT Beacon Configuration" will stay none activated after save and reboot.

Workaround:
Select a different "Beacon Tone" and then go back to "1".
Then "Use DSHOT beacon" will stay active after save and reboot.

wind0r, 2018-05-24:
Do you use the nightly configurator? I think that but got fixed in the configurator

Lord-Quake, 2018-05-24:
No, I'm using "Configurator: 10.2.0"
I will search for the nightly version and report back.

Lord-Quake, 2018-05-24:
OK, installed it and the configurator seems to be working. I'm going to take your word for it else I would have to reset my setup. I have(had) another problem to solve which might have to do with me using v10.2 on nightly BF builds.
Thanks @wind0r 
Closing this for now.

closed 2018-05-24 by Lord-Quake