betaflight/betaflight #5969: While arming via stick motors fire up and disarm is not possible!

Lord-Quake, 2018-05-25:
Everyone,

I've been asked to post this issue as a separate post.

For the last 4 days I was plagued with a problem that I consider dangerous. I am posting this because it concerns the proposed default settings found here #5852 , however I don't know if the problem has to do with settings directly or the fault is a combination of factors.
I've tried several 3.4 builds (793, 849, 855) with the same results. Right now I'm running build 859. and using Configurator 10.3.0 build 202.
The installation is always the same. Delete the whole flash, install new FW, change my settings and then add #5852 mentioned proposed default settings.
I arm my QC via stick. When I move the stick to the right to arm the QC the motors immediately go up to speed. If the QC would have props it could potentially lift off and flip which I consider to be a very dangerous situation. Now if I try to disarm by moving the throttle stick to the left the motors will fire up again. The only way to disarm the QC is to turn off the transmitter so the QC enters fail safe mode.
**I'm no expert** but it seems like the min throttle turns to 1000 even though it is set to 1050. At least that is where I started to concentrate my search for the possible fault. Long story short.... the culprit turned at to be  "set rc_interp_ch = RPYT". However I don't know if it is the cause of the problem. Anyhow, when I configure "set rc_interp_ch = RP" the problem disappears. BTW, the problem also shows up when setting "set rc_interp_ch = RPY".
I am pointing this problem out because if someone like me had a working QC and did above mentioned updates they would get a bad surprise.

Please take a look at my DIFF (here from build 859) and look to see if a combination of factors can be found which would cause the unpleasant and dangerous situation.

`
# version
# Betaflight / MATEKF405 (MKF4) 3.4.0 May 24 2018 / 14:05:19 (71a1a9789) MSP API: 1.39

# name
name QC_3

# resources

# mixer

# servo

# servo mix

# feature
feature -SOFTSERIAL
feature -TELEMETRY
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE

# map
map TAER1234

# serial

# led

# color

# mode_color

# aux
aux 0 2 0 900 1300 0
aux 1 2 0 1700 2100 0
aux 2 3 0 1700 2100 0
aux 3 13 2 1700 2100 0
aux 4 19 1 1700 2100 0
aux 5 28 0 1300 1700 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 4 s 1000
rxfail 5 s 1000
rxfail 6 s 2000

# master
set gyro_lowpass_hz = 120
set gyro_lowpass2_hz = 300
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set acc_calibration = 33,70,-191
set mag_hardware = NONE
set rssi_channel = 12
set serialrx_provider = SUMD
set blackbox_p_ratio = 64
set motor_pwm_protocol = DSHOT600
set failsafe_off_delay = 50
set failsafe_throttle = 1200
set failsafe_procedure = AUTO-LAND
set bat_capacity = 1550
set beeper_dshot_beacon_tone = 4
set pid_process_denom = 1
set osd_warn_esc_fail = OFF
set osd_cap_alarm = 1500
set osd_alt_alarm = 200
set osd_vbat_pos = 2505
set osd_rssi_pos = 2081
set osd_tim_1_pos = 471
set osd_tim_2_pos = 2103
set osd_throttle_pos = 2087
set osd_ah_pos = 2126
set osd_current_pos = 2518
set osd_mah_drawn_pos = 2512
set osd_altitude_pos = 2094
set osd_warnings_pos = 2409
set osd_avg_cell_voltage_pos = 2497
set osd_disarmed_pos = 2346
set osd_stat_max_dist = ON
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_used_mah = OFF
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set debug_mode = NOTCH
set vcd_video_system = PAL

# profile
profile 0

set dterm_lowpass_type = PT1
set dterm_lowpass2_hz = 200
set dterm_notch_hz = 0
set anti_gravity_gain = 3000
set setpoint_relax_ratio = 0
set dterm_setpoint_weight = 50
set throttle_boost = 50
set throttle_boost_cutoff = 40
set p_pitch = 47
set i_pitch = 20
set d_pitch = 36
set p_roll = 45
set i_roll = 23
set d_roll = 33
set p_yaw = 72

# rateprofile
rateprofile 0

set roll_rc_rate = 90
set pitch_rc_rate = 90
set yaw_rc_rate = 90
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 20
set roll_srate = 82
set pitch_srate = 82
set yaw_srate = 82
set tpa_rate = 15
set tpa_breakpoint = 1050

#
 `

(set rc_interp_ch = RPYT is removed)

flint723, 2018-05-26:
By the way don't fly with those throttle boost settings without rc_interp_ch = RPYT, it is not kind to your escs/motors.  `set throttle_boost = 0` to disable.

No idea about the issue though, sorry.

flint723, 2018-05-26:
I've tried to reproduce this on a Matekf405 boad, I could not.  Even went as far as to copy your diff in (minus serial setup, and different rx protocol).  RPYT on or off, it stick armed and disarmed normally.

Lord-Quake, 2018-05-26:
OK, that is very interesting!

All my sticks have a range of 1000 - 2000.

I will try it again no and log it. I would guess the log should show my throttle positions when armed.

Lord-Quake, 2018-05-26:
I was able to reproduce it.
Log LOG00002.BFL has "rc_interp_ch = RPYT" set.
Log LOG00003.BFL has "rc_interp_ch = RP" set.
No other changes were made.
[LOGs_RPYT.zip](https://github.com/betaflight/betaflight/files/2041313/LOGs_RPYT.zip)

flint723, 2018-05-26:
Wow, I see what is happening, but I don't know why.  Yaw P is responding to yaw stick (despite rc command reading 0 in log) and driving motors 2 and 3 to 40%.  Looks like some type of bug.

Does it still behave this way if you keep all BF defaults (except what you need to function) and then turn on `rc_interp_ch = rpyt`?

Lord-Quake, 2018-05-26:
I entered "defaults" in the CLI, did min settings for function and added "rc_interp_ch = rpyt".
I was able to reproduce the problem.

diff
# version
# Betaflight / MATEKF405 (MKF4) 3.4.0 May 24 2018 / 14:05:19 (71a1a9789) MSP API: 1.39
# name
# resources
# mixer
# servo
# servo mix
# feature
# beeper
# map
map TAER1234
# serial
# led
# color
# mode_color
# aux
# adjrange
# rxrange
# vtx
# rxfail
# master
set acc_calibration = 33,70,-191
set mag_hardware = NONE
set rssi_channel = 12
set rc_interp_ch = RPYT
set serialrx_provider = SUMD
set motor_pwm_protocol = DSHOT600
set pid_process_denom = 1
# profile
profile 0
# rateprofile
rateprofile 0
# 

Here my log: 
[Defaults_RPYT.zip](https://github.com/betaflight/betaflight/files/2041360/Defaults_RPYT.zip)

flint723, 2018-05-26:
Ok I reproduced it as well after flashing new master 863 (was previously on 796).  I didn't change anything from BF defaults except serial Rx protocol, my port setup, motor resource swap, BB logging to internal flash, and `set rc_interp_ch = rpyt`.  Here is log and diff.  Someone who can look at code will have to check it out.
[stick arming issue matekf405.zip](https://github.com/betaflight/betaflight/files/2041366/stick.arming.issue.matekf405.zip)
[diff stick arming issue matekf405.txt](https://github.com/betaflight/betaflight/files/2041367/diff.stick.arming.issue.matekf405.txt)

Lord-Quake, 2018-05-26:
OK, at least I now know I'm not going crazy! :-)

I'd like to continue with my PID tuning. In your opinion would it be safe to use your recommended settings with the exception of setting = RT and throttle_boost = 0?

flint723, 2018-05-26:
Yep, perfectly fine without RPYT and throttle boost.  The only reason RPYT is turned on is to use throttle boost.

Lord-Quake, 2018-05-26:
Thanks! It's morning here with beautiful weather all day :-)

Let's hope this gets sorted ASAP!

flint723, 2018-05-26:
May be target specific, so far cannot reproduce on Omnibusf4 with the same procedure (flashing 863 and only changing the minimum settings to get the board to stick arm).

etracer65, 2018-05-26:
I just tried using the current master and wasn't able to replicate. This is with all defaults except serial and receiver.  Any advice how to replicate?

I see you're both running `map TAER` where I'm running `map AETR`

This is a SPRACINGF3EVO target.

flint723, 2018-05-26:
I'll try on a few other targets later tonight.  Only other thing I can add is it was still present with different motor protocols on the Matekf405.  I tried with more features off, didn't make a difference.  Tried with auto and manual 11ms interpolation, no difference.

etracer65, 2018-05-27:
The logic is designed so that if using stick arming (no arming mode set up on the Modes tab), then if throttle stick is below min_check then yaw PID's are zero'd.  So if you're seeing yaw P then we need to confirm that your throttle is getting below min_check.

The only other real exception is if your were using the AIRPLANE or FLYING_WING mixer which I assume you're not.

@flint723 If you have a board that can replicate, can you try to narrow down the Jenkins build that seems to start having the problem? I guess it's after 796 because you said that was ok.

etracer65, 2018-05-27:
I usually do a binary search by splitting the difference between one I know doesn't work and one that does.  Then depending on the result split again until you narrow down on the problem.

flint723, 2018-05-27:
I am able to confirm OMNIBUSF4 does have the issue as well.

flint723, 2018-05-27:
So far: 
-796 is fine
-801 is fine

flint723, 2018-05-27:
Verified on OMNIBUSF4, IMPULSERCF3, and MATEKF405 that 845 is ok, and 846+ has the issue.

etracer65, 2018-05-27:
Thanks to @flint723 identifying the build, I think I found the problem.  I'm still not able to replicate with a variety of flight controllers - even tried changing to map TAER1234 to match your setup.  However I think I see what might be happening and it appears be a timing issue in incoming RX data that may be explained by differences in protocols.  I run Frsky and can't reproduce but I think both of you are running Spektrum.

In the identified build 846 the only change was the fix I made for motor spikes with rc interpolation enabled for throttle #5905.  This shifted the ordering of how rc data was processed to prevent the mixer from seeing un-smoothed data.  I can see that there's some logic related to throttle and yaw reset that probably should have moved as well (although it really shouldn't matter).  So it seems like a timing related issue - probably explaining the cases of some FC's having the problem and others not.

I made a fix for you to test since I can't replicate.

[betaflight_3.4.0_MATEKF405.hex.zip](https://github.com/betaflight/betaflight/files/2042591/betaflight_3.4.0_MATEKF405.hex.zip)

This is built from latest master.  Let me know if you need other targets to test and I'll build.  Once we confirm that this fixes it I'll submit a PR.

etracer65, 2018-05-27:
I was finally able to get a combination that duplicated the problem so I can confirm it's fixed.  It only affected cases where `rc_interp_ch = RPYT`. In reality stick command disarming was working.  The problem was that yaw input to the PID controller wasn't being suppressed like it should when throttle was below min_check so it would cause the quad to react to the yaw.  If you held the yaw it would disarm, but I can understand why you wouldn't think so.

I'm not surprised it wasn't noticed until now. Very few people use stick arming anymore and the use of `rc_interp_ch = RPYT` has been very rare as well up to this point.

Created https://github.com/betaflight/betaflight/pull/5992

flint723, 2018-05-27:
@etracer65 so in your tests `rc_interp_ch = RPY` did not have the issue but RPYT did?

And in the last batch of testing I realized the same, that it did indeed stick disarm as well, probably did not hold it quite long enough in initial testing.

etracer65, 2018-05-27:
@flint723 No, RPY has the issue as well. I just skipped over that and went right to RPYT to duplicate.

I definitely understand the problem now and the fix will resolve it.  The cause was due to the ordering change in the processing to fix the motor spikes.  Before the logic was zeroing out the yaw rcCommand channel *before* interpolation and then the PID controller would use that value on the next loop.  Now the interpolation is happening *before* the PID controller but the zeroing was still after.  Without interpolation the rcCommand data exactly matches the actual RX frames from the receiver but with interpolation enabled for a channel the rcCommand values come from an internal buffer that's generating the smoothing.  So with the zeroing at the end of the chain it was okay without interpolation because the actual rcCommand data was overritten with zero.  But with interpolation even though rcCommand was being zeroed the next pass through got the value from the internal buffer and the PID controller used that before it was again zeroed.  So that's why you didn't see any traces for rcCommand but the PID controller responded to yaw.  In fact that was the entire problem - not that disarming didn't work but rather that yaw wasn't being suppressed below min_check. So a moderately complicated timing issue but definitely now understood and solved.

flint723, 2018-05-27:
Confirmed it is fixed with 98fb4d7c6 on omnibusf4, impulsercf3, matekf405.  Any need to test the later commit?

etracer65, 2018-05-27:
No. That was just to address fast ram overflow on F7's.

flint723, 2018-05-27:
Sorry, I made a mistake and edited, I did test the last commit, but had flashed one of my boards back to 863 when I went to look what commit number I tested.

etracer65, 2018-05-27:
Actually the fact that this was really not a disarming issue and rather a yaw suppression problem is why I had trouble duplicating.  I was initially just testing bare boards on the bench and was confirming that the arming led would go back off to indicate it worked.  I wasn't testing in the configurator motors tab or with a real quad so I couldn't see the motor response to yaw.

flint723, 2018-05-27:
Yes, I should have shown images in configurator, it was visible in the log however.

PR #5992 (2018-05-27): Fix yaw reset and throttle angle correction to be before rc interpolation
Fixes #5969 

In #5905 the ordering of rc interpolation processing was moved out of `subTaskMainSubprocesses()` and into a separate `subTaskRcCommand()` process that ran earlier in the PID loop. Previously the logic to handle stick arming yaw reset was immediately before the rc interpolation logic and after the previous change it ran later in the process and after rc interpolation.

This fix moves the related throttle/yaw processing into the `subTaskRcCommand()` before rc interpolation to match the ordering that existed  [... cut]
closed 2018-05-27 by mikeller