betaflight/betaflight #5972: Port failsafe_recovery_delay from Inav for gps_rescue in Betaflight

ChilledoutUK, 2018-05-26:
I have been testing the Gps_rescue mode out and recently had a major issue with an intermittent radio link interrupting the Rescue mode which in it self works very well.

Here is a video showing how what happened when it kept switching Failsafe on and off.
https://www.youtube.com/watch?v=EWCF6x3sjU8

Here is the Blackbox log from the flight
https://drive.google.com/file/d/1QIBjHBlia8WjW10zp5I80izF6bTtCE4U/view?usp=sharing

From Inav wiki:-
failsafe_recovery_delay
Guard time for failsafe de-activation after signal is recovered. This is the amount of time the flight controller waits to see if the signal is consistent before turning off failsafe procedure. Usefull to avoid swithing in and out of failsafe RTH. Does not apply when disactivating the FAILSAFE aux mode.

ChilledoutUK, 2018-06-03:
Gps rescue works great but the current failsafe settings intermittant signal will keep resetting it i have recovered my drone from the sea and got it working again so i can test if someone wants to port this code.

s0up, 2018-06-05:
@ChilledoutUK Just out of curiosity, does rescue engage properly as a failsafe method?  It is very hard to test the failsafe code so it would be nice to hear if it actually does in fact engage properly.

ChilledoutUK, 2018-06-06:
On the build from 12 may 2018 from your gps_rescue github site the mode does indeed engage fine I tested it the other day by turning my transmitter off I however wasn't able to rearm the drone in time before landing pretty damn hard. (knocked my camera loose)

The only issue i had at range was that it seemed the failsafe was getting interrupted by the radio link hence the need for a guard time or something similar.

If the gps_rescue returns to a loiter then a guard time would make it function perfectly as that way if you set it too long you can just wait whilst it loiters to regain control.

ChilledoutUK, 2018-06-10:
Oh dear you guys going to release gps_rescue in 3.4 without this necessary feature? 

People will loose drones and get very upset if you don't put this in 3.4

s0up, 2018-06-10:
Honestly, coming from a background of flying inav a lot, it goes both
ways.  Fail-Safe recovery delay can actually make it hard to regain control
of craft with an intermittent link when you regain control and try to drop
altitude and then Fail-Safe brings it back up.  I agree we should make this
a setting so the burden is on the user to decide for themselves.  It is
unwise to have rth as a failsafe but not have a switch as well.

This feature is going to be used more by advanced users and anybody
familiar with RTH knows there are inherent risks.  Updates will be coming
to make this better and better, but there is a huge divergence in Fail-Safe
code between inav and betaflight, so it isn't as simple as just copy
pasting it over.

If you feel this is something you'd like to take on, it would be awesome to
see someone pull request this 😀

ChilledoutUK, 2018-06-10:
Regardless whether gps_resue is on a switch the failsafe guard time of the radio link combined with betaflight means In fringe reception gps_rescue has little to no chance at working reliably. 

If gps_rescue returns to loiter then there is no reason to need to regain control before the guard time is up.

s0up, 2018-06-10:
If you Fail-Safe, you should also flip the switch to enable the mode so it
doesn't flip flop.  This has already gone through fairly extensive testing
in horrible RF environments (San Fransisco) and  it has been fine when used
correctly by the pilot.  This is a good idea even when using things like
inav with a Fail-Safe recovery delay because something as simple as a trim
control being one click to the side or touching the stick accidentally will
negate the delay.

This is likely something to be added, but to imply that it is entirely
broken as-is I would disagree.   RTH without a barometer or compass is and
will likely always be a risk that is meant to simply increase your odds of
recovery in an emergency situation, not something to be leaned on.  That is
why we called it rescue mode and not RTH.  The idea is to not imply this is
a DJI style return to home method that should be heavily relied on.

ChilledoutUK, 2018-06-11:
OK i see what your saying i think what my primary issue was is that i lost video feed when the quad turned around in rescue mode and i had no information so didn't know that it was going in and out of rescue mode.
I think in future I will make sure the antenna is better oriented and train myself to manually switch recovery mode on via the switch as soon as I notice a failsafe as you suggest.

I Assume rescue mode is ok with being reinitialized over and over until signal is stable?
or would it be better to have a failsafe switch instead so that the quad is kept in gps_rescue via failsafe and not repetitively reinitialized?

How about an option to just force the mode to stay active until the descent distance is reached?

s0up, 2018-06-11:
It actually won't toggle on and off.  If the switch matches the mode it
won't re initialize multiple times.  I think the best solution is to
eventually add a recovery delay optionally and make stick movements cancel
it probably 😎

ChilledoutUK, 2018-06-11:
Oh that's great to know.
Out of interest will rescue mode try to land or just loiter if you don't disable it when its back in range?

s0up, 2018-06-11:
It will loiter at an increasingly lower altitude until it touches down.  It
will attempt to disarm if it senses a bump (landing) or if groundspeed hits
close to zero for 5 seconds or so.

ChilledoutUK, 2018-06-11:
Wow that's really cool. Is this something added recently or does the 12 may build also do this?

s0up, 2018-06-11:
All builds should attempt to do that.  The bump detection logic may need
some fine tuning because that kind of stuff is hard to test while
developing.

ChilledoutUK, 2018-06-11:
Do you think it is advisable to turn stage 1 of the failsafe off if your receiver already has a guard time?Because as it seems right now if you loose RF link in Air mode it will center the channels and put throttle low for 400ms before enabling gps_rescue via failsafe.
If your flying low this would probably be enough for you to hit the ground.

Sm1thson, 2018-06-13:
>  RTH without a barometer or compass is and
will likely always be a risk that is meant to simply increase your odds of
recovery in an emergency situation, not something to be leaned on

@s0up Firstly, I fully agree. Secondly, I have a BF'd quad with GPS, Barometer and compass, in the new recovery mode will the barometer and compass be of any additional benefit of have they been made redundant? (I ask as I will likely add GPS to another quad, but if the compass and barometer aren't needed that will simplify things). 

As a side note, wheres the best place for the general discussing of the GPS / recovery mode configuring and stuff? I've sen the Betanav channel on FPVchat slack but with al the other channels been so busy it gets flushed too often. (I see github as been more the place for discussing the code related stuff)

Bakerfpv, 2018-06-20:
I've flashed betaflight 3.4 rc1 and enabled rescue mode on a switch. How do I set it to rescue mode when it failsafes. (I'm using tbs crossfire with v2 receiver which is set to cut) I've test the rescue mode with a switch which works fine. However when I switch my taranis off it cuts power. I then set stage 1 failsafe to keep my rescue and arm switch on but it fails after half a second. I think it ia going to stage 2 failsafe after 0.4 seconds and cutting everything. How can I set it up to get me home?

ChilledoutUK, 2018-06-20:
Have you set failsafe in the cli to gps_rescue instead of drop?

Bakerfpv, 2018-06-20:
How do you do that? I did try on code in cli but it didn't seem to do anything. It was 
set failsafe_procedure = GPS-RESCUE

ChilledoutUK, 2018-06-20:
did you then save?
you can test this with props off just arm it props off in the garden then turn off your transmitter and if should go into failsafe and show Resc in the osd flight mode.
make sure you have 10 sats or more

s0up, 2018-06-20:
Rescue needs to be over 100 meters from home to engage and will also be
prevented from triggering via the min throttle check in the faailsafe
logic.

Bakerfpv, 2018-06-21:
I'm sure I did press save. Can you confirm I have the correct line of code to set failsafe as rescue?

Also the rescue works when I trigger it from my switch when I tested it out in my garden. What is the best way to check failsafe works with actually flying it. I don't want to crash my quad.

I might fly it over 100m from 'home' and switch off my transmitter when I'm just off the ground. Either that or arm it in my car and drive up a hill 100m away and switch it off. I will test again tomorrow. Thanks

Bakerfpv, 2018-06-21:
Additional to above can you explain what you mean by min throttle check? Will failsafe/rescue not work at min throttle or doing a min throttle dive?

Bakerfpv, 2018-06-21:
Update:

Thanks for all your help guys. I have entered the cli code pressed save and selected stage 2 on the drop down menu on the failsafe tab. It now comes up 100 fs 100 in my goggles so assume it is entering into fs mode. I will test this later. I have also set my arm switch to work when its back so I can test it by turning off my taranis. It will stay armed when inturn it back on then. Thanks again

ChilledoutUK, 2018-06-21:
Sadly to test failsafe turning the transmitter off is the only way but if you try to rearm the drone you might not be able to before it hits the ground from the rth height.
Your probably better off just taking the props off arm the quad take it 100m and then turn of transmitter and it should show resc in the osd plus beep the failsafe beep if its enabling properly.

ChilledoutUK, 2019-02-04:
I think this needs to be done before 4.0 released guys as it is now switching gps-rescue via a switch does stop the quad falling out of the air but it also causes the quad to gain a lot of altitude taking it way above 400ft legal limit.

This is because gps-rescue keeps re-initializing and climbing over and over until a solid signal is obtained.

mikeller, 2019-02-05:
@ChilledoutUK: Feel free to open a pull request to implement it if you think that this needs to be done.

ChilledoutUK, 2019-02-07:
Pull request for what I haven't written any code for this has anyone else?

If i could write the code for this i would but i am not a programmer.

This is not a special feature that i am requesting but rather something very important to make sure gps-rescue is even viable as a failsafe option.

Right now failsafe will kick in spind the drone arround you probably will loose video and it will then proceed to yoyo up and down as it comes into and out of radio reception.

The solution at the moment is to rush to enable gps rescues on a switch so that it can instead gain about 200m of altitude before gaining a stable rf link.

ChinaMan28, 2019-02-12:
@ChilledoutUK dude, guy did this shit for free, to the best that he can. people have lives. learn some programming, help fix the problem. everything is a special feature when you are doing work for free.

closed 2026-09-26 by blckmn