betaflight/betaflight #5982: crash prevention feature burns motors?

m013, 2018-05-26:
as far as i know , mpu6000 accelerometer is affected by drift. 
i had a nasty crash yesterday , when during rapid ascend i hit a branch and quad started to spin and lost control. all i could do is to disarm. it fell off a tree with motors stopped.

gyro overflow was set to yaw at that time. 

one of the motors got burned , although esc were fine.

so, could it be that my quad tried to stabilize (but failed because of acc accumulated error), and the  motor was commanded to spin so fast that it got damaged?

unfortunately i cannot provide any logs
[Leopard_btf.zip](https://github.com/betaflight/betaflight/files/2041670/Leopard_btf.zip)

etracer65, 2018-05-26:
The MPU6000 gyro is not affected by overflow - only ICM gyros need gyro_overflow_detection (this is auto-configured in BF3.4).

The flight controller can't command the motors to spin beyond 100% throttle.  So if the motors can't take that they would be damaging themselves in flight all the time.  It's most likely that you damaged the motor when hitting the branch and that was part of the reason for the high-speed spin - the motor wasn't producing thrust and destabilized the quad.

It only takes a fraction of a second for a motor stuck in an unlucky position to burn out.

m013, 2018-05-26:
@etracer65,  so,  it s like when you jam a motor but at the same time pushing throttle up?
The motor is spinning well, considering it is shorted.
Maybe dterm went mad or smth...
I wonder why blheli32 current limiting feature didnt prevent it.
I ll try to lower this value to something like 30amps (had it set  40a or so, the motor was rated up to 38a)

mikeller, 2018-05-26:
@m013: The current limiting feature may be helpful in situations where the current limit is approached slowly. But in a situation where a loaded and spinning motor is suddenly stopped, the current increase caused by this will be way too fast for any kind of limiting to be effective before damage is done.
Or to put it into another perspective: In order for motors to be able to accelerate as fast as possible, the protection must be designed in a way to allow fast increases in current, which is fine if the current is used to accelerate the motor. If the motor is blocked, the current will be dissipated as heat instead, causing the motor to burn up.

m013, 2018-05-27:
I guees i d better disable this feature on every quad i have. 
noticed today when i had a crash,  a quad just stopped,  or bounced back without these funny uncontrollable spins while trying to get itself leveled. 
Maybe crash prevention is usable for woops, but not  for outdoors quads, not even for 100mm ones.

mikeller, 2018-05-27:
@m013: Not sure what feature you are talking about, you seem to be a bit confused about what the individual features do. But there is no such feature that would result in 'a quad just stopped' in mid-flight (i.e. not right when taking off), you probably just had enough of an impact to cause (momentary) loss of power.

watageek, 2018-06-02:
I am more in doubt you smoked a motor running it with damage prop or "forcing" it. 

You have deactivated the notch filter
activated PT1 and dynamic filter and stage2 filter at 150hz, this can be the culprit.

No blackbox, version of firmware and lack of detail on your build will lead to guessing.

you can close this issue since it is not really an issue...

closed 2018-06-02 by m013