betaflight/betaflight #6000: WS2812b RGB - Protocol Change

ingleburn, 2018-05-28:
Pls can you confirm if beta flight has adopted the new protocol standard for WS2812b RGBs. The reset interval has been increased from 50us to 280us. I note that the old chips are no longer being manufactured we are not able to source anymore. To add to the confusion, they are still called the same! 

My understanding is that the old chips will work just great with the 280us reset low but not visa versa.

Pls see attached notice from worldsemi.
[image]

mikeller, 2018-05-29:
Betaflight only sends one LED_STRIP data frame every 10 ms:

https://github.com/betaflight/betaflight/blob/b2f7aa3c363dc645fbf854fc10f0cff12604e635/src/main/fc/fc_tasks.c#L522

With the maximum of 32 LEDs, the length of the frame will be 32 * 24 * 1600ns (~ 1.2 ms), which puts the remaining 8800 us of reset time comfortably above the 280 us limit.

mikeller, 2018-06-02:
@ingleburn: Does this address your concerns? If so, can you please close this?

mgbolts, 2018-06-04:
Hold off, I am doing some testing.  Results are not clear yet....

closed 2018-07-07 by mikeller