betaflight/betaflight #6006: Blackbox Explorer Incorrectly Reporting RC Command deg/s and Causes PID Error to be Incorrect

Mewogoginspin, 2018-05-29:
In the blackbox explorer, when displaying RC command, the calculated deg/s is incorrect. My max rates are 515 deg/s for roll and at full stick deflection the rcCommand displays 293 deg/s which causes the PID error to display ~200 deg/s even though when viewing the PID I see minimal activity on the roll axis and the Gyro is reporting 504 deg/s which would mean the error is much smaller. 

[image]

McGiverGim, 2018-05-29:
Please test the master version of blackbox you can get it from here andwho.sytes.net:8080 I think this problem is solved

McGiverGim, 2018-05-29:
I forgot to say, if the problem persists open the issue in the blackbox github and not the firmware github https://github.com/betaflight/blackbox-log-viewer

flint723, 2018-05-31:
By the way for me the problem (as I originally reported it in BB issues) is no longer there with latest master (standalone BBE not chrome app version).

closed 2018-07-07 by stale[bot]