betaflight/betaflight #6007: Latest build (from 1ab7b7816b9bdc54ad47298537555913654cb5d4) SDCARD blackbox logging not working

cjbeck, 2018-05-30:
I am fairly certain there is something new with the blackbox logging on my FC (TARGET=MLTEMPF4), which isn't working. I have tried 3 different sd cards, 2 of which are no longer working (which were previously working), and the third looks like it logs, but never actually writes the log file.

I flashed "a different fork" of betaflight, and the logging is working on that version. I am happy to try other versions of betaflight on this flight controller. On a different build, I am running 3.2.3, and the blackbox logging is working on that version.

Config block for version that doesn't work:
```
➜  Downloads egrep 'sdcard|blackbox|debug' BTFL_cli_KONASTY_20180529_114703.txt
set blackbox_p_ratio = 128
set blackbox_device = SDCARD
set blackbox_record_acc = ON
set blackbox_mode = NORMAL
set sdcard_dma = OFF
set osd_debug_pos = 234
set debug_mode = NOTCH
```

Config block for a version that does work:
```
➜  Downloads egrep 'sdcard|blackbox|debug' /Users/cjb/Documents/BTFL_cli_EPICV2_20180420_215449.txt
set blackbox_p_ratio = 32
set blackbox_device = SDCARD
set blackbox_on_motor_test = OFF
set blackbox_record_acc = ON
set sdcard_dma = OFF
set osd_debug_pos = 1
set debug_mode = NOTCH
```

McGiverGim, 2018-05-30:
Maybe is a too high blackbox_p_ratio? Use 32 as in the working version to see if it works.

mikeller, 2018-07-07:
@cjbeck: Can you please confirm that this is working now, and if so, close this?

cjbeck, 2018-07-12:
I'm going to close now since I don't have that same FC in the frame anymore. I will reopen if needed.

closed 2018-07-12 by cjbeck