betaflight/betaflight #6011: Own Firmware file for HGLRC FCs

naimadfpv, 2018-05-30:
Hey Guys,

is it possible to do an own Firmware hex file for HGLRC? 

Many People have problems flashing the HGLRC Boards with the correct firmware file (Omnibus F4).

Im thinking about the easy way to do this, just rename the Omnibus F4 hex file to HGLRCF4 or something like that and add it to Betaflight GUI so people can choose HGLRC FW instead of Omnibus F4.

Thanks in advice.

[betaflight_3.3.2_HGLRCF4.zip](https://github.com/betaflight/betaflight/files/2054555/betaflight_3.3.2_HGLRCF4.zip)

naimadfpv, 2018-05-30:
They choose the wrong hex file. 

They have to flash "OmnibusF4" hex file instead of "OmnibusF4FW", "Omnibus F4SD" or something other.

Many People ask daily on Facebook HGLRC Group which Firmware they should use.

It is way easier if the GUI has HGLRCF4 Firmware in the Dropdown to choose.

wind0r, 2018-05-30:
As allready told you plenty of times this will not happen.
Creating Targets just for the name would cost us way to much for the CI Infrastructure for not any sane reason.

Travis takes already around 30 Minutes to build. We need to wait that time for every pull request.

A Travis Licence that would allow such Targets with Custom names would cost $489 per month. If HGLRC and other Companys would donate that on a monthly base it maybe would feasible to do. Atleast for the CI Part. Jenkins and the Development overhead also is there.

Lucklily it is planed to just build one Target per Archtitekture, so all those "problems" are gone.

I think there is no reason to keep this issue open since all this question got answerd multply times. i will close it.

PS: what confuses me the most is that it looks like companys dont understand how targets are build or git works ;)

closed 2018-05-30 by wind0r

naimadfpv, 2018-05-30:
Okay thank you for your repy. 

I will talk that to HGLRC.

wind0r, 2018-05-30:
@naimadfpv you arent from HGLRC?
The last few days they pmed us through different ways and also created some strange pull request without any information. #6003

naimadfpv, 2018-05-30:
Not directly, I’m a TeamPilot and they ask me to ask you to do that.

That’s all I know.

naimadfpv, 2018-05-30:
But you understand me/us right? I’m not a developer, just a dump Pilot as I told :D 
We don’t want a new CI. 

We thought about just to copy the OmnibusF4, rename it to HGLRCF4 and implement that into betaflight GUI. So the inside of the hex file is the same like OmnibusF4. 

Just saying because I don’t know if you understand me right...

wind0r, 2018-05-30:
As written above. That will not happen. You allways need to compile the target for the name. So more ci resources are needed just for some name.

naimadfpv, 2018-05-30:
Okay i understand that. 

Sorry for double asking. I didn’t know that they had ask you in the past....

blckmn, 2018-05-30:
@naimadfpv it’s a common issue agreed, however when a manufacturer wants their branding in the configurator they should be prepared to donate and assist the Betaflight project.

Make no mistake it’s about branding. It’s not hard to simply post in a Facebook group or on a help page which firmware to use. They can even provide their own hex for download and flashing.

Soon the number of targets will go, and we will base it on architecture only, creating generic targets.

naimadfpv, 2018-05-31:
You’re right! I thought about that too. But I think if the user knows how to save their config in the log with „diff“ command or ask „version“ I think they are also able to know which firmware they should flash imo.

We are thinking about the newer users, which don’t know much about betaflight. 

I mean, sure they can ask on Facebook and I think mostly they will get an answer which Firmware they should use. 

You told me how much work and costs it takes to do that and I understand and accept it. 
I thought it would be easier....