betaflight/betaflight #6015: "Get" do not show the real settings

Vidalcris, 2018-05-31:
Hi,
I got a weird problem with my OmnibusF4 (HGLRC F440).
When i do a get on some settings the result is false, i mean this is not the actual setting.

But a "Diff" show the actual setting.

This is not from this build (6c9ec2625), i got the problem with "dterm_lowpass_type" for long.

I tried with another F440 and also with a F428 the problem is always the same.
I'm always flashing with a fullerase.

Thanks in advance :)

```
Diff
# Betaflight / OMNIBUSF4 (OBF4) 3.4.0 May 31 2018 / 14:05:10 (6c9ec2625) MSP API: 1.39
[...]
# profile
profile 1
set dterm_lowpass_type = PT1
set iterm_rotation = ON
set smart_feedforward = ON
set iterm_relax = RP

# get smart_f
smart_feedforward = OFF
Allowed values: OFF, ON

# get iterm_relax
iterm_relax = OFF
Allowed values: OFF, RP, RPY

# get dterm_lowp
dterm_lowpass_type = BIQUAD
Allowed values: PT1, BIQUAD

DieHertz, 2018-05-31:
I think you're on a different profile than 1

Vidalcris, 2018-05-31:
Damn ... so this is normal ? :)

DieHertz, 2018-05-31:
Type `profile 1` and try `get` commands again

Vidalcris, 2018-05-31:
same : 
diff
[...]
set smart_feedforward = ON

# profile 1
profile 1

# get smart
smart_feedforward = OFF
Allowed values: OFF, ON

Vidalcris, 2018-05-31:
Even if i change the profile with the PIDs page

DieHertz, 2018-05-31:
What does `profile` say? Weird indeed

Vidalcris, 2018-05-31:
```
#profile 
profile 0

Vidalcris, 2018-05-31:
oh btw if i type profile 1, then profile the result is profile 1 ;)

Vidalcris, 2018-05-31:
when i restart the board the result is profile 0

Vidalcris, 2018-05-31:
So yeah in fact from start i'm in profile 0 ;)

Vidalcris, 2018-05-31:
Ok the problem is my diff ... idk which part ..

```
# Betaflight / OMNIBUSF4 (OBF4) 3.4.0 May 29 2018 / 08:08:58 () MSP API: 1.39

# name
name BH130-Crisssss

# resources
resource MOTOR 1 B01
resource MOTOR 2 A02
resource MOTOR 3 B00
resource MOTOR 4 A03
resource SERIAL_TX 3 NONE
resource SERIAL_TX 11 B14

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature SOFTSERIAL
feature LED_STRIP
feature AIRMODE
feature ESC_SENSOR
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -SYSTEM_INIT
beeper -ON_USB

# map

# serial
serial 0 64 115200 57600 0 115200
serial 2 1024 115200 57600 0 115200
serial 5 16384 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200

# led
led 0 7,8::A:13
led 1 8,8::A:13

# color

# mode_color
mode_color 6 0 4
mode_color 6 1 13

# aux
aux 0 0 0 1700 2100 0
aux 1 13 2 1700 2100 0
aux 2 26 1 1700 2100 0
aux 3 35 2 1300 1700 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 3 h
rxfail 6 s 1000

# master
set gyro_lowpass_hz = 100
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 215
set gyro_notch2_hz = 0
set gyro_notch2_cutoff = 150
set gyro_overflow_detect = YAW
set acc_hardware = NONE
set acc_calibration = 49,148,-124
set mag_hardware = NONE
set baro_hardware = NONE
set rc_interp = MANUAL
set rc_interp_ch = RPYT
set rc_interp_int = 9
set fpv_mix_degrees = 45
set max_aux_channels = 4
set serialrx_provider = IBUS
set dshot_idle_value = 600
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT1200
set failsafe_delay = 10
set align_board_yaw = 90
set vbat_min_cell_voltage = 32
set vbat_warning_cell_voltage = 33
set current_meter = NONE
set small_angle = 180
set pid_process_denom = 1
set tlm_inverted = ON
set osd_tim_2_pos = 2416
set osd_vtx_channel_pos = 2071
set osd_craft_name_pos = 2048
set osd_pidrate_profile_pos = 2105
set osd_avg_cell_voltage_pos = 2408
set osd_disarmed_pos = 2347
set osd_esc_tmp_pos = 2065
set osd_esc_rpm_pos = 2444
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set debug_mode = ITERM_RELAX
set vtx_band = 5
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5658

# profile
profile 1

set dterm_lowpass_type = PT1
set dterm_lowpass_hz = 90
set dterm_notch_hz = 0
set vbat_pid_gain = ON
set anti_gravity_gain = 4000
set dterm_setpoint_weight = 100
set smart_feedforward = ON
set iterm_relax = RP
set p_pitch = 28
set i_pitch = 40
set d_pitch = 16
set p_roll = 24
set i_roll = 35
set d_roll = 16
set p_yaw = 30

# rateprofile
rateprofile 0

set roll_srate = 50
set pitch_srate = 50
set yaw_srate = 50
set tpa_rate = 15

Vidalcris, 2018-05-31:
If i dont paste my diff in the CLI everything is working fine.
After using my diff then saving, the problem happen.

Vidalcris, 2018-05-31:
This is the only part that i'm not sure about : 
```
# resource SERIAL_TX 11 B14

NOTE: B14 already assigned to PPM.

NOTE: B14 already assigned to PWM 1.

NOTE: B14 already assigned to ESCSERIAL.

Resource is set to B14

DieHertz, 2018-05-31:
Thank you, I'll try to reproduce and study once I get home.
@Vidalcris AFAIK you don't have to free a resource unless the other assignment may kick in, e.g. if you use PPM/PWM receiver or ESCSERIAL.

Vidalcris, 2018-05-31:
i use esc telemetry .. not sure if this is related to escserial

Vidalcris, 2018-05-31:
Also i freed SERIAL_TX 3 .. idk if this is really necessary with my config.

[image]

Vidalcris, 2018-05-31:
If i change my diff with profile 0 instead of profile 1, everything is working fine.
This is the first time is use another profile because i use this model with 3S & 4S.

```
# # Betaflight / OMNIBUSF4 (OBF4) 3.4.0 May 29 2018 / 08:08:58 () MSP API: 1.39

# # name

# name BH130-Crisssss
name BH130-Crisssss

# # resources

# resource MOTOR 1 B01

NOTE: B01 already assigned to MOTOR 2. MOTOR 2 disabled.

Resource is set to B01

# resource MOTOR 2 A02

NOTE: A02 already assigned to MOTOR 4. MOTOR 4 disabled.

Resource is set to A02

# resource MOTOR 3 B00

Resource is set to B00

# resource MOTOR 4 A03

Resource is set to A03

# resource PPM 1 NONE
Resource is freed

# resource PWM 1 NONE
Resource is freed

# resource SERIAL_TX 3 NONE
Resource is freed

# resource ESCSERIAL 1 NONE
Resource is freed

# resource SERIAL_TX 11 B14

Resource is set to B14

# # mixer

# # servo

# # servo mix

# # feature

# feature -RX_PARALLEL_PWM
Disabled RX_PARALLEL_PWM

# feature RX_SERIAL
Enabled RX_SERIAL

# feature SOFTSERIAL
Enabled SOFTSERIAL

# feature LED_STRIP
Enabled LED_STRIP

# feature AIRMODE
Enabled AIRMODE

# feature ESC_SENSOR
Enabled ESC_SENSOR

# feature ANTI_GRAVITY
Enabled ANTI_GRAVITY

# feature DYNAMIC_FILTER
Enabled DYNAMIC_FILTER

# # beeper

# beeper -SYSTEM_INIT
Disabled SYSTEM_INIT

# beeper -ON_USB
Disabled ON_USB

# # map

# # serial

# serial 0 64 115200 57600 0 115200
serial 0 64 115200 57600 0 115200

# serial 2 1024 115200 57600 0 115200
serial 2 1024 115200 57600 0 115200

# serial 5 16384 115200 57600 0 115200
serial 5 16384 115200 57600 0 115200

# serial 30 2048 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200

# # led

# led 0 7,8::A:13
led 0 7,8::A:13

# led 1 8,8::A:13
led 1 8,8::A:13

# # color

# # mode_color

# mode_color 6 0 4
mode_color 6 0 4

# mode_color 6 1 13
mode_color 6 1 13

# # aux

# aux 0 0 0 1700 2100 0
aux 0 0 0 1700 2100 0

# aux 1 13 2 1700 2100 0
aux 1 16 2 1700 2100 0

# aux 2 26 1 1700 2100 0
aux 2 28 1 1700 2100 0

# aux 3 35 2 1300 1700 0
aux 3 36 2 1300 1700 0

# # adjrange

# # rxrange

# # vtx

# # rxfail

# rxfail 3 h
rxfail 3 h

# rxfail 6 s 1000
rxfail 6 s 1000

# # master

# set gyro_lowpass_hz = 100
gyro_lowpass_hz set to 100
# set gyro_notch1_hz = 0
gyro_notch1_hz set to 0
# set gyro_notch1_cutoff = 215
gyro_notch1_cutoff set to 215
# set gyro_notch2_hz = 0
gyro_notch2_hz set to 0
# set gyro_notch2_cutoff = 150
gyro_notch2_cutoff set to 150
# set gyro_overflow_detect = YAW
gyro_overflow_detect set to YAW
# set acc_hardware = NONE
acc_hardware set to NONE
# set acc_calibration = 49,148,-124
acc_calibration set to 49,148,-124
# set mag_hardware = NONE
mag_hardware set to NONE
# set baro_hardware = NONE
baro_hardware set to NONE
# set rc_interp = MANUAL
rc_interp set to MANUAL
# set rc_interp_ch = RPYT
rc_interp_ch set to RPYT
# set rc_interp_int = 9
rc_interp_int set to 9
# set fpv_mix_degrees = 45
fpv_mix_degrees set to 45
# set max_aux_channels = 4
max_aux_channels set to 4
# set serialrx_provider = IBUS
serialrx_provider set to IBUS
# set dshot_idle_value = 600
dshot_idle_value set to 600
# set use_unsynced_pwm = OFF
use_unsynced_pwm set to OFF
# set motor_pwm_protocol = DSHOT1200
motor_pwm_protocol set to DSHOT1200
# set failsafe_delay = 10
failsafe_delay set to 10
# set align_board_yaw = 90
align_board_yaw set to 90
# set vbat_min_cell_voltage = 32
vbat_min_cell_voltage set to 32
# set vbat_warning_cell_voltage = 33
vbat_warning_cell_voltage set to 33
# set current_meter = NONE
current_meter set to NONE
# set small_angle = 180
small_angle set to 180
# set pid_process_denom = 1
pid_process_denom set to 1
# set tlm_inverted = ON
tlm_inverted set to ON
# set osd_tim_2_pos = 2416
osd_tim_2_pos set to 2416
# set osd_vtx_channel_pos = 2071
osd_vtx_channel_pos set to 2071
# set osd_craft_name_pos = 2048
osd_craft_name_pos set to 2048
# set osd_pidrate_profile_pos = 2105
osd_pidrate_profile_pos set to 2105
# set osd_avg_cell_voltage_pos = 2408
osd_avg_cell_voltage_pos set to 2408
# set osd_disarmed_pos = 2347
osd_disarmed_pos set to 2347
# set osd_esc_tmp_pos = 2065
osd_esc_tmp_pos set to 2065
# set osd_esc_rpm_pos = 2444
osd_esc_rpm_pos set to 2444
# set osd_stat_max_spd = OFF
osd_stat_max_spd set to OFF
# set osd_stat_endbatt = ON
osd_stat_endbatt set to ON
# set osd_stat_battery = ON
osd_stat_battery set to ON
# set osd_stat_min_rssi = OFF
osd_stat_min_rssi set to OFF
# set osd_stat_max_curr = OFF
osd_stat_max_curr set to OFF
# set osd_stat_used_mah = OFF
osd_stat_used_mah set to OFF
# set debug_mode = ITERM_RELAX
debug_mode set to ITERM_RELAX
# set vtx_band = 5
vtx_band set to 5
# set vtx_power = 3
vtx_power set to 3
# set vtx_low_power_disarm = ON
vtx_low_power_disarm set to ON
# set vtx_freq = 5658
vtx_freq set to 5658
# # profile

# profile 0
profile 0

# set dterm_lowpass_type = PT1
dterm_lowpass_type set to PT1
# set dterm_lowpass_hz = 90
dterm_lowpass_hz set to 90
# set dterm_notch_hz = 0
dterm_notch_hz set to 0
# set vbat_pid_gain = ON
vbat_pid_gain set to ON
# set anti_gravity_gain = 4000
anti_gravity_gain set to 4000
# set dterm_setpoint_weight = 100
dterm_setpoint_weight set to 100
# set smart_feedforward = ON
smart_feedforward set to ON
# set iterm_relax = RP
iterm_relax set to RP
# set p_pitch = 28
p_pitch set to 28
# set i_pitch = 40
i_pitch set to 40
# set d_pitch = 16
d_pitch set to 16
# set p_roll = 24
p_roll set to 24
# set i_roll = 35
i_roll set to 35
# set d_roll = 16
d_roll set to 16
# set p_yaw = 30
p_yaw set to 30
# # rateprofile

# rateprofile 0
rateprofile 0

# set roll_srate = 50
roll_srate set to 50
# set pitch_srate = 50
pitch_srate set to 50
# set yaw_srate = 50
yaw_srate set to 50
# set tpa_rate = 15
tpa_rate set to 15
# get smart
smart_feedforward = ON
Allowed values: OFF, ON
Default value: OFF

Vidalcris, 2018-05-31:
I think the Diff should look like this : 
```
set dterm_lowpass_type = PT1
set dterm_lowpass_hz = 90
set dterm_notch_hz = 0
set vbat_pid_gain = ON
set anti_gravity_gain = 4000
set dterm_setpoint_weight = 100
set smart_feedforward = ON
set iterm_relax = RP

# profile
profile 1

set p_pitch = 28
set i_pitch = 40
set d_pitch = 16
set p_roll = 24
set i_roll = 35
set d_roll = 16
set p_yaw = 30
```

I just tried and this is working.
So all parameters not related to profile must be placed before the "profile"

Vidalcris, 2018-05-31:
Ok i think i should be able to make a PR for that ;)

mikeller, 2018-05-31:
@Vidalcris: You have to be aware that there are **profiles** and **rateprofiles**. They are independent, and each one has certain settings associated. So you could be using (for example) the settings from profile 1, and the settings (of different parameters) from rateprofile 2.

Vidalcris, 2018-05-31:
smart_feedforward (as an exemple) is related to profiles ?

Vidalcris, 2018-06-01:
For me there is a bug as i said when pasting a diff done from another profile than 0.

Those values are not supposed to be different depending on the profile : 
```
# profile
profile 0

# set smart_feedforward = ON
smart_feedforward set to ON
# profile 1
profile 1

# get smart
smart_feedforward = ON
Allowed values: OFF, ON
Default value: OFF

# profile 2
profile 2

# get smart
smart_feedforward = ON
Allowed values: OFF, ON
Default value: OFF

Vidalcris, 2018-06-01:
Ok now i know where is the problem imo ... finally.

"Get" always give the value from profile 0.
It should imo give the value of the actual profile or even better show the value for each profiles who have a value.

```
Diff All
[...]
# profile
profile 0
set smart_feedforward = ON

# profile
profile 1

# profile
profile 2
set smart_feedforward = ON

# profile 0
profile 0

# get smart
smart_feedforward = ON
Allowed values: OFF, ON
Default value: OFF

# profile 1
profile 1

# get smart
smart_feedforward = ON
Allowed values: OFF, ON
Default value: OFF

# profile 2
profile 2

# get smart
smart_feedforward = ON
Allowed values: OFF, ON
Default value: OFF

mikeller, 2018-06-01:
Yes, I agree that `get` should return the value of the current profile.

And yes, `dump all` is what is recommended for backup / restore.

Vidalcris, 2018-06-01:
Ok thanks ;)

DieHertz, 2018-06-01:
Nice find, I was sure `get` returns current profile.

etracer65, 2018-06-01:
I found the problem and will work on a fix.  It was an unintended side effect of #5935 

The `resetConfigs()` function used to restore the defaults so it could be printed also resets the profile back to 0.
```
void resetConfigs(void)
{
    pgResetAll();

#if defined(USE_TARGET_CONFIG)
    targetConfiguration();
#endif

#ifndef USE_OSD_SLAVE
    setPidProfile(0);
    setControlRateProfile(0);
#endif

#ifdef USE_LED_STRIP
    reevaluateLedConfig();
#endif
}
```

etracer65, 2018-06-01:
Created https://github.com/betaflight/betaflight/pull/6021 to fix the problem.

@Vidalcris Here's a hex with the fix for you to try.

[betaflight_3.4.0_OMNIBUSF4.hex.zip](https://github.com/betaflight/betaflight/files/2063884/betaflight_3.4.0_OMNIBUSF4.hex.zip)

Vidalcris, 2018-06-01:
Awesome, its working correctly now :) 
Good job !

Vidalcris, 2018-06-02:
I think that could be usefull to have the actual profile number displayed after a "Get".
Even better to show it only if the value is part of a profile.
Just a ergonomic choice though ;)

mikeller, 2018-06-02:
@Vidalcris: Good idea, I have been stuck there, trying to figure out if a value is part of the profile or not, in the past as well.

Try this for size:

[betaflight_3.4.0_OMNIBUSF4.zip](https://github.com/betaflight/betaflight/files/2064903/betaflight_3.4.0_OMNIBUSF4.zip)

Vidalcris, 2018-06-02:
Nice but i think like this could be more practical  :)
```
# get iterm 
iterm_rotation = OFF (profile 0)
Allowed values: OFF, ON
```

mikeller, 2018-06-02:
@Vidalcris: 

More practical: Yes.
More costly: Yes too, unfortunately.

The solution that I am proposing is re-using the existing implementation of `profile` and `rateprofile`, making it very cheap to implement. Considering this is very much a 'cherry on top' feature, I think we should go for the cheap one, instead of spending extra effort on it, and causing it to add to the code size (which is already problematic).

closed 2018-06-03 by mikeller

Vidalcris, 2018-06-03:
Oh ok understood ;)
Its already nice this way.

PR #6023 (2018-06-03): Fixed dumping of profile values in CLI.
Fixes #6015.
Alternative to #6021.

This one is a bit more complex. but it does away with the totally unnecessary profile switching when dumping values.