betaflight/betaflight #6016: vTX Remote Control overwriting button and Tramp NFC config changes

joshuabardwell, 2018-05-31:
When vTX Remote Control is enabled, it overwrites channel and power changes made by vTX button or Tramp NFC tag. Alternative configuration methods are rendered essentially non-functional. This is especially a big deal for Tramp users, who want to be able to use wand at races, but vTX remote control at home.

wind0r, 2018-05-31:
Disabling the VTX Feature for racedays is one way to solve this.

joshuabardwell, 2018-05-31:
It's a workaround, but not ideal.

Ideally, the FC would be notified of channel/power changes that come in through any other method and remember them. So if I change channel via NFC or button, the OSD and Lua Script are updated accordingly, instead of overwriting.

joshuabardwell, 2018-05-31:
Why?

DieHertz, 2018-05-31:
Because they don't notify you about changes.

mikeller, 2018-05-31:
One of the problems to solve here is to define who is authoritative for the VTX settings. The current implementation makes the flight controller authoritative, and the settings stored there are pushed the the VTX. This makes the implementation straightforward, works around some tricky bits like 'what to do when an MSP command to change the VTX setting is received, but the VTX isn't responding to commands (like when it's powered down on the bench), and can be used for all supported VTX hardware types.
Reversing this implementation for tramp would complicate the already complex VTX support implementation and probably result in a different way that the tramp has to be configured, compared to the other VTX types.
As a compromise, to make this a workable solution, how about a 'VTX control disable' switch? This way, a pilot could use this switch to disconnect the VTX control from the flight controller (for a race), and give it back authority, and then, after the race, simply toggle the switch again, and have the firmware take control of the VTX again. Alternatively, this functionality could also be added to the TX lua scripts.

joshuabardwell, 2018-05-31:
It's not just for Tramp, but for any FC that has an alternative method of changing setting. Most common of course is push-button, and the obvious response is, why use push-button if you have vTX remote control, but hey, it's there, and some users want to use it. Additionally, some vTX have other methods of control such as Furious Stealth which has a Bluetooth interface. So please don't think of this as a single-vendor hack.

Who is authoritative should be the last input method received. So if you change setting via Lua script, then later press a button, the button press should be authoritative.

@DieHertz not being notified of changes: could this be solved via polling? If the vTX won't "push" notification, can we not just read settings from the vTX to become aware of changes from external source?

mikeller, 2018-05-31:
@joshuabardwell:

 > Who is authoritative should be the last input method received. So if you change setting via Lua script, then later press a button, the button press should be authoritative.

So on startup, there are different settings on the VTX and in the firmware. Who is authoritative?

joshuabardwell, 2018-05-31:
If that were the main problem case, then I could see accepting it as un-solvable, if we could still solve the problem of totally disabling alternate configuration methods.

So we make an arbitrary choice of whether to treat the vTX or FC as authoritative on startup, but at least button, NFC, Lua, and OSD all appear to operate smoothly most of the time.

How often do the vTX and FC get out of sync like this? Maybe it's way more common than I think, but it seems like when the vTX is hooked to the FC and installed in the quad, then the vTX and FC should stay in sync most of the time.

mikeller, 2018-05-31:
@joshuabardwell:

> So we make an arbitrary choice of whether to treat the vTX or FC as authoritative on startup, but at least button, NFC, Lua, and OSD all appear to operate smoothly most of the time.

Not sure if 'the VTX does not switch settings without being prompted most of the time' is something that any user would want to use - but there is no way to guarantee that this does not happen if we go for a solution that treats two independent systems as 'most of the time authoritative'.

nellikb, 2018-06-01:
Hi, 1st time posting here, just thought I'd add some info from my own recent experiment:

When using an IRC tramp hooked up to BOTH the NFC tag AND Betaflight, the Tramp will power up and start transmitting briefly with settings from the NFC tag, then it will change to whatever BF sends it.

**This is potentially dangerous as it may cause a crash by briefly transmitting on someone else's frequency!** 

I agree with Joshua, Please change the way the VTX functionality works - I'd suggest perhaps an option to disable VTX setting changes on initial quad power up and to ONLY send data to the VTX when specifically requested via the OSD menu or LUA scripts etc.

I much prefer using the wand to change settings but the reason I wanted both options is that I recently got a quad stuck in a tree at a race event with a basically fresh pack near the pilots area and my quad was hogging 1 of our 4 available channels for an hour or so preventing 1 pilot from racing while I tried to get it down...

mikeller, 2018-06-01:
@nellikb:

> When using an IRC tramp hooked up to BOTH the NFC tag AND Betaflight, the Tramp will power up and start transmitting briefly with settings from the NFC tag, then it will change to whatever BF sends it.

That is exactly what I am referring to - having multiple 'authoritative' systems work concurrently will always expose you to this problem.

> I agree with Joshua, Please change the way the VTX functionality works - I'd suggest perhaps an option to disable VTX setting changes on initial quad power up and to ONLY send data to the VTX when specifically requested via the OSD menu or LUA scripts etc.

That is what I suggested in https://github.com/betaflight/betaflight/issues/6016#issuecomment-393657299: Make it possible to have a switch (or TX lus script toggle) that switches between the settings on the VTX and the settings in the firmware.

wind0r, 2018-06-01:
You could add something to the LUA Script that enabled vtx feature then change channel and then disable the feature again. so betaflight will not overwrite it again.

to explain the problem: betaflight works similar like the nfc tag. this is because there are more vtx then just tramp or some vtx with buttons and some control, there are also some fc with build in vtx those need betaflight as the storage else it will get potentiall dangerous since the vtx will send some undefined channel.
But the more important but is that betaflight is just another "nfc tag". e.g i connect my mobile phone to the fc with the vtx unpowerd and change the channel. on the next boot the vtx will use those settings. so it is very save to change channels since you can do that without the vtx being powerd. (as known from the nfc tag)
Having another source of "truth" aka.the real nfc tag is a problem since we dont know which setting the unser wants.

because of this the solution for race events that only allow the nfc tag to disable the betaflight feature and just enable it when you 

some technical solution would be that tramp implements something like locking. e.g some raceday mode which when enabled the tramp only accepts nfc updates and it need to be disabled via some wand setting again. it even could be implemented that some password is needed which only the orga knows so the pilots cant change settings with thier wand.

Knifa, 2018-06-06:
Some info from the ImmersionRC Hub Facebook group:

https://github.com/betaflight/betaflight/wiki/VTX-CLI-Settings

At startup the settings are applied to the transmitter. There is a 'vtx_freq' setting that operates as follows:  If vtx_band=0 and vtx_freq!=0 then the 'vtx_freq' value (in MHz) will be configured on the transmitter at startup.  If both are zero then the settings will be ignored. 

tl;dr: *if vtx_band==0 and vtx_freq==0 then the settings will not be sent out to the VTX*

Really bad defaults, imo. Looks like a bug rather than a neat feature for those without wands.

mikeller, 2018-06-08:
@bseishen: Some of the problems of this are that if this is done, there will be no way to change the VTX settings without powering up the VTX, and that, if the VTX settings are changed with the VTX offline, the new settings will vanish into thin air without any notice of this happening.

wind0r, 2018-06-08:
@bseishen could you eleobrate further how reading the settings helps with the multply authoritatives we have here? 
As descriped above. I use the TNR Tag and change the VTX Setting in Betaflight via USB Mobile App. How we know which one the user wants. Tag, BF and VTX all could have different Settings.
Solution for that is allready descriped abouve. Just disable one authoritave. As an Example for Betaflight disabling VTX Feature or set the Settings to 0 so no overwrite will happen.

If you or IRC have any Technical Information we are missing here please help us but currently there isnt any good solution without some protocol change or feature changes.

codecae, 2018-06-08:
The VTX can be made authoritive, but if there is an issue transmitting settings changes from the FC due to heat, timing, etc, you may be required to send the settings change from Betaflight more than once in order for it to stick.  By having Betaflight assert authority over the state control, you can usually be guaranteed that the system will become eventually consistent.  By releasing this authority, there is no guarantee that the settings change you require from Betaflight will result in an update on the VTX.

codecae, 2018-06-08:
State polling is generally undesirable.  If the VTX would proactively transmit its settings via it's protocol when they are changed by manual or wand input, then we would not have this state issue.   Unfortunately, though, VTX hardware is not typically capable of accepting firmware updates, so to change the way they work is pretty much impossible without forcing the entire community to invest in new stuff.

bseishen, 2018-06-08:
@mikeller @wind0r  Very true. Offline changes over USB would get blown away. 

Maybe If tramp telem is enabled, then have the VTX "authoritative"? I personally don't know anyone to use USB configuration to set VTX channels atm, but there could be people out there deem it a necessary feature. Usually its TNR/LUA/OSD. Which the only offline way of changing that is the TNR. I have not dug into the BF code base yet, don't know how reasonable it would be to have a CLI register for setting the authoritative source between FC/VTX.

codecae, 2018-06-08:
Changes are not limited to USB -- Lua scripts and OSD are also responsible for updating this parameter group.  The CLI is just another means of accessing it.

codecae, 2018-06-08:
Temporary delegation of authority would not enforce transient mode states like low_power_disarm and pitmode on a switch.  The FC would ask for the mode to change, but if it doesn't maintain authority, there's no guarantee that these modes would be properly enforced.  Releasing authority would basically render these capabilities useless and limit functionality to ad-hoc frequency, channel and power updates with the acceptance that manual retries may be required.  Implementing the transfer of authority is not difficult and could be enabled as a mode.

bseishen, 2018-06-11:
I would have to go back and verify, but before 3.3 it seemed like OSD/LUA/TNR were working like one would assume. I know polling seems like a terrible Idea, but its not like a fast update interval is needed. Also pit mode on a switch is an important feature for me in team racing :) Let me know how I could be helpful. I am an embedded guy, but I have not really dove into the code base. 

PS Polling the tramp, we could get VTX temps on the OSD 😃

codecae, 2018-06-11:
Vtx temps?  That's not part of the protocol.

codecae, 2018-06-11:
Polling interval is not the issue.  In order to maintain authority over the state of the vtx, it must be asserted by the flight controller if you wish to have mode switches like pit mode.  These states are not committed and maintained by the vtx on its own.  I'll take a look and see what I can come up with.

codecae, 2018-06-11:
One option might be only to assert authority when a transient state is being enforced.   Still requires state polling, though...which sucks.

codecae, 2018-06-12:
I see where you've pulled that from vtx_tramp.c, I just want to verify with immersion that nothing has changed in the spec since the time this was written.

timmytron, 2018-06-14:
Encountered this issue after upgrading into 3.3.x territory from 3.2. The behaviour of this in 3.2.x was perfect. I could set the VTX channel using the wand or the FC and it would not interfere with each other. On my quads I am testing 3.3 I've had to input vtx_freq = 0 and vtx_band = 0 to just let the TNR wand do its thing. Any reason the changed code for this feature cannot be reverted?

wind0r, 2018-06-14:
@timmytron did you read the posts above?

timmytron, 2018-06-14:
Yes but I could not determine what the flaw with the 3.2.x implementation is? I’ve read it all again and it’s still unclear to me.

wind0r, 2018-06-14:
3.2 did not worked well with build in vtx. 3.2 was not able to change settings offline. With 3.2 "Soltuion" it would not be possible to implement low power disarm / high power arm and overall it was a mess.

Possible Solution are also described above. It simply is not possible to have multiple authoritative.

CarlovdMeer, 2018-06-16:
A good discussion about authoritative: Tramp NFC/button (off-line) or BF (on-line). 

**Timmytron** mentions an excellent point. On pre BF3.3 with Lua or OSD, you changed the channel, band and power setting on the fly and will be saved in the Tramp directly. Lua/OSD pre BF3.3 is sometimes a bit buggy, but functional wise: obvious and logical. This won’t work anymore in BF3.3, and the most people don’t understand why the NFC tag is ignored. On BF3.3.3 I notice that Lua display the values of the CLI parameters. It would better if reading out the actual settings on the Tramp and show them when possible (as pre BF3.3).

The slider switch option might be a good option but when Tramp is authoritative, always give Lua/OSD (and CLI) the possibility to change settings on the fly. Consider Lua/OSD (and CLI) as a Wand. Maybe code wise: save in the Tramp (memory) and then read-out the Tramp and show the actual Tramp settings in the Lua/OSD (and CLI). Be aware that the EU and USA versions of the Tramp do not accept all frequencies and power levels.

Most likely this ends in a CLI “VTX_AUTHORITATIVE” parameter.

When redesign BF – Tramp, please use the power levels of Tramp. Immersion has recently added in the Wand (1.07 and up) a 2 mW power level. Less confusion when mixing usage Wand (or Android App) and Lua/OSD (Levels: 2-25-200-350-600 mW).

mikeller, 2018-06-16:
@CarlovdMeer:

> It would better if reading out the actual settings on the Tramp and show them when possible (as pre BF3.3).

One of the reasons this was changed was because the handling of the tramp VTX protocol was unified with the two other VTX standards supported in Betaflight. This had to be done because before this, maintaining and extending the two different VTX implementations inside Betaflight was hard to impossible.

> always give Lua/OSD (and CLI) the possibility to change settings on the fly.

This is **exactly** what we are doing now.

> save in the Tramp (memory) and then read-out the Tramp and show the actual Tramp settings in the Lua/OSD (and CLI). 

This was never supported. lua would always overwrite whatever changes were made directly to the VTX, because there was no way to push updates back to lua once the VTX screen was loaded.

> Most likely this ends in a CLI “VTX_AUTHORITATIVE” parameter.

This sounds a bit limiting. Users will have to bring their laptop to change from 'practice mode' to 'race mode'. As I suggested early on, this could easily be put onto a switch. Users who rather want to use the configurator than a switch can just set up a 'switch' covering the entire input range of a channel.

timmytron, 2018-06-16:
I would be happy with the switch setting if that's an easy way to overcome the issue. 

I'm a Futaba user, so I don't have the LUA scripts in my transmitter, but I can use the OSD to change the VTX over smart audio remotely. I use the TNR wand because I would not be able to change VTX channels offline without it. 

With the switch solution I would default to the TNR wand, and if I needed to remotely change VTX or power on a non race day I could toggle the switch to the FC and then change the required settings.

AndersHoglund, 2018-06-17:
We already have an "Authority switch", even though it is not labelled exactly so. In the BFC ports tab, the "Peripherals" column, you can select Tramp/SmartAudio/Disabled.
But a new "VTX Authority" as a mode on a switch overriding this would be nice.

bseishen, 2018-07-11:
Bump! Anything I can do to resolve this let me know. MultiGP IO is coming up soon and would like a solution to use VTX disable on a switch while team racing and have the tag both coexist. Right now the solution for me enable tramp telemetry in the BF configuration when I need to remotely disable the VTX.

ctzsnooze, 2018-07-26:
The problem really is due to the design of the protocol, which was outside Betaflight's control.  The protocol provides no notification, by the camera to the FC, of when there is a channel change made by pressing a button, or when a wand is used.  That's the fundamental problem.  The FC isn't told of these changes.

As a result, if the FC doesn't know that the channel has been changed.  If the FC does **not** have authority over the VTx, a manual channel change on the VTx would set it to a different channel from what is shown on the OSD.  Clearly this is undesirable.  So the current model is to enforce FC over VTx to ensure consistency.

I guess we are seeking some improvements to that model that give manual inputs or wand inputs equal authority to OSD/LUA, including writing those changes to the FC just the same way.

Due to the protocol limitations, for the FC to be updated when the user changes channel with a button, the FC has to continually ask for the current channel value from the VTx, and display whatever the VTx sends, over-writing and saving the current channel settings when it changes.  This ensures that under all circumstances the OSD is consistent with what is actually happening, and that a reboot will go to the last used channel all the time.  Good.  Wand or button presses changes will then have equal authority to an OSD change. 

I think that's what we want. But there are some issues.

There is a possibility that the VTx data might get corrupted in transit. The VTx settings would be polled frequently, the chance of this happening is non-trivial.  There are two mitigating factors.  First, presumably channel change is only possible while disarmed.  Second, if a non-FC commanded channel change occurred, a warning could appear on-screen.  Third, non-commanded channel changes would change the displayed channel (assuming it was displayed all the time).  So this would require some OSD code changes to provide external channel change notifications / warnings.  But there is one more issue.

In such a model, how does the FC change the channel on the VTx via OSD?  The FC would need to remove authority from the VTx, transmit the OSD change channel request, poll for a confirmation, keep transmitting request and wait for confirmation until a timeout, and not return to Vtx authority until the change was confirmed.  

These changes do require a bit of coding  :-)

Could people clarify if the above summarises the best way forward?

timmytron, 2018-07-26:
That's a good approach @ctzsnooze but I think it could be simplified a little as we will only use the wand or button right after power on (or in the case of the wand, powered off) so perhaps if the FC only queries the VTX during power on and continues to do so until:

- Quad is armed
- a change of channel initiated by the LUA / OSD through the FC

Then make the FC authoritative until next power cycle. 

I believe every situation where the button or wand will be used is right when its powered on (or before in the case of the wand), anyone using these methods will only want to use the FC control later such as when wanting to switch to graveyard mode, or change channel / power remotely because you've become stuck in a tree.

Or does that not work with your proposal?

ctzsnooze, 2018-07-26:
I was thinking that the FC should ignore all changes while armed, so that you can't change channels via OSD, LUA, wand, button or anything, while armed.  The VTx data is CRC checked so corrupt data is unlikely.  

Once disarmed, the channel would be editable by any route, and the value stored in both VTx and FC, if changed.  That's simple enough.  And unless the pilot changed something via LUA or OSD (and they could), the channel would stick.

For added certainty, whenever a change is written, some kind of on-screen message says 'VTx channel changed to X'.  That would be a good confirmation (and a good indicator of some unwelcome or unexpected change).

The user could disarm, change power levels and enter pit mode etc via buttons or Lua or OSD at the end of the race, and expect the quad to reliably power up in that mode reliably.

If a quad got stuck high in a tree near the track, the pilot could disarm, use OSD to cut VTx power, and the race could go on.  But if VTx has absolute authority, that would not be possible.

closed 2018-09-01 by stale[bot]

anthonycake, 2018-10-18:
Any chance of re-opening this issue?. The BF/NFC incompatibilities are causing a lot of confusion with pilots, and since NFC is being used at some of the big events (DCL as one example) we need a clean solution. Right now the solution is to take a pair of wire cutters and simply cut the connection to Betaflight. 

So one way to move forwards is to use some of the undocumented (for the moment) API features of the Tramp. The Tramp can be queried and return all kinds of interesting data to Betaflight, including it's temperature, race lock status, frequency, power, pit mode status, regional settings (min/max frequency, and power limit). 

One thing I would like to see is that if the tramp is race-locked on BF startup, BF steps aside and lets the NFC wand take control. 

There is no notification mechanism in the protocol, but a simple poll (yeah.. yeuch... ) from time to time (rate doesn't need to be very high) would allow the two systems to co-exist a lot better than they do now. 

Thoughts?

wind0r, 2018-10-18:
Wow. Cutting a wires is a strang solution but at least it is fast and a user can revert it fast.
Just disabling the VTX Feature in Betaflight would have done the Job and is the advised way.

I still do not think that on a DCL event you need both BF VTX and Wand.

anthonycake, 2018-10-18:
Well, in my opinion the wand should always be 'boss'. For an event like DCL they do need BF control and the Wand for the following reason. 
When pilots are flying alone, and don't posses the wand, they need to change channel using BF control. 
When they arrive at a race, and race-locking is performed using the wand, that needs to take priority, always. 

It is relatively simple to allow the two to co-exist using a few simple queries to the Tramp. 

This may have already been handled, but however it has been handled it is not working. I help people to resolve these conflicts at every race that I am a part of, and it has generated a lot of confusion.

anthonycake, 2018-10-18:
I just reviewed 6680, this doesn't go far enough. When a quad is race locked at an event this needs to override all other control methods. The entire purpose of race locking is to allow pilots to power up safely at large events (in pit mode), and for organizers to be able to decide frequencies and power levels with the race wand. 
If all of this relies on some obscure setting in BF then the system fails.

wind0r, 2018-10-18:
As i said above. Just do not setup the tramp protocol in the ports tab and you are done. There will not not a single communication to the device since betaflight doenst know it exist. If that realy is not the case you need to explain how bf still does the change.

the changes in #6680 are not for racedays and i know those arent the solution to your problem.

AndersHoglund, 2018-10-18:
"VTX Control Disable" must be usable even on racedays, disable VTX conttrol and the the Wand has full exclusive control. Or are there some problems with this mode? In that case, disable Tramp in the Ports tab. Or cut the wire as a last resort. 
Polling VTX status is not an option.

anthonycake, 2018-10-18:
These options have been available for a while, but yet still pilots, unaware of them, turn up at events like DCL and cannot use Wand control. The current solution is not working, period. 

The entire concept behind the race wand, and the control that it puts into organizers hands, relies on the fact that the wand is always 'boss'. A setting in BF that can override it is a huge issue and does not solve the problem. 

The other way I can fix this is in the firmware of the Tramp, but that may cause other problems, and then BF will be completely ignored when the wand is in control. It will mean that pilots will need to turn up at events with only 'v3' tramps, which will result in extra cost, and yet more confusion. 

From the pilots' perspective, BF is broken, and as a result wires need to be cut at events, removing BF control completely.

So why is polling VTX status not an option? because we need to fall in like with the other vTx's out there which don't support a bidirectional interface?, of because of some other reason that is not clear to me yet?

wind0r, 2018-10-18:
Polling does not work. It got explaind MULTIPLE times here and it even got explained in my last message again. 

I just will copy past it from above.
 Similar to the NFC Tag you can change vtx settings when the vtx is offline and they will get enabled with the next boot. Similar to the NFC Tag. If we query the VTX on boot this feature is not possible anymore, because of that you need to say bf which solution you want to use. If you want to use bf you enable it and if you want to use an external solution you disable it.

If there is anything unclear about that please write it down and i will try to explain more in detail.

Just to ask a similar question. So why there is a Problem with just connecting to BF Configurator and Disable the Tramp Procotol in the Ports feature?

mikeller, 2018-10-18:
@anthonycake: Please read the above thread for an exhaustive discussion of why a 'multi-master' system for controlling the VTX settings will not work, and result in a system that acts in a random way.

However, what could be done is for Betaflight to read the 'race lock' flag, and if it is found to be set, inhibit sending of any updates to the VTX. (Which is something that could be implemented on the VTX side alone, without the need to extend the protocol.)

anthonycake, 2018-10-30:
> @anthonycake: Please read the above thread for an exhaustive discussion of why a 'multi-master' system for controlling the VTX settings will not work, and result in a system that acts in a random way.
> 
> However, what could be done is for Betaflight to read the 'race lock' flag, and if it is found to be set, inhibit sending of any updates to the VTX. (Which is something that could be implemented on the VTX side alone, without the need to extend the protocol.) This will make 'race lock' act in the same way as the 'disable VTX control' master switch, except that it's controlled from the wand.
> 
> Can you please supply a link to the specification of the extended protocol, so this can be evaluated?

This would make sense, I think we need two changes though to resolve 90% of the confusion: 
1) BF respects the racelock, and during the first query to the Tramp, if detecting a racelock, inhibits sending any updates, as you suggest. 
2) Also during this first query the power and frequency are ready from the Tramp, and used to set the current BF value. 

Case 1) solves the issue for races and leagues (DCL for example) which race lock the tramps, since the locking is always done prior to power-up. 
Case 2) solves the issue for mere mortals, who are using the wand to set the Tramp in the local park before powering up. 

The only case that remains is for the user who is changing settings using the wand after powerup, but I believe this is much less of a problem than the two cases described. 

Is this a good compromise?

timmytron, 2018-10-30:
The value of the wand for all pilots even with SmartAudio and LUA has always been about using it when powered off. This is a suitable compromise as anyone with all those components in play use BF to conduct the channel change after being powered on.

mikeller, 2018-10-30:
@anthonycake: I am not going to discuss with your for yet another time why the multi-master system you are proposing in 2. is not a viable option. I guess your users will just have to use the 'disable VTX control' mode.