betaflight/betaflight #6018: Arm drop - after quick throttle drop to zero, one motor goes to zero, other three stay about 20%

ctzsnooze, 2018-06-01:
About five people have asked me about 'arm dropping' issues with 3.4.  

Typically, on quickly dropping throttle to zero, one arm seems to drop.

In the logs I've seen, the motors trace show one motor with zero drive, whereas the other three are stuck on 15-30%.  It seems as if one motor is failing to stop properly.

These are sporadic reports, so I've opened this issue for people to post logs and setup.

Before posting, please report Betaflight version, ESC brand and type, ESC/BLHeli mode, ESC settings, FC type, loop time, and check CPU%.  If possible add a log or a link to a log.

SteveKfpv, 2018-06-01:
Just tried rolling my quad back from 3.4 to 3.3.2, same problem.
There's a drop then some time later an uncommanded correction the other way.
[image]
Racerstar 35A 4in1 on blheli s 16.7 defaults (all reversed for reverse prop)
Omnibusf4SD on BF 3.3.2
8k8k loop dshot600

I have another identical quad but it is using an omnibus f4 non SD. No problem with that one.

May try rolling back to BF 3.2.5 to see if problem persists

ProcessDestroyer, 2018-06-01:
Not entirely sure it's the same issue and will be a while before I can look into it more,, but I was having noticeable dip on BF 3.4, April 9th build, and captured a log with several flips, and almost everyone has a strange 'knee' on the pitch gyro trace, as well as roll, coming out of a maneuver.  I've posted a log and a screenshot compilation in Slack, under Test_Feedback.

McGiverGim, 2018-06-01:
@SteveKfpv can you attach the log of the image?

theSkonk, 2018-06-02:
I have build 890 loaded on a new stack and esc in the qwad I can try for a log with iterm relax turned off.. the tune has been set up to take advantage of it though. For the record the issue has been since I built it and was flying test code for the dual dterm filter based on 3.3

etracer65, 2018-06-02:
I just got done testing this and can't replicate.  Did a bunch of big LOS punches and zeroed throttle and took my hands off the sticks as it fell.  All gyros stay rock solid at zero during the fall and motors are all around zero.  Note that I don't have any of the i-term manipulation features, throttle boost, etc. enabled.  Anti-gravity is enabled and set at 4500.

SteveKfpv, 2018-06-04:
I tried rolling back to 3.2.5 with the same problem. I now am more certain it is some hardware issue, but can't imagine what. Bit more detail, my light weight quad with 2306 motors and heavy props (5x5x3) are showing the issue most. My freestyle quad with identical hardware but weighing 25% more and using 5x4.3x3 props does not show any sign of the problem. Perfect on full punch to zero throttle. Going to try different combinations of props, quad and switching FC to see if there is anything that shows up.
Here's a screenshot of the issue again and a link to the log (shown clearest at 27 seconds) https://1drv.ms/u/s!Arl8iTLBSLo1g50U1sj7cPujMEd_tA

[image]

etracer65, 2018-06-04:
In every case I just see the PID controller and motors responding correctly to some uncommanded pitch caused by external forces.  Possibly some aerodynamic effect with your setup.  I noticed that your P gains are quite low.  Increasing them would allow a more direct correction rather than relying on slow-acting I to correct the movement.

SteveKfpv, 2018-06-04:
Its a reasonably aero frame (canopy, thin arms) with central bottom mount battery. Ill give tuning it with higher P gains a go. Otherwise feel free to drop my experience with the issue as an input to this potential bug. To me, once I installed 3.2.5 and the issue remained, then that ruled out firmware as the cause.

theSkonk, 2018-06-05:
@etracer65 when I've logged LOS ( I'm not much of a LOS pilot i haven't had the issue) for me it requires forward flight to make it happen.. I've got a new stack in my racing rig and it needs tuning before i can get back to trying to solve this.. its not flying as well as the old stack at this point. so when the rain backs off i'll try and get it sorted

etracer65, 2018-06-05:
I can’t really tell from the log alone, but I’m wondering it if it occurs at the top of the punch at the point the quad goes weightless and starts to fall.  Transitioning from aero drag on top to the bottom.

ctzsnooze, 2018-06-05:
I noticed that the effect always a *diagonal* drop - just one motor alone is affected.  And the gyros show a combined pitch *and* roll deflection, i.e. a diagonal / on the arm deflection of the quad.

I can't envisage any aero feature causing this.  All quads on which it happens on have been perfectly symmetrical in COG and aero in the roll axis, and generally symmetrical in pitch also.  If it was aero or COG related on pitch, I'd expect both the roll or both the pitch motors to move together.  We all see this sometimes.  But with this problem, it's not pairs of motors, it's just one.  And on any given log, it's *always* the same arm.

So the strange things are:

- always one arm
- only when cutting full throttle to zero stick, and only happens when one motor hits zero, then the other three stay too high.
- doesn't happen when cutting hard to say 10% residual throttle, then everything is perfectly normal.
- absolutely no sign of any motor inequality or aero offset or anything at all of that kind during normal flight, all motors do the right thing, punch-outs and drops are flat and happy.  

Therefore I do not think it is aero or cog related.  

I am not clever enough to know if upwards deflection of the pitch trace means pull back or nose down.  But I anticipate that if we consider the arm where the motor drive goes down, I expect that the logs would show that the affected arm actually goes up, causing both pitch and roll deflections simultaneously, and the PIDs respond to try to stop that arm climbing.

If so, the only possible explanation is that one motor fails to properly go to zero throttle on sudden stops.  Instead, it keeps generating non-zero thrust, so that arm goes up relative to the others, and the PIDs cut throttle drive to it.  The conceptual opposite of what happens when a motor/ESC loses thrust, then the motor drive to that motor goes high and the arm drops.  Here we have the precise opposite.

It might be possible to replicate this hand-held, but it might be a little dangerous :-)

It's a very strange phenomenon.  I'm wondering about the ESCs themselves or some weirdness about the communication of DShot commands or the mixer with very sudden throttle drops.

ctzsnooze, 2018-06-05:
OK so looking at the above logs...
- Skonk's DShot 300 log, motor 3 gets stuck on zero, quad pitches forward and right = motor 3 high
- SteveK's log, motors 1 and 3 go to zero, quad pitches forward and a bit right, seems like both rear motors were a bit high, then prolonged recovery
- Example 3, below: Motor 1 gets cut to zero, quad rolls left and pitches forward = motor 1 high
- Example 4, below: Motor 3 gets cut to zero, quad rolls right and pitches forward = motor 3 high
 Strangely never motors 2 or 4?
Weird.

Example3, raptor romeo log - multiple occurrences, always motor 1, always the same:
[image]
likewise same log:
[image]

Example 4, raptor alpha - on this quad it is always motor 3, with pitch forward and roll right, again multiple similar events in the log:
[image]

bkeinert, 2018-06-06:
Does this only happen when using DSHOT?

bkeinert, 2018-06-06:
Sometimes there's a reason for a certain question. All I'm asking is: Has anyone seen the same behavior with MULTISHOT (assuming reasonable settings)?
There have been some changes around DSHOT. It might already help to simply rule out that this has anything to do with DSHOT.

DSHOT is super common these days on relatively recent machines, maybe that's why all above reports mention DSHOT. Having more indicators for the root cause is always helpful.

theSkonk, 2018-06-06:
@bkeinert totally i'll try to get a log or 2 if the weather starts playing ball.. been grounded for over a week now

theSkonk, 2018-06-11:
so i tried with multishot and it was the same motor 3 is flatlining.. @ctzsnooze seems to think its protocol or hardware related.. for me its only happening on blheli 32 rigs.  I've got some blheliS 3" quads and its not an issue. also while the motors that dip are different the behaviour is the same on 2 different esc brands and one brand its been over 4 different esc's 
what esc's are on the other effected rigs..??

mine are all running dshot 1200 and firmware 32.40, the first lot ( 4 different esc's ) were the HGLRC dinoshot 40amp 20x20 escs and the other one ( current build ) is an airbot ORI32 20x20 
im reaching out to one of the Blheli32 devs and see if he knows any thing

SteveKfpv, 2018-06-11:
Update for me, I changed the configuration of my quads to squash X (which I prefer anyway) instead of stretch and raised pitch P. The dipping issue seems to be much more reduced. Still there but not so bad that I feel the need to rectify it. As @etracer65 suggested, higher P does cover the problem but I can't go high enough to remove it completely without causing bad prop-wash response. I also tried multishot and it made no discernible difference. I would upload a log but lost my SD card in a little crash... Oops

butterSnake, 2018-06-12:
What esc's are you running @stevekfpv?

theSkonk, 2018-06-12:
What esc's are you running @stevekfpv?

SteveKfpv, 2018-06-12:
@theSkonk Racerstar REV35 blheli S 4in1

RipperDrone, 2018-06-12:
Arm Dip upon throttle cut is there from BF 3.0, on many of my 20+ builds. Has been reported as issue before, under titles like throttle blip, strange behavior under throttle cut, strange attitude etc. 
Unfortunately it seems all these issue tickets have been auto-closed in the meantime...

Issue has never been fixed, if you asked me. Can be related to some quads being a little offset with cog vs cot (center of thrust)...

I have tried PIDs, different ESCs and motors, always there is a dip upon very hard throttle cuts.

etracer65, 2018-06-12:
I still contend that the only thing that can cause a quick "arm dip" on suddenly cutting throttle is if the motor opposite the dip doesn't brake as well as the others.  The logs support this.

theSkonk, 2018-06-12:
Motor braking is a function of the ecs though if I'm correct @etracer? 
I've got another machine built up with the ori32 ecs will try for a bit of flying tomorrow. 
I'm also Gona try proshot out see if that changes any thing

avr39-ripe, 2018-06-13:
@SteveKfpv are you use low ESR capacitor on battery lead on your ESC? Recently I also suffer from random sky fall of my newly rebuild quad with Racerstar REV35A 4in1 ESC.. only adding 470uF 25V low ESR capacitor get rid of this problem. If not your case - sorry to disturb.

mikeller, 2018-07-07:
@ctzsnooze: Have you had any feedback on this? If not, I think this should be closed.

theSkonk, 2018-07-07:
no one has managed to work out the issue..

RipperDrone, 2018-07-08:
Here's my today's recordings of BBL viewer: Looks like 2 motors respond strangely after throttle cut... no other RC inputs that I intentionally overlaid over the throttle cut :-(

BF 3.4, OMNINXT7, Typhoon32 ESC (4in1)
[image]

RipperDrone, 2018-07-08:
Quad is a little top heavy, so I was wondering if this can be produced by wind + inertia effects only. But it really seems the FC / gyro does not do its best to make the PID controller compensate these effects.

Important to mention: This quad has a 108Hz resonance in its delicate frame, so I had to set gyro + D term lowpass filters to 50Hz. Linked?

RipperDrone, 2018-07-13:
tried 2 more quads today, boh Omnibus F4 SD based FW (Merica + Floss 2). Same behavior. Irrespective of throttle idle settings from 3% up to 7%...

RipperDrone, 2018-07-13:
here's a diff, just in case...
https://drive.google.com/file/d/1fYR0oSDfUycnxzhDK45meef48po9S-0d/view?usp=sharing

kenkamm, 2018-07-19:
I have this issue as well and it has remained through all kinds of different hardware. It even exists on Butterflight as well. For me the problem manifests as an uncommanded pitch move immediately after dropping throttle. Antigravity doesn’t solve it. Raising P to the point of oscillations doesn’t solve it. Different FCs, ESCs, frames, motors, top mount battery, bottom mount battery, it’s always there. 

[image]

tyrelcorp1, 2018-07-21:
I am having the same issue with my Armattan rooster.  If I punch out and drop throttle after a few seconds it will roll till im upside down while im falling.  Also if I do a loop as I pass inverted I feel a hesitation in the elevator as I go upside-down.  All other characteristics are fine.   
[nose drop.zip](https://github.com/betaflight/betaflight/files/2216089/nose.drop.zip)

kenkamm, 2018-07-21:
That’s a different issue entirely. Sounds like airmode is off.

RipperDrone, 2018-07-21:
can tell that on my quads, Airmode is definitely On. (RC mode switch, not permanent by feature switch)

kenkamm, 2018-07-21:
Same here. I have airmode permanently enabled. I believe the “dip” or “drop” described in this issue is a brief bobble. That’s how it is for me, anyway, as you can see in the log I posted.

RipperDrone, 2018-07-21:
yes, after abrupt throttle cut, quad shortly loses attitude (bobs in one direction, very often front right on my ~10 quads which are affected), then recovers back to original attitude.

tyrelcorp1, 2018-07-22:
Tried simulating the engine stop with some full throttle with my drone in a clamp.   saw no noticeable engine difference.
[throttle run.zip](https://github.com/betaflight/betaflight/files/2216672/throttle.run.zip)

tyrelcorp1, 2018-07-22:
definitely in acro mode
drone at idle showing gyros are working

tyrelcorp1, 2018-07-22:
[idle11 acro.zip](https://github.com/betaflight/betaflight/files/2216689/idle11.acro.zip)
definetly in acro mode.
video shows gyros working normal with no throttle

kenkamm, 2018-07-22:
I think you’re right in that more powerful motors make the problem worse. Unfortunately there is no setting in BLHeli that can limit how quickly the braking force (damped light) is applied. I already asked the BLHeli 32 developer about it.

etracer65, 2018-07-22:
I've said this previously, but in every log it looks like the flight controller is doing exactly what it should - react to an external force.  The cause of the external force is what needs to be determined.  It could be aerodynamic but I think not.  More likely related to motor performance - particularly braking.

Assume the case with the front-right (motor 2) arm dipping.  If motor 3 (back-left) was not braking as well as the others then it'll have some residual thrust that will cause the quad to tip towards motor 2.

Has anybody with a repeatable situation tried swapping motors/esc's?  In the above example (front-right dipping), try first swapping the two rear motors.  If the motor is the cause then the dipping will shift to the front-left.  I think it's unlikely to be a motor but possible.

The more likely scenario is the ESC.  So try the same test but this time swap the rear ESC's.

RipperDrone, 2018-07-22:
I have it on >10 quads, so it is not limited to one brand of ESCs or motors at least (I have >5 makes of ESCs, single + 4in1's and >8 makes of motors on the builds which show this behavior). 

When I pull back throttle quickly but leave it around 20%, the effect ist not there (maybe because I go slower in the braking trying to stop around 20% thrust?). 

I have set the pid_at_min_throttle switch to OFF (motors just spinning constant when arming, no PID controller engaged), and I had to increase min_throttle to 1030 due to RC having some variation on my throttle stick (Taranis X9D). Can this Min_throttle setting or pid_at_min_throttle have an effect on dropping one motor to zero rpm or cause different braking behavior among the motors?

RipperDrone, 2018-07-22:
@etracer65 Don't the log show that one motor out of the 4 does not spin down to idle rpm but stop completely - which it is NOT supposed to do? If so, how can a motor go below idle rpm, is idle_rpm the lower limit for the PID controller?

RipperDrone, 2018-07-22:
Taking pristine builds apart is painful - especially since fixing 10+ in case it's hardware related will be a nightmare... so my hope is with something software related. Yes, you are right - currently it's more or less fishing around. I thought by supplying all the logs there should be an indication where to look...

RipperDrone, 2018-07-22:
I'm running DShot as my mainstream protocol, and up to 7% idle the problem persists. So question is why can the ESC brake one motor down to standstill whereas the others seem to hit the intended floor of dshot_idle_percentage...

Maybe there is something to try out on the ESC settings...

RipperDrone, 2018-07-22:
higher or lower startup power might have an effect...

kenkamm, 2018-07-22:
I have tried a few experiments on one of my quads to try to figure out if it is a hardware issue. I used the same quad each time to minimize the number of changes being made.

First I rotated the 4-in-1 ESC (with motors still connected) 180 degrees, making the front ESC outputs the back and the front motors the rear motors. No change - the quad still dips in the same direction when throttle reaches zero. (Note I have just about zero bobbing on yaw or roll - it’s all pitch).

Next I switched the front and rear motors, leaving the ESC in place. Still dips in the same direction, by the same amount. 

The one thing I have noticed is that balancing the quad perfectly left/right/fore/aft does not fix the issue. However, interestingly, if the quad is nose heavy, it seems to help. Still experimenting with that one.

etracer65, 2018-07-22:
You're missing that there's a difference between what is being **commanded** to happen vs. what is **really** happening with the motor.  The flight controller can only command and react if it doesn't happen correctly.  That's what you're seeing.  Exactly why could be hardware (defective or weak motor, bad ESC, etc.) or possibly some issue with the ESC firmware or settings.  I'm not a BLHeli developer so I can't help you with those settings.

It's obvious where to look for the problem - it's the motor/esc diagonally across from the arm that's dipping.

kenkamm, 2018-07-22:
Ripper, there is nothing in the ESC settings that will help this. It’s the braking effect that we would want to control, and we don’t have control over that. (Except to turn it off, which would be very bad.)

kenkamm, 2018-07-22:
Etracer, not sure who you are addressing, maybe you should indicate it so we can see which one of us you’re trying to insult. ;)

We’ve already done what you’re suggesting, many times, and found that the problem doesn’t follow the motors and ESCs, it follows the orientation of the craft.

etracer65, 2018-07-22:
I'm not trying to insult anybody.  This is a long thread with many contributors so it's not possible to keep track of who has tried what.

The short answer is I don't see anything in the blackbox logs that the flight controller could do differently.  Some external force is acting on the quad and the flight controller is correctly responding to it.  I have built dozens of quads and never experienced this problem.  While I'm not disputing that the group active in this thread are seeing something, the vast majority of users are not experiencing this problem.  So unless you can identify a commonality between your setups I'm afraid it just might be a flight characteristic of your particular frame or configuration.

kenkamm, 2018-07-22:
Etracer, the flight controller is not responding with enough authority and it’s not responding quickly enough. 

I think most people actually do have this issue, but they either don’t care because they aren’t trying to get HD footage, or they don’t realize it because they haven’t been flying long enough, or they fly in such a way that this issue isn’t realized. 

I’ve had two people who claim they don’t have this issue send me their blackbox logs, and I pointed it out to them within seconds. Care to share some blackbox logs yourself? I’ve yet to see a log from an experienced, aggressive pilot that doesn’t have this issue. I would really, really love to find a setup that truly does not exhibit this problem. I WANT to be shown that there is a way to fix this.

RipperDrone, 2018-07-22:
@etracer: The problem persists since BF 3.0 or even slightly before, have never experienced it before. I have built 30+ quads now, all of them being different frames, 3-5", X/stretched, top/bottom loader LiPo, 20+ diferent sets of motors, 10+ different makes of ESCs, 15+ different FCs. Setups with large, small and no extra caps. Half of them are affected severely by the arm drop issue, smaller ones are not so much affected, sfter motors are better. All strong / heavy motor builds are affected. 

To me, this doesn't seem like a small minority of builds which is plagued by the problem, sorry to say that.

From what I have understood, you are 100% sure that the PID controller will NEVER command any motor to go below its idle rpm, correct? I have seen so many evolutionary extensions of the code related with pidsum, different rpm thresholds monitored, Airmode and Antigravity so I am no longer sure that there isn't a glitch somewhere in the code which allows the PID to command a lower rpm than idle and might make the motor stop completely.

If this assumption can be ruled out, we have to look for the ESC braking algorithm and see how this can be influenced (in BLHeli...). 

@kenkamm  I was just referring to Startup Power since I read in BLHeli manuals that this has an influence on high torque speedup vs. fast rpm speedup, so couldn't this also have an effect on the braking which is basically shorting out the motor windings in a controlled way which might be driven by similar parameters like the speedup?

Anyway, even if I may sound stubborn: I have not fully digested the concept why the PID controller can be ruled out already by looking at the logs: Can it be verified in the logs or in the code that the commanded output is never below idle throttle for any of the motors?

RipperDrone, 2018-07-22:
My 2 cents on many users not noticing the issue: If you are not cutting throttle very hard, the quad flies nicely, no question. I guess not too many pilots are flying aggressively enough to really cut hard on the throttle by fast-flipping it down to experience the dip. Try it on some of your builds with strong motors (like 2207, 2306, 5" builds)...

RipperDrone, 2018-07-22:
@etracer65 HEre is my diff, to be able to compare against yours which do not show the issue... maybe you find something in the settings. Vice versa, I'm happy to check your diffs for differences if you wish...

https://drive.google.com/open?id=1PD-H_9OGwnP7vYVieSq3KS48oPFjYnU_

kenkamm, 2018-07-22:
@Ripper, my understanding from the dev is that startup power does not affect the braking, and the damped light function is not adjustable. The maximum acceleration feature also does not influence braking. I asked him another question, will report back.

RipperDrone, 2018-07-22:
@kenkamm <sigh> one little piece of hope just died... :-)

Yes, damped light was on/off switchable until BLHeli_S, from then it wasn't anymore. sskaug didn't have any other clue for us regarding ESCs, just to make sure we follow all the potential directions?

kenkamm, 2018-07-22:
@Ripper, well, we definitely would not want to turn it off anyway. What I would like to see would be variable braking, or one that even allows the braking to taper off slightly as minimum throttle is reached. This might prevent the shock to the system that causes the violent twitch that the flight controller cannot compensate for.

RipperDrone, 2018-07-22:
I see. 

Maybe this find helps as well: https://github.com/bitdump/BLHeli/issues/310

Fast decel noise (in conjunction w/ strong motors, high magnetization to be inverted by braking current) introduced into the system could blind one ESC/motor to not accept the commanded values anymore for a short period of time?

kenkamm, 2018-07-22:
I don’t think that really applies to us. In that case, PWM was being used, and control was being lost. If that was happening to us, we would be crashing. 

Sskaug responded quickly. He said, “Nope. Braking power is fixed, it can not be altered for a given ESC/motor.”

So there you have it, right from the developer. We do not have control over braking force.

[102 more replies]

closed 2020-06-16 by ctzsnooze