betaflight/betaflight #6019: GPS Hold mode drops quad out of sky, Target: DALRCF405

mmakam2, 2018-06-01:
Hello,

GPS initializes properly, I get a 3d fix, shows a number of satellites and accuratish speed on the OSD. As soon as I enable GPS HOLD, the quad tilts over and starts falling until I disable GPS HOLD. I tried 2 other GPS units with the same results.

GPS Modules tried:
OP GPS
BN-880
Micro Ublox M8N (readytoflyquads)

McGiverGim, 2018-06-01:
The GPS Hold does not work in Betaflight. 

Maybe is better to remove this mode (and others that don't work) from the Configurator.

mmakam2, 2018-06-01:
Did not realize that, i couldn't find any documentation on the mode. I agree better its removed. Curious, out of all the baseflight forks, is there one that does support gps hold and the DALRCF405 target?

McGiverGim, 2018-06-01:
The only one that supports GPS Hols is iNav, but I don't know it they support this target.

mikeller, 2018-06-01:
@mmakam2:

> Did not realize that, i couldn't find any documentation on the mode. I agree better its removed. Curious, out of all the baseflight forks, is there one that does support gps hold and the DALRCF405 target?

The upcoming 3.4 release will remove these, and instead give you `GPS_RESCUE`, which actually works.

andrejpodzimek, 2018-06-02:
Could it be related to #6028? That problem has been around since the dawn of ages; I think I spotted it last year already. But luckily I didn't try GPS hold, because baro hold doesn't work in Betaflight (#3823) (at least not by default), so there was no point in trying GPS hold. (Why would one do GPS hold without altitude hold?)

How `GPS_RESCUE` is supposed to work (without a proper altitude hold and without a working compass), that remains a big mystery to me.

ChilledoutUK, 2018-06-03:
works great i guess it calculates the heading on gps vector data.

https://www.youtube.com/watch?v=uMCezyTeU_U

just needs a failsafe guard time and fine tune the landing some and it will be great.

mmakam2, 2018-06-03:
I tried GPS rescue, works as advertised 😀

s0up, 2018-06-03:
When it gets home on rescue mode it continues to loiter.  Using GPS to emulate compass data requires movement to happen to not go out of whack.  Holding a position without the requirement of a compass is not very feasible as the only sensor you have to rely on to keep orientation is the gyro which is prone to drifting.

ChilledoutUK, 2018-06-04:
is loitering the new condition for returning to home with gps_rescue now?

closed 2018-07-11 by stale[bot]

andrejpodzimek, 2018-07-16:
The GPS safety feature sounds quite useful per se, but it's in no way a solution to the whole compass + gps ordeal. (Just stating the obvious here.) Both compass and GPS appear to be usable in [iNav](https://github.com/iNavFlight/inav), [LibrePilot](https://github.com/librepilot/LibrePilot) and [Pixhawk / PX4](https://github.com/PX4/Firmware) (and on a number of closed-source platforms), so this doesn't look like a fundamental hardware flaw affecting compass chips. (Admittedly, I've only *seen* it work on LibrePilot some time ago; never tried iNav with GPS or Pixhawk (at all).) Could Betaflight perhaps reuse some of the iNav code if it's been through more autonomous flight testing than Betaflight? Many people say *just use iNav then*, but hey, iNav doesn't have DShot&hellip;

s0up, 2018-07-16:
Betaflight is geared towards performance -- getting hold mode to work with
a compass is not a particularly hard thing to do, but getting a compass to
work on a miniquad reliably is, which is 99% of all users on betaflight.  I
would suggest using inav if you need truly autonomous features.   I think
bringing back assisted navigation modes is probably not in the scope of
betaflight.

andrejpodzimek, 2018-07-16:
I’ve been planning to try iNav for a long time, but as long as it doesn’t have DShot, it’s just not worth it. It would be sad to use an analog ESC protocol with BLHeli32. ESC telemetry is yet another important feature related to DShot.

I’ve never understood the single-purpose drone idea. When I travel for business / vacation, I can’t take multiple drones with me. That’s why I build all my drones so that they could possibly do 5g+ thrust and flips for fun, but also a hold in 100 meters of altitude for taking a nice video, which has become very easy with RunCam Split. Whichever drone I take with me on a trip, I want it to have as many capabilities as possible. (“Get a Mavic” is not a solution here — a Mavic doesn’t flip.) That’s why I keep asking about these “universal” features that can make a drone more versatile.

Also, it doesn’t necessarily have to be a compass from the very start. It would be already nice if at least baro hold worked in Betaflight — but it doesn’t (#3823, #3824).

IMO, all this^^^ could give Betaflight a way broader audience. For example, LibrePilot appears to be somewhat behind its zenith and very few new devices support it. Pixhawk is maintained, but also doesn’t offer much of a hardware choice. Betaflight with its broad hardware support could attract many people from the other platforms if it just really had all the functionality it “kind of” had already — altitude hold, GPS + compass etc.

I’ve just got a few sonata and lidars and will be trying them with my new F7 builds. In Betaflight, of course. I’m not giving up on it just yet.

aadhoc, 2018-11-23:
> I’ve never understood the single-purpose drone idea. When I travel for business / vacation, I can’t take multiple drones with me. That’s why I build all my drones so that they could possibly do 5g+ [...]

Yes, Yes!
I've got whoops for playing around, and drones with 5" and 8" props for general acro, exploring and video. I *LOVE* that Betaflight is running on all of them, while the limitations are unfortunate. I don't want to build a big dedicated video rig with big hardware and new unfamiliar FC software (same as Andrea above). 

I was just configuring to use GPS HOLD, when I discovered this thread and that the feature isn't ready/implemented. I have a GPS and compass on my custom F7 quads, and was hopeful.

I love Betaflight, and am excited to see where it goes in the future.

marcelinojorgeromero, 2020-01-20:
the only thing i wish betaflight had is **gps hold** and would be perfect.