betaflight/betaflight #6020: OmniNXTF7 Baro drop to negative Values

RoadyFPV, 2018-06-01:
OmniNXTF7 Baro shows on Bat conect 0m and drop to around -2m to -5m.
Also the OSD shows not corect negative value. 
I am at 0m and go down to -1.5m the "-" sign will be displayd first at 1,0m

The OmniNXT use a new Baro type?

jflyper, 2018-06-02:
@RoadyFPV 
Yes, OmniNXT F7 uses a new baro chip LPS, but baro readings drifting into negative region is not a bug by itself. I think OSD presentation is the problem.

@conkerkh @DanNixon Thoughts?

conkerkh, 2018-06-02:
@jflyper  Do you know if it's LPS25HB or LPS22HB?

mikeller, 2018-06-02:
@jflyper: My thoughts too - negative altitude readings in OSD used to work, so it's probably a recent regression.

jflyper, 2018-06-02:
@conkerkh It's LPS25HB.

McGiverGim, 2018-06-02:
The problem with the osd is because multiply zero by -1 is zero. 

https://github.com/betaflight/betaflight/blob/master/src/main/io/osd.c#L273

mikeller, 2018-06-02:
Lol, yes, good find @McGiverGim. :-)

McGiverGim, 2018-06-02:
The easy solution is to use a float, but I suppose that is being done with integers, abs and conditionals to not make a float operation by performance?

mikeller, 2018-06-02:
@McGiverGim: It's not so much a question of performance, but one of size: The `tfp_printf` library that we are using for string formatting, which is considerably smaller than `stdio`, does not have support for float formatting. I think the easiest will be to explicitly set the `-`.

DanNixon, 2018-06-02:
>  I think the easiest will be to explicitly set the -

I'm pretty sure that is how it used to be done. Not sure why it would have been changed.

conkerkh, 2018-06-03:
Btw from my own experience LPS25xx is very unstable baro and definitly not usable for alt hold. It should be avoided and instead LPS22xx or LPS33xx should be used if we talk about ST barometers.

PR #6031 (2018-06-05): Fixed OSD altitude display for negative altitude > -1.
Fixes #6020.
closed 2018-06-05 by mikeller