betaflight/betaflight #6025: reintroduction of ** ibus_sensor = 1,2,3,4,0,0,0,0,0,0,0,0,0,0,0** for F3 Targets.

lamerton, 2018-06-02:
Hi guys,

I have 2 x BFF3 FC boards and run a Flysky I6 with Flit 10 RX, which has telemetry functionality for IBUS to receive voltage telemetry to the FS-I6. Since release 3.25 was updated this functionality has vanished and is only available on my BFF4 in the new releases.

Could you please reintroduce the *ibus_sensor = 1,2,3,4,0,0,0,0,0,0,0,0,0,0,0* functionality for the F3 targets.

Keep doing great work, Cheers.

mikeller, 2018-06-03:
@lamerton: The problem that we are facing for F3 based flight controllers is that they come with a relatively small amount of flash space. The improvements and optimisations in flight behaviour that we made have resulted in an increase in size of these parts of the firmware. As a trade off, we have had to remove other parts of the code from the firmware that is built for F3 based flight controllers, amongst them the extended IBUS telemetry.

What you can do to get around this is to build your own version of the firmware, and by disabling things like RX / telemetry protocols that you don't use, you'll probably be able to reclaim enough space to fit in the features you want.

There are instructions available online on how to build a custom firmware:
https://github.com/betaflight/betaflight/wiki/Building-Betaflight-on-Windows,-the-Linux-subsystem-way
https://github.com/betaflight/betaflight/blob/master/docs/Customized%20Version.md

lamerton, 2018-06-03:
@mikeller

Thank you for the rapid response.
Unfortunately, I don’t have windows 10 or know anything about coding. Is there anyone out there who might be able to do this mod to the F3 firmware for me?

Cheers.

closed 2018-07-10 by stale[bot]