betaflight/betaflight #6026: My OMNIBUSF4 SoftSerial hacks don't work in 3.3.3, but worked in 3.3.2

andrejpodzimek, 2018-06-02:
First of all, long story short: I traced it back to one particular file modification that causes the problem (`git diff v3.3.2 v3.3.3 src/main/target/OMNIBUSF4/target.h`):

```diff
diff --git a/src/main/target/OMNIBUSF4/target.h b/src/main/target/OMNIBUSF4/target.h
index a64e67562..af301018f 100644
--- a/src/main/target/OMNIBUSF4/target.h
+++ b/src/main/target/OMNIBUSF4/target.h
@@ -135,6 +135,10 @@
 #define MAX7456_SPI_CLK         (SPI_CLOCK_STANDARD) // 10MHz
 #define MAX7456_RESTORE_CLK     (SPI_CLOCK_FAST)
 
+// Globally configure flashfs and drivers for various flash chips
+#define USE_FLASHFS
+#define USE_FLASH_M25P16
+
 #if defined(OMNIBUSF4SD)
 #define ENABLE_BLACKBOX_LOGGING_ON_SDCARD_BY_DEFAULT
 #define USE_SDCARD
@@ -149,12 +153,18 @@
 
 #define SDCARD_DMA_CHANNEL_TX                   DMA1_Stream4
 #define SDCARD_DMA_CHANNEL                      0
+
+// For variants with SDcard replaced with flash chip
+#define M25P16_CS_PIN           SDCARD_SPI_CS_PIN
+#define M25P16_SPI_INSTANCE     SDCARD_SPI_INSTANCE
+
 #elif defined(LUXF4OSD)
 #define ENABLE_BLACKBOX_LOGGING_ON_SPIFLASH_BY_DEFAULT
 #define M25P16_CS_PIN           PB12
 #define M25P16_SPI_INSTANCE     SPI2
 #define USE_FLASHFS
 #define USE_FLASH_M25P16
+
 #else
 #define ENABLE_BLACKBOX_LOGGING_ON_SPIFLASH_BY_DEFAULT
 #define M25P16_CS_PIN           SPI3_NSS_PIN
@@ -252,8 +262,6 @@
 #define RANGEFINDER_HCSR04_ECHO_PIN        PA8
 #define USE_RANGEFINDER_TF
 
-#define DEFAULT_RX_FEATURE      FEATURE_RX_SERIAL
-
 #define DEFAULT_FEATURES        (FEATURE_OSD)
 
 #define DEFAULT_VOLTAGE_METER_SOURCE VOLTAGE_METER_ADC
```

Reverting this^^^ change makes my little dronie ~great~ fly again.

Sadly, I have to admit that it's not a stock Betaflight setup. I'm using [this stack](https://www.banggood.com/HGLRC-XJB-F440-TX20_V2-Omnibus-F4-Flight-Controller-OSD-40A-Blheli_32-3-5S-ESC-25100200350mW-VTX-p-1242937.html) and here's how I came to modify the code:

1. It's an F4 and I want SmartPort to work. Out of the 3 free UARTs, 1 has S.Bus, 3 has IRC Tramp (VTX) and 6 has BLHeli_32 telemetry. No place to install the half-duplex diode. &rarr; OK, SoftSerial it is.
2. I configured a timer for `PA0` to have SoftSerial on the RSSI pad which I'm not using for RSSI.
3. Oh no, there's a timer conflict with LEDs! But I want my LEDs to shine! &rarr; OK UART 6 only uses RX for BLHeli_32, TX is free and can have a timer. &rarr; Let's put LEDs on TX6 to avoid the timer conflict.

The thoughts above expressed as a diff (the `USED_TIMERS` had been wrong, BTW):

```diff
diff --git a/src/main/target/OMNIBUSF4/target.c b/src/main/target/OMNIBUSF4/target.c
index 92b9b3fb6..a9eb76657 100644
--- a/src/main/target/OMNIBUSF4/target.c
+++ b/src/main/target/OMNIBUSF4/target.c
@@ -45,10 +45,13 @@ const timerHardware_t timerHardware[USABLE_TIMER_CHANNEL_COUNT] = {
     DEF_TIM(TIM2,  CH3, PA2,  TIM_USE_MOTOR,               0, 0), // S4_OUT D1_ST1
 
 #if defined(OMNIBUSF4SD) || defined(EXUAVF4PRO)
-    DEF_TIM(TIM5,  CH2, PA1,  TIM_USE_MOTOR,               0, 0), // S5_OUT
+    DEF_TIM(TIM5,  CH1, PA0,  TIM_USE_NONE,                0, 0), // SoftSerial
+    DEF_TIM(TIM2,  CH2, PA1,  TIM_USE_MOTOR,               0, 0), // S5_OUT
     DEF_TIM(TIM4,  CH1, PB6,  TIM_USE_LED,                 0, 0), // LED strip for F4 V2 / F4-Pro-0X and later (RCD_CS for F4)
 #else
-    DEF_TIM(TIM5,  CH2, PA1,  TIM_USE_MOTOR | TIM_USE_LED, 0, 0), // S5_OUT
+    DEF_TIM(TIM5,  CH1, PA0,  TIM_USE_NONE,                0, 0), // SoftSerial
+    DEF_TIM(TIM2,  CH2, PA1,  TIM_USE_MOTOR,               0, 0), // S5_OUT
+    DEF_TIM(TIM3,  CH1, PC6,  TIM_USE_LED,                 0, 0), // TX6 -> LED
 #endif
     DEF_TIM(TIM1,  CH1, PA8,  TIM_USE_MOTOR,               0, 0), // S6_OUT
     DEF_TIM(TIM1,  CH2, PA9,  TIM_USE_NONE,                0, 0), // UART1_TX
diff --git a/src/main/target/OMNIBUSF4/target.h b/src/main/target/OMNIBUSF4/target.h
index a64e67562..221513fd5 100644
--- a/src/main/target/OMNIBUSF4/target.h
+++ b/src/main/target/OMNIBUSF4/target.h
@@ -266,10 +266,9 @@
 #define TARGET_IO_PORTC (0xffff & ~(BIT(15)|BIT(14)|BIT(13)))
 #define TARGET_IO_PORTD BIT(2)
 
+#define USABLE_TIMER_CHANNEL_COUNT 16
 #if defined(OMNIBUSF4SD) || defined(EXUAVF4PRO)
-#define USABLE_TIMER_CHANNEL_COUNT 15
-#define USED_TIMERS ( TIM_N(1) | TIM_N(2) | TIM_N(3) | TIM_N(4) | TIM_N(5) | TIM_N(10) | TIM_N(12) | TIM_N(8) | TIM_N(9))
+#define USED_TIMERS ( TIM_N(1) | TIM_N(2) | TIM_N(3) | TIM_N(4) | TIM_N(5) | TIM_N(8) | TIM_N(10) )
 #else
-#define USABLE_TIMER_CHANNEL_COUNT 14
-#define USED_TIMERS ( TIM_N(1) | TIM_N(2) | TIM_N(3) | TIM_N(4) | TIM_N(5) | TIM_N(12) | TIM_N(8) | TIM_N(9))
+#define USED_TIMERS ( TIM_N(1) | TIM_N(2) | TIM_N(3) | TIM_N(5) | TIM_N(8) | TIM_N(12) )
 #endif
```

Now this^^^ worked for me up to version 3.3.2, but stopped working in version 3.3.3. When I revert the flash chip support modifications mentioned (far) above, everything works again.

BTW, with the flash chip modifications, it's not just the SoftSerial that stops working! Also my UART 1 (i.e., S.Bus receiver) doesn't work. So this may have some consequences even for the stock firmware after all.

mikeller, 2018-06-03:
@andrejpodzimek: I think you got a problem somewhere in your repository:

```
mikeller@pc:~/git/betaflight$ git diff v3.3.2 v3.3.3 src/main/target/OMNIBUSF4/target.h
diff --git a/src/main/target/OMNIBUSF4/target.h b/src/main/target/OMNIBUSF4/target.h
index 5850cd9d2..af301018f 100644
--- a/src/main/target/OMNIBUSF4/target.h
+++ b/src/main/target/OMNIBUSF4/target.h
@@ -262,8 +262,6 @@
 #define RANGEFINDER_HCSR04_ECHO_PIN        PA8
 #define USE_RANGEFINDER_TF
 
-#define DEFAULT_RX_FEATURE      FEATURE_RX_SERIAL
-
 #define DEFAULT_FEATURES        (FEATURE_OSD)
 
 #define DEFAULT_VOLTAGE_METER_SOURCE VOLTAGE_METER_ADC
mikeller@pc:~/git/betaflight$
```

jflyper, 2018-06-03:
@andrejpodzimek 
v3.3.3 SBUS works on Omnibus F4 AIO (original).

1. With v3.3.3, do `defaults` (or full-erase flash).
2. Configure UART1 as SBUS

Can you do the same?

andrejpodzimek, 2018-06-03:
> @andrejpodzimek: I think you got a problem somewhere in your repository:

Yes, now I see it's broken. Git is quite infamous for various inconsistencies of this kind and here we go, I got burned again. :angry: A fresh clone shows exactly the diff you've posted. There are 24 commits missing from my repository.

I guess the master must have been re-tagged and for some reason the changes didn't propagate.

`git checkout v3.3.2; git log -n 1;` on a fresh clone:

```
commit 64a292f3725ed0c8c86d82ed2bb067b8aac0cf53 (HEAD, tag: v3.3.2)
Author: Michael Keller <github@ike.ch>
Date:   Fri May 4 21:11:49 2018 +1200

    Merge pull request #5765 from etracer65/gps_speed_units
    
    Fix GPS speed and max speed to respect the units selection
```

`git checkout v3.3.2; git log -n 1;` on my repository:

```
commit cba6db4f15a8461902ff883b1dcd9885be18ed05 (HEAD, tag: v3.3.2)
Author: Michael Keller <github@ike.ch>
Date:   Wed Apr 11 23:32:32 2018 +1200

    Fixed help text for 'diff defaults'. (#5668)
```

That's why Git has never been my favorite version control system, although I've had to use it more than any other VCS. This is just annoying.

OK, I'll try to rebuild this from a fresh clone with my hacks patched in. It may well be the case that what I'm reporting has been fixed already.

andrejpodzimek, 2018-06-03:
> It may well be the case that what I'm reporting has been fixed already.

Actually, that's most likely not the case, because my v3.3.3 tag and its derivatives appear to be unaffected by the Git inconsistency, at least as far as `git log` is concerned.

> @andrejpodzimek
> v3.3.3 SBUS works on Omnibus F4 AIO (original).
> 
> 1. With v3.3.3, do defaults (or full-erase flash).

I always full-erase and then restore from `diff all`. That was also the case here.

> 2. Configure UART1 as SBUS (and nothing else)
> 
> Can you do the same?

Well, that sounds like a lot of re-flashing hassle. OK, I can try that once I get back to this build, but I don't really doubt that it would work with S.Bus alone. The new issue will cause it to stop working with SmartPort/SoftSerial, LEDs on TX6 or both.

Well, maybe I should switch to F.Port, which wouldn't solve the inversion problem on F4, but at least it would require one UART less. :confounded:

mikeller, 2018-06-03:
@andrejpodzimek:

> I guess the master must have been re-tagged and for some reason the changes didn't propagate.

Yes, it has. Unfortunately GitHub's web interface is such that it makes it very easy to mix up the buttons for saving the draft version of an unpublished release, and publishing the release. I accidentally published v3.3.3 early when wanting to update the draft, and this caused GitHub to tag the tip of the release branch with `v3.3.3`...

cudakeer808, 2018-06-03:
I have to disable some “parallel_ppm” something in cli to get my softserial back on 890 nightly. Might work on 3.3.3 too.

mikeller, 2018-06-03:
@cudakeer808: `PARALLEL_PWM` is the RX protocol that is chosen before you configure your RX. As soon as the RX configuration is done, this will be disabled, no need to disable it first (it will be re-enabled anyway if you don't enable another RX protocol).

mikeller, 2018-06-04:
Ok, I can see the problem now - the ordering that the config is applied in means that  `RX_PARALLEL_PWM` does not get disabled before the config for SOFTSERIAL is applied. I don't think there is much we can do for this though, because pretty much all other configuration requires the serial ports to be configured first.

closed 2018-07-11 by stale[bot]