betaflight/betaflight #6028: Compass is unusable (and likely dangerous) in Betaflight

andrejpodzimek, 2018-06-02:
I have 3 instances of HMC5983 and one QMC5883 (fake HMC5883) connected to `BLUEJAYF4`, `ANYFCF7` and `OMNIBUSF7V2`. None of them works. After getting through the initial hassle of getting all the devices detected, I found a few disturbing malfunctions that make a compass unfit for flight with Betaflight.

1. Calibration always succeeds, no matter if you get the sensor orientation setting right. There are no warnings or errors if you don't get it right. That's already quite dangerous.
2. Some of the sensor orientation settings work just like placebo, i.e., they don't work at all. The *flip* configuration has **no effect** whatsoever, it "works" (i.e., malfunctions) the same way no matter if you set *flip* or not. The 0° / 90° / 180° / 270° configuration mostly gives a bit of a hint when it's wrong (the compass on the Setup panel spins fast all by itself).
3. When you let the drone sit on the floor switched on, the compass will eventulally **reverse its direction** in about 2&ndash;3 minutes. Just wait long enough and watch. This happens faster when you keep an eye on the compass on the Setup panel and rotate the drone slowly to keep the compass at north. A few minutes later you will be pointing to the south and beyond, with the compass showing north! When you rotate the drone to point to the north again, it will show south initially and then slowly drift back to the north within a couple of minutes. **This also happens in flight.** It happens regardless the *flip* setting; *flip* has no effect.

I'm not sure whether this is an issue with this particular device type (QMC5883, HMC5883, HMC5983) or whether it's a more fundamental issue in Betaflight, but I can't really imagine flying on GPS or on compass yaw hold under these circumstances. That just asks for a bad crash.

It may be better to disable the particular device type or compass in general in Betaflight until this gets resolved. I took the time to re-calibrate the compass under **all** 8 possible orientations and the result is plain and simple: It doesn't work.

jflyper, 2018-06-03:
Can't reproduce the raw mag *reversal* phenomenon.

Tested under v3.4, with following configuration. 

SPRacing F3 : HMC5883L (onboard) : Stable for 10min.
Omnibus F4 V2 (OMNIBUSF4SD target) : HMC5883L (on BN880 compat puck) : Stable for 10 min.
Omnibus F7 V2 (OMNIBUSF7V2 target) : HMC5893L (GY-282 in I2C mode) : Stable for 30+ min.

jflyper, 2018-06-03:
However, I do feel like there's some bad interaction between mag heading and gyro-based heading.

McGiverGim, 2018-06-03:
My HMC5883 works right with MATEKF405.
What you describe usually happens when your mag orientation is not right.

andrejpodzimek, 2018-06-03:
The raw data appears to be stable — as long as it’s within ±1. Sometimes two of the axes show crazy values like +12 and +14.

I have tested all the 8 mag orientations and a of them malfunction. Some of the orientations yield an obviously wobbly heading (on the compass instrument in the GUI) and some “only” cause the very slow reversals. In particular, “flip” has no obvious effect.

There must be indeed something wrong about the interaction between compass and gyro. It also surprises me that the GUI lets me start the compass calibration even before the three beeps. I’m not sure whether compass calibration makes any sense without a calibrated gyro.

andrejpodzimek, 2018-06-03:
BTW, stability while pointing towards the north is mostly much better than for other orientations, which can be deceiving. (It made me believe than my compasses worked for ~months.) A simple test to reveal the problem:

`for side in ['south', 'east', 'west']:`\
` ` ` ` ` ` ` `Point the drone to the `side`, double-check whether the GUI instrument matches and wait for a while … it will drift.

andrejpodzimek, 2018-07-06:
Just a poke to tell @stale that this is still a problem.

andrejpodzimek, 2018-08-06:
Nope, the issue is still there. Sometimes this stale bot can get quite annoying. It would be nice if issues like this could be flagged “long-standing problem” or the like, because it’s neither a minor problem, nor something that’s likely to be fixed in the near future, considering that the GPS hold mode has been removed.

McGiverGim, 2018-08-06:
@andrejpodzimek the problem with this is that only you have this problem. I use the compass, I use the GPS Rescue, and I don't have any problem with it. I have always in the OSD the RTH arrow, and it works without problem. To me is very reliable, at least with the use of Betaflight (that does not have autonomous flight modes). 

I will like to help you, but without reproducing the problem I can't.

mikeller, 2018-08-06:
Closing this - two testers can not reproduce.

closed 2018-08-06 by mikeller

infrared4ever, 2023-07-02:
Hello you all,
I have got the same bad behavior with a magnetometer inside of a GPS device. I worked for days on the problem to fix the glitch.
This is what I have done to fix the problem:
- 4.4µF condensator at the input of the power inside of the GPS module
- 2 x 3k resistor as pull up resistors on the I2C port (SDA & SCL). The size of resistors is 0805 (there is no room to place common resistors). Resistors are connected between the positive 5v power wire and the ports (SDA & SCL).
- According to the position of the magnetometer inside of the GPS box, we need to show: 270 CW FLIP
- If you are working with Betaflight you could disable the magnetometer (but the magnetometer is still powered up...) and have a try to see whether the compass is working well at showing the true North and following the positions of your drone.
- To start a flight you will need to place your drone facing the true North if you disable the magnetometer (ACC and Gyro will make the job)
- I know that it is a bit tricky but my magnetometer is working well and it is possible to enable the GTPS RESCUE on my drone.
- Good luck