betaflight/betaflight #6033: Feature Request:

spatzengr, 2018-06-03:
Make a mode to toggle the FC into MSC mode.  This way users could put on a switch on the TX.
1. Plug-in USB (on may FCs that powers to RX)
2. Toggle the switch on TX to flip into MSC mode.

[for bonus option: have a variable for auto-download and erase.]

mikeller, 2018-06-03:
@spatzengr: As an alternative, this could be implemented as a function in the TX lua scripts, for users who don't have a spare channel they can use for thing they won't ever need in flight. Also, be aware that if this is implemented, this will be an immediate kill-switch, and if it's accidentally activated while in the field, the craft will stop responding to anything until it's power cycled.

Re the auto-download, this would have to be done on the host (Windows) side. But as far as I know, there are ways to trigger these actions on media insertion.

DieHertz, 2018-06-03:
... which could be considered malicious by some antivirus software :-)
We need a full-fledged configurator app and a service running in background, the way it's done for software like iTunes which synchronize media and library upon insertion

tavdog, 2022-05-07:
Is this still a wanted feature?

closed 2022-05-08 by haslinghuis