betaflight/betaflight #6039: SmartAudio Settings Refresh

linus1999, 2018-06-03:
I wanted to use Audio parallel to SmartAudio which is totally possible:
http://team-blacksheep.freshdesk.com/support/solutions/articles/4000100084-tbs-smartaudio-what-it-is-and-how-it-works

The problem is, betaflight seems to refresh the VTX Settings all the time with 2 Hz, which results in a weird ticking-like noise.

Solution would be to update the settings only, when entering the OSD or the Lua Script is started, and after changing the settings.

Can anyone tell me, where the refreshing of the vtx-Settings is handled?

wind0r, 2018-06-03:
Just arm and vtx will stop getting updates ;)

linus1999, 2018-06-03:
Ahaa, great, thanks:D

mikeller, 2018-06-03:
@linus1999: Is this solved? If so, can you please close this issue?

closed 2018-06-05 by linus1999