betaflight/betaflight #6045: Lost motor using DSHOT protocol

dengorman, 2018-06-04:
Updated my Asgard32 F7 to latest 3.3.3. Using Multishot all motors working. Use Dshot, one motor doesn't spin. Flashed with ButterFlight 3.5.1 hex from Airbot, all working.

jflyper, 2018-06-05:
This is a bug, induced by combination of followings.

1. F4 and F7 has different `dmaopt` ordering on TIM8 channels.
F4:
https://github.com/betaflight/betaflight/blob/c648c5460ce2330d86986edccc3c642ad3648cea/src/main/drivers/timer_def.h#L451-L453
F7:
https://github.com/betaflight/betaflight/blob/c648c5460ce2330d86986edccc3c642ad3648cea/src/main/drivers/timer_def.h#L556-L558

2. 1. AG3XF4 and AG3XF7 share a single `target.c` file, in which a motor output from PC8 (TIM8_CH3) is configured with `dmaopt = 0`, which is `D(2,4)` for F7.

3. `D(2,4)` collides with ADC1.

4. `ENABLE_DSHOT_DMAR` is not defined (defaults to `false`).

jflyper, 2018-06-05:
I will be fixing this by
1. Defining `ENABLE_DSHOT_DMAR` to `true`.
2. Fixing `dmaopt` for non-burst operation.
Personally, I want to remove inconsistencies in how `dmaopt` is handled among different MCUs, but this requires scanning entire `target.c` ... hmm... thoughts???

dengorman, 2018-06-05:
Great, thanks jflyper! I wonder how it's working for Diehertz? 🤔

jflyper, 2018-06-05:
Using ESC telemetry instead of onboard ADC mask the problem (this is another workaround).

dengorman, 2018-06-05:
Oh right! Noted.

DieHertz, 2018-06-05:
@jflyper just reorder dmaopt entries in `timer_def.h`

silversurfer48, 2018-06-06:
i have one motor out on flip 32 f4 board and it was working fine up until I flash again for the repair on the video cam

mikeller, 2018-06-06:
@dengorman: Can you please check if this is fixed with the latest development build of 3.4, and if so, close this?

dengorman, 2018-06-06:
@mikeller Sure, where do I download the development build of 3.4?

closed 2018-07-07 by mikeller