betaflight/betaflight #6050: Super_s F3

OGblmk, 2018-06-05:
After finishing my 90mm quadcopter i plugged it into the pc and opened betaflight configurator. I cant seem to flash the newer software it says succesfully closed serial port but it says failed to open serial port. and if i try to connect nothing happens. If i unplug the usb and plug back in it works as normal without the firmware flashed. I choose omnibus on the flashing tab and 3.3.3 software. But it cant flash the board. And i cant get the board into DFU mode aswell. I soldered my boot joint and if i connect the board after i have soldered the joint. Nothing happens .

jflyper, 2018-06-05:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

closed 2018-06-05 by jflyper

richard-scott, 2018-06-06:
You don't solder the boot joint, you briefly short-circuit it when connecting USB and then remove the metal device you used to join the two boot pads.