betaflight/betaflight #6053: Arming only works once (possibly only when crash flip mode is enabled) and never again until power cycle.

flint723, 2018-06-06:
I first saw the issue myself with one of the builds (I think before it was rebased) on the 3.4 default PR.  But another friend of mine reported the problem on build 900 (currently trying to confirm he had crash flip mode enabled).

However @Saxin has the issue documented in a video with commit 6bac0cd and c0d25b1.

Video: https://www.dropbox.com/s/5iayfke31wt5l99/2018-06-03%2016.07.11.mp4?dl=0

I have not pursued further to see if the issue still exists in current master as of 6/6/2018.

flint723, 2018-06-06:
Also, @Saxin said the problem was only there on omnibusf4sd target, regular omnibus is fine.

cudakeer808, 2018-06-06:
I have same issue with OmnibusF4 hex on HGLRCF440.

Saxin, 2018-06-06:
@cudakeer808 - please check and advise if its related to "flip after crash" mode being configured vs. not in modes tab ?

cudakeer808, 2018-06-06:
@Saxin, I do have have on in the modes tab.

mikeller, 2018-06-06:
Guys, is this something that can be reproduced repeatedly on any craft that it's happening on, or does it happen only once, and then it's good again?

Also, we need diffs in order to identify and fix this.

Saxin, 2018-06-06:
@cudakeer808 - please try to check if it's related to "flip after crash" mode being set in modes tab - try when set and when not set. (In my case this was the offensive feature - with it set I can arm only once, without it set - I can arm regularly)

**Remark:** When say set I mean setup in modes tab, not activated.

flint723, 2018-06-06:
So far I wasn't able to reproduce on the bench with 900 OMNIBUSF4 (hglrc F440) or the same quad that it happened on for me (impulsercf3).  In both cases accel is on and max arm angle set to 180.  Both have turtle mode set up as a switch, however when I arm I have to move through turtle mode to get to armed on a 3 position switch.

Will try to test a little with battery on the quad it happened on for me (I haven't reflashed it since it happened).

cudakeer808, 2018-06-06:
@Saxin, yes I have it assigned on a switch in modes tab. On both quad.

Saxin, 2018-06-06:
@cudakeer808  - Can you please try **without** it assigned (in modes tab) - see if issue is resolved ?

cudakeer808, 2018-06-07:
@Saxin, I’ve already flashed 903 on it and can no longer replicate the issue.

Saxin, 2018-06-08:
Flashed 911 today on OmnibusF4SD target and can't replicate any more.

flint723, 2018-06-08:
I haven't had the problem on later builds either.  Should I close this?

Saxin, 2018-06-08:
yes Ive tested on two boards, its ok now

flint723, 2018-06-08:
Closed for now because it seems to be no issue anymore.

closed 2018-06-08 by flint723

Saxin, 2018-06-14:
@Randy45 - please post in issue #6088 , please provide target name and steps to reproduce.

RANDY45, 2018-06-14:
3.4.rc1 disable turtle mode it works and I did post on issue page but it
says it's closed

510Rascloud, 2018-06-17:
Hey explain like I’m lame how do I get msp off of hobbywing xrotor f4 omnibussd