betaflight/betaflight #6057: Current sensor value overflow

wysiwyng, 2018-06-06:
When using the built-in hall current sensor on an Omnibus Fireworks V2, the current in the blackbox log shows erroneous peaks going all the way up to 650 A. When disabling unit display in the blackbox viewer, "650 A" shows up as a value in the ballpark of 65520 - 65535. 

[image]

The problem should be reproducible by using any FC with ADC current sensor input and setting a large **negative** ```ibata_offset``` (maybe via CLI, in my case a value lower than the GUI allows is needed). The problem shows only in the Blackbox logs, in GUI or Spektrum telemetry no such peaks are present. I killed my other FC I could have used to test this behaviour, so verified reproducible only on Omnibus Fireworks V2.

I had a look at the necessary code in ```src/main/sensors/current.c``` and it seems inside that file current is handled with signed variables. However once it gets logged to the blackbox the current is cast to an unsigned type, explaining the wrap-around when negative current values are present.

diff output:
```
# version
# Betaflight / OMNIBUSF4FW (OBFW) 3.3.3 May 31 2018 / 13:24:16 (3565182d3) MSP API: 1.38
# name
name Fury 5X
# resources
# mixer
# servo
# servo mix
# feature
feature TELEMETRY
# beeper
# map
map TAER1234
# serial
# led
# color
# mode_color
# aux
aux 0 0 4 900 1100 0
aux 1 1 0 1900 2100 0
aux 2 2 0 1400 1600 0
aux 3 28 5 900 1100 0
# adjrange
# rxrange
# vtx
# rxfail
# master
set gyro_notch1_hz = 190
set gyro_notch1_cutoff = 140
set gyro_notch2_hz = 253
set gyro_notch2_cutoff = 210
set mag_hardware = NONE
set serialrx_provider = SRXL
set blackbox_p_ratio = 128
set motor_pwm_protocol = DSHOT300
set pid_process_denom = 2
# profile
profile 0
set dterm_notch_hz = 0
set dterm_notch_cutoff = 130
set yaw_lowpass = 60
set p_pitch = 40
set i_pitch = 40
set d_pitch = 15
set d_roll = 15
# rateprofile
rateprofile 0
set roll_rc_rate = 130
set pitch_rc_rate = 130
set yaw_rc_rate = 130
set yaw_expo = 20
```

sarunoatama, 2018-06-10:
What is your offset and scale? Can you post your blackbox log?

wysiwyng, 2018-06-10:
[BTFL_BLACKBOX_LOG_Fury_5X_20180605_204522.BBL.tar.gz](https://github.com/betaflight/betaflight/files/2087710/BTFL_BLACKBOX_LOG_Fury_5X_20180605_204522.BBL.tar.gz)

Offset is -18500, scale is 197.

PR #6082 (2018-06-13): Make current in blackbox signed (fixes #6057)
This makes measured current in the blackbox signed. Fixes #6057.

Maybe just making the current signed is not what you want, one could expand this to a feature to either log the current as-is with sign, log its absolute value or just truncate it to never go below 0.
commit 11d8171c (2018-06-13): Merge pull request #6082 from wysiwyng/signed_current

Make current in blackbox signed (fixes #6057)
closed 2018-06-13 by mikeller